keyleds/keyleds #2: Planned features

spectras, 2017-07-26:
Mooaaaaare plugins!
Seriously though, the skeleton is mostly done, all it needs now is a richer set of animations.

For showing off, those would be cool to have:

 * Wave-style animation (based on key layout).
 * Star animation (light up random keys and let them fade out)
 * Keyboard-only game. Whack-a-mole using colors? Minesweeper? You name it.

More serious features:

 * Keypress feedback (requires matching hiddev device with Xinput device - it's mostly done but untested).
 * Application feedback (implement a plugin skeleton that allows application-specific to set key colors depending on internal application state, eg: emacs/kate plugin, email notification, etc…)
 * Filters plugins. Most notably: low-pass filter (to smooth transitions), dawn filter (tune down a color, possible less blueish at night).

Now on the system side of things, there are some fields that need more work, namely:

 * initial fade-in.
 * untying of plugin lifecycle from context changes, so they have the opportunity to linger / play fade out animations / etc.
 * systemd interaction and seat management. At the moment we just grab all devices we have permissions for.

spectras, 2019-02-01:
Some more:

 * [x] Drop Qt dependency for keyledsd service.
 * [ ] Add automated tests, in particular for public APIs (plugins and lua).
 * [ ] Add support for G-keys.

msagerup, 2019-04-13:
Amazing work. I finally got it to work, after 2 days of scratching my head. I am having alot of fun with your program.

spectras, 2019-04-13:
Hello again @msagerup,
Thanks for your message, it's great to have feedback.

Out of curiosity, would you mind describing what made you scratch your head? The goal is to make it head-scratching-free, so I would love to know what I can improve :)

msagerup, 2019-04-15:
I had some issues with the looping, and I did not know how to build the program with "make" I also had some issues when trying to run the program after using "make" . 
I keept getting this: "error while loading shared libraries" 

After some searching I found the solution here. https://itsfoss.com/solve-open-shared-object-file-quick-tip/

It been a good learning curve for me, since I am not used to digging around in the root system. I also had some trouble with finding the profile. I belive it was in the etc ?folder. 

+ It is the first time trying to code with Lua. So that has been a  challange also. 
I am building  a little program that light up keys I commonly use when coding, they will light up when I press the shift key. So while I am here, I cant seem to find anything on a key hold feature? Is it something you will add ? I am using the keypress atm. That works also. But it would work better to check if its beeing pressed and release a trigger when unpressed. 

OMG> While typing this. I recall there is a paramater called isPressed!. Can that be used to check if a key is beeing held down?

spectras, 2019-04-15:
@msagerup > yes, compiling manually might be a bit harder than installing the packaged version.
Had you seen there is an [install guide](https://github.com/keyleds/keyleds/wiki/Installing)?

For your key hold, yes, that's indeed what isPress is for. It will be `true` if the key is being pressed, and `false` if the key is being released (therefore one keystroke always generates 2 events, the first one with true and the second one with false).

If you want to keep the information around you can store it in some variable, eg:
```lua
shiftHeldDown = false

function onKeyEvent(key, isPress)
    if key.name == "LSHIFT" then
        shiftHeldDown = isPress
    end
end
```
Now, in all the rest of your script you can use `shiftHeldDown` to know whether the left shift key is held down right now.

msagerup, 2019-04-15:
Amazing! Thank you !