keyleds/keyleds #3: Need to replug device after login

pyrocks66, 2017-09-15:
I installed the ppa on my ubuntu. Program works fine, but after I open a new session, keyboard does not respond to events. Unplugging and replugging my keyboard fixes the problem.

spectras, 2017-09-15:
Hello,

Thanks for reporting. I have seen this happen a few times. It seems to come from a backwards time adjustment that causes the animation routine to freeze. Technically it doesn't crash, it's just that if your system time is adjusted backwards, say, 2 hours, it will believe next animation refresh has to wait 2 hours + usual refresh time.

Alas, I don't have a solution right now, as the issue comes from underlying thread library itself (namely, `pthread_cond_timedwait`). While I think about a fix, as you noted, you can replug your keyboard, that will reset the animation routine.

spectras, 2017-09-15:
I just published v0.4.2, which includes a workaround.

closed 2017-09-15 by spectras