keyleds/keyleds #10: Unable to start service

spacemeld, 2017-10-13:
An error occurs when starting the service:

`keyledsd: /build/keyleds-OukYOw/keyleds-0.6.1/keyledsd/src/tools/accelerated_sse2.c:33: blend_sse2: Assertion 'length % 4 == 0' failed.
Aborted (core dumped)`

Trying this out on Ubuntu 17.04 with a G810 Orion Spectrum

spectras, 2017-10-13:
Hello,
Thanks for reporting, I'll look into it today. I have an idea of what might be going on.

spectras, 2017-10-13:
Hello again.
Issue should be fixed in the pull request I just created. Are you willing to try it, or would your prefer that I release it in packaged form?

spacemeld, 2017-10-13:
I installed the package as noted by your installation instructions.

Usually, I prefer packaged but it wouldn’t make sense to upstream your changes if it’s isnt fully tested. I assume I can simply check out your pull request and follow the steps for a manual install. Let me know if that works for you.

spectras, 2017-10-13:
Well, you may either install yourself (using debuild to build a package would be best) or i can package the pull request version tonight, as you prefer.

If you choose to use debuild, from outside the project directory, run
```
tar czf keyleds_0.6.1.orig.tar.gz --exclude=.git --exclude=debian keyleds
cd keyleds
sed -i "s/:native//g" debian/control
sed -i "s/unstable/zesty/g" debian/changelog
debuild -uc -us
```
That should build the package, unless some dependencies have to be installed (debuild will tell which)

spacemeld, 2017-10-13:
Ok, I won’t be back to my desktop until tonight. I’ll try the debuild method.

spacemeld, 2017-10-14:
I replaced the service using the debuild method. Keyboard responds accordingly without issue.

Thanks!

spectras, 2017-10-14:
Great! Well, thanks to you for taking the time to report and follow.
I will publish a bugfix release this weekend.

If you have time, I'll be interested to know if you have ideas for improvements, and possible defects. For instance, I didn't have a chance to try G keys since my keyboard doesn't have them. So it's very likely the feedback effect won't work on them until I have a chance to inspect how they behave.

spectras, 2017-10-14:
Fix is included in release 0.6.2.
Thanks for your time @spacemeld ! Without you it would still be broken on G810.

If it is not asking too much, I would be very interested in knowing whether the wave effect works fully on G810 (including G keys), how the feedback effects handle those as well and, if possible, a log with full debug output (`keyledsd -vvv`) that would include keypresses on G keys.

closed 2017-10-14 by spectras