keyleds/keyleds #13: g610 missing keys

ibrokemypie, 2017-10-29:
The shift keys, alt keys, meta keys and right control key all get ignored on my g610, showing as "unknown key some  number" when running with -vvv

ibrokemypie, 2017-10-29:
42, 125, 56, 100, 126,  54, 91

spectras, 2017-10-29:
Hello, thanks for reporting here!
Would it be possible for you to tell which is which? Like, 42 is left shift, etc?

zwostein, 2017-11-04:
Hi,

I think I have a similar problem with the G810:
Some of my modifier keys cannot be set to a color and `keyledsd -vvv` reports them as unknown keys when pressed:
- Super Left (unknown key 125)
- Alt Left (unknown key 56)
- Alt Right (unknown key 100)
- Super Right (unknown key 126)
- Control Right (unknown key 97)
- Shift Right (unknown key 54)

The multimedia keys produce unknown key messages, too.
I cannot set them using `keyledsctl set-leds -d /dev/hidraw1 MUTE=ffffff` (no invalid key error, though)
But unlike the modifiers, they seem to work fine when an effect is used.

Is this a layout issue? I'm using a german keyboard btw. c33100000000_0004.xml is loaded.
I've seen that other working keys are named after their US-layout counterpart (e.g. Z and Y are swapped)

The output of `keyledsctl info && keyledsctl get-leds`: [keyledsctl-g810-deu.txt](https://github.com/spectras/keyleds/files/1443640/keyledsctl-g810-deu.txt)

spectras, 2017-11-05:
Hello,

Thanks for taking the time to track this! And you have a G810, that's great, I didn't have a chance to get one to test. Let's see what I can learn from your output.

Hmm, the G810 is internally split into 4 different key blocks, and there are a few tricks it seems. From what I see:

* Multimedia keys are on a separate key block. That's why keyleds fails to find them when pressed and that's why `keyledsctl set-leds` does not find them. However it will still animate them as unknown leds, which matches what you see. This should not be too hard to fix. I believe they should report the following unknown keys values in keyledsd log:
    - Mute: 226
    - Play: 205
    - Stop: 183
    - Prev: 182
    - Next: 181

  Could you please confirm that's correct? And would it be possible to post the output of `keyledsctl get-leds -b media`? I would need that to match them correctly (so keyleds knows which led is which key).

* Now, for modifier keys, oddly enough it seems the keyboard does not report them. On the other hand there are a few strange keys. I wonder whether they are the same. Could you try this ?
    `keyledsctl set-leds RO=blue KATAKANAHIRAGANA=green YEN=red HENKAN=white MUHENKAN=pink`
  => If it does nothing, then those are spurious keys and the G810 does not report missing modifiers at all.
  => If it changes modifiers, then the G810 misreports them as other keys.
  Both should be easy fixes, but I'd need to know which :)

So you're right with the layout issue :) It's not from the layout file though (it's normal keys are inverted, the keyboard actually reports US keys and layout translation is done in the display server). Rather, keyleds fails to interpret it correctly for media keys, and it seems the keyboard incorrectly reports some modifier keys.

If you have some time to answer the questions above, it should be easily fixed.
Thanks for your time!

ibrokemypie, 2017-11-05:
@spectras

left shift-42
left windows-125
left alt-56
right alt-100
right windows-126
right control-97
right shift-54

us layout

zwostein, 2017-11-05:
- Almost:  
  ```
            | key in keyledsd | get-leds -b media
  Mute      | unknown key 113 | xe2
  PlayPause | unknown key 164 | xcd
  Stop      | unknown key 166 | xb7
  Prev      | unknown key 165 | xb6
  Next      | unknown key 163 | xb5
  Just in case - the volume wheel (without LEDs):
  VolUp     | unknown key 115
  VolDown   | unknown key 114
  ```
  And by using `keyledsctl set-leds -b media xcd=ffffff` etc. the colors are set accordingly.

- Those strange keys: when I try to "set-led" them, nothing changes.  
`get-leds` always reads them as #000000 except KATAKANAHIRAGANA, which always contains some previous color. (Actually, it's the color the alt_right key was set to using g810-led in my case)

spectras, 2017-11-05:
Now, that's interesting, the values I expected as unknown key values are those reported by `get-leds -b media`.
Thanks both of you!

I think I now how to fix this, I'll push a commit today :)

spectras, 2017-11-05:
The issue should be fixed on `master` I believe.
Do you think you would have some time to spare and give it a try?
I can give the instructions to package it locally for your distribution if you wish, just tell.

zwostein, 2017-11-05:
That was quick, thanks, the keys now work for me (with effects and set-leds).
Just `keyledsctl set-leds MUTE=#ff0000` (without -b media) now turns on the left alt key ;)

spectras, 2017-11-05:
Great! The fix will be included in next release then.
Well, yes, the keyboard uses the same value for mute key in media group and left alt key in regular group, and I didn't forbid the conversion. I guess I could :).

Thanks for your time. If you will, I'd love to have more feedback, such as which features you're using, how they could be improved, suggestions for new features, ...
Anyway, even if you don't, I'm glad I now know it works 100% on G810. And perhaps G610 when @ibrokemypie will have had a chance to test it too.

spectras, 2017-11-15:
Release 0.7.1 includes the fix, so I'm closing this issue.
Thanks again for reporting!

closed 2017-11-15 by spectras