keyleds/keyleds #14: Looking for G910 owner

spectras, 2017-11-06:
Hello,

Keyleds is looking for someone willing to spend a bit of time to help me add G-key support.
No dev skills required, I would just need to collect some data from that model and the way Linux sees it.

Palanthis, 2017-11-21:
Hey there!

I have a G910 and I have been messing with your (very awesome and much wanted) project. I would be happy to help out in any way I can.

spectras, 2017-11-21:
Hello @Palanthis !

That's great. I am glad you like keyleds and would really appreciate your help.

Well, there are several steps that would allow me to better understand that keyboard model, if you are willing to spend a bit of time:

1. First, simply checking what works right now:
    - Do full-keyboard effects use all keys, or are some of them missing?
    - Do all keys respond to reactive effects?

2. Second, gathering some information on the keyboard. If you can run the following...
    ```bash
    keyledsctl -v info          # see what keyleds finds
    sudo lsusb -vvd 046D:       # list all device bearing Logitech vendor ID
    ```
    And
    ```bash
    sudo evtest
    ```
    This should show a list of event devices and ask to select one. For now, I'm just interested in knowing the list of Logitech devices it finds. This is because the keyboard might show up as several devices.

3. Third, key testing! This would aim at identifying each keys Keyleds doesn't know about yet. So that would be: all specials keys (G keys, M keys, ...) and any key that didn't work in step 1.
    * Run `sudo evtest` again and pick the first device for the keyboard.
    * Try to press each of the keys once, in order (I'll need to know which is which).
    * Hit Ctrl-C to terminate evtest.
    * Do the same again for each device you identified in step 2. The idea is some keys could show up on one device, and some on another.

If you could do this and post the outputs, that would be amazing!
_(and be careful not to type anything sensitive while `evtest` is running, it would show up in the output)._

If we are lucky, this could be enough to understand how Logitech mapped those extra features to events, and from there keyleds could pick them up. If we are less lucky, a USB trace will be needed, but it's a lot more messy so let's hope it won't be required :)

Thanks a lot for your time!

spectras, 2018-01-03:
Update on that one: I managed to buy a second-hand g910, I'll get it in a few days.
Expect some activity here soon!

michaelolson01, 2018-03-13:
I have a 910 if you just wanna pull some info from it.

m1st0, 2019-04-25:
I found this [g910-gkey-macro-support project](https://github.com/JSubelj/g910-gkey-macro-support) which I hoped to combine efforts from. It almost simplifies to sending a initialization code to the keyboard and accessing it accordingly (udev rules, etc).  However, I haven't solved how to match the enumerations of the keyboards between the projects only allowing me to use one or the other, either colorizing it or using G-keys.

spectras, 2019-04-25:
@m1st0 Hello!
As far as I know, keyledsd should function properly alongside the project you linked.
What exactly do you need? I don't understand the "match the enumerations of the keyboards between the projects" part? Would you care detailing a bit more?

m1st0, 2019-04-25:
When I run _g910-gkey-macro-support_, python's uinput takes over.  I am then unable to get keyledsd to actually see the keyboard as it immediately stops effects.  Restarting keyledsd results in `<E>service: not opening device /dev/hidraw1: feature not found on device` .  I am confused therefrom as to how to get both to play along.  Here is some more info of what is done from their side:

- [Initial understanding of g-key access](https://github.com/CReimer/g910-gkey-uinput/issues/3)
- [Current code for g-key access initialization](https://github.com/JSubelj/g910-gkey-macro-support/blob/master/lib/keyboard_initialization/usb_and_keyboard_device_init.py)

So I assume the udev mapping is changed in a way, "matching enumerations" being different in each project?

spectras, 2019-04-25:
Aww, I skipped over that line.
No, you are right, it won't work. The reason is this part in g910-gkey-macro-support:
```python
usb.util.claim_interface(dev, USB_IF)
```
It takes exclusive ownership of the interface, so other programs such as keyleds can longer use it.

There is not much to be done from keyleds side. For further reference, keyleds does not take exclusive ownership, so if g910-gkey-macro-support is updated to play nice, it will work.

spectras, 2019-04-30:
Hardware library now has full support for G keys.
Next step is to implement it in the keyledsd service on top of it.

I'm looking for a second hand G910 right now so I can have G-keys to develop and debug with.
Full support should come in May!

(software architects please don't look at the code, the initial effort will focus on making it work, there will be a refactor after >< )

JSubelj, 2019-06-02:
Hey, just found this issue. My driver releases the interface every loop after it reads from it as seen [here](https://github.com/JSubelj/g910-gkey-macro-support/blob/5caa954019a835906f80950d1f21877e9b1b7405/lib/g910_gkey_mapper.py#L97). If you have any more problems with my software please report them.

In the future if you find problem with another project please inform them by opening an issue. I found this thread purely accidentally and most of the developers don't have time browsing issues from similar software.

Also as I see you are working on GKeys support, we can join forces. If you are up for it contact me.

spectras, 2019-06-07:
Hello @JSubelj 

First, thanks a lot for getting in touch. You are right, it would have been much better to talk to you about this directly.

About the issue at hand, I wonder how the exact details work out, but it turns out while g910_key_mapper is running, the hidraw interface goes down and never comes up again. A wild guess would be that either the hidraw driver does not kick in again once disengaged, or it only does so after some time. But I don't know the internals of Linux usb stack.

I am indeed working on GKeys support, though past month had me more busy than I expected (I struggle finding a second hand g910 as well). I am completely open to joining forces, though I am not sure exactly how. What would you suggest? :)

JSubelj, 2019-06-07:
@spectras no problem.

That's weird. Maybe it's because pyusb is doing some weird stuff in the background, I will also try to change the polling strategy into something less resource consuming, when I'll find the time, maybe that will solve the problem. It also seems weird that [g810-led](https://github.com/MatMoul/g810-led) controller works fine with my driver running, but maybe your controller does a lot more in the background (didn't check it out yet because it failed to install half a year ago).

Well I was thinking that if we figure out what's the problem with running both our drivers simultaneously, you could use my project for your GKeys support. Afaik, the GKey support of my driver works quite well but it does need a GUI that you could provide while I provide the functionality.

spectras, 2019-06-07:
Yes, they are very different projects. `keyleds` is a background service that plays scripted animations, reacting to keypresses and context changes (you can do stuff such as changing the animation when you are viewing a specific website and writing your own animations in LUA).

If you ever used Logitech's Gaming software, the RGB animation thing on windows, you can imagine keyleds as basically that software on steroids, but with no GUI. And if/when it gets a GUI, it will be detachable, blocking 150MB or RAM for animating the keyboard is not good programing in my view (keyleds is <3MB footprint).

On the technical level,
* g810-led uses the same API you use, the USB subsystem through libusb,
* while keyleds uses the HID subsystem, through hidraw.

The Linux HID subsystem driver sits on top of the USB subsystem (in fact, USB is just one of several possible protocols for HID devices).

So I believe what actually causes the issue is that g910_key_mapper detaches the HID driver [here](https://github.com/JSubelj/g910-gkey-macro-support/blob/master/lib/keyboard_initialization/usb_and_keyboard_device_init.py#L36). A quick test with some unrelated hid devices (I'm on holidays without a keyboard) shows that calling `detach_kernel_driver` makes the `/dev/hidraw` device nodes disappear.
I do not know how to solve this yet.

As for a GUI… at the moment we stand at the same spot: keyledsd is a background service that would definitely want a configuration GUI, but has none. It would be a full project. Perhaps some Py/Qt project would be nice, or plain Qt/C++. When all is done, I would love to see some integrated user experience, where current application context can switch GKeys profiles dynamically. :)

nickbclifford, 2019-12-17:
Has this been fixed at all? I still can't use keyleds alongside g910-gkey-macro-support and I would hate to be stuck with g810-led.

spectras, 2019-12-17:
Hello @nickbclifford !

As the previous message explains, `g910-gkey-macro-support` unloads the HID driver, so the keyboard is no longer recognized as a HID device. This is not an issue with keyleds, if you try to uses tools from `hid-tools` package after starting `g910-gkey-macro-support`, they will not see the keyboard either.

I am not sure how this could be resolved on keyleds side, apart from implementing G keys support so you no longer need g910-gkey-macro-support at all.

I was going in this direction (in fact the [latest commit](4917e0490cd984920b9a81c5fbd6fc1e27ff50a2) adds support for Gkeys in the hardware layer of keyleds), but the fact that all programs that deal with keystrokes will be broken with Wayland soon (that includes both keyleds and g910-gkey-macro-support) kind of demotivated me. See #42 for details.

zocker-160, 2021-09-13:
@nickbclifford
I know this issue is almost 2 years old, but I personally had the exact same problem.

Since not much seems to be happening in this regard, I have created my own GUI application for mapping Gkeys and it does **not** unload the HID driver, which means it works alongside of other RGB software like `keyleds` or `OpenRGB` without issues.
You can check it out here: https://github.com/zocker-160/keyboard-center

Since keyleds does "only" need a configuration file, I would be willing to do a colab with @spectras, but he sadly does not seems be active anymore... :/

frostworx, 2021-12-04:
@zocker-160
Wow, your keyboard-center is great!  Thanks a lot for your great work!

zocker-160, 2021-12-05:
thank you @frostworx, appreciate it :heart: 
I hope I can offer a keyleds integration at some point in order to support RGB animations (not just static like with openRGB right now).

unfortunately according to his own comment here: https://github.com/keyleds/keyleds/issues/70#issuecomment-968659109

@spectras does not seem to have any time left for this project. :(

frostworx, 2021-12-05:
Thanks @zocker-160 for linking @spectras official statement, I would have missed it otherwise.
Unfortunately, open-source projects often slow down or even stop completely because of jobs.
In a perfect world everybody would have their hobby as job, RL often looks differently.

zocker-160, 2021-12-05:
yes that is true, I personally would not have the time to take over maintenance of keyleds either.