keyleds/keyleds #15: Assertion Null

m1st0, 2017-12-07:
On Kubuntu 17.10, Logitech G910 keyboard. When I start `sudo keyledsd -vv` (needing sudo otherwise permissions denied error), I get the following:

    service: not opening device /dev/hidraw1: synchronization with device failed
    keyledsd: /build/keyleds-fijYzj/keyleds-0.7.1.1/libkeyleds/src/feature_leds.c:100: keyleds_get_leds: Assertion `keys != NULL' failed.
    Aborted

I also installed logitech.rules into /etc/udev/rules.d/ and restarted udev.

m1st0, 2017-12-07:
I installed from source and fixed some library issues with my system which made it work over installing the deb file.

spectras, 2017-12-07:
Interesting, that means that the deamon crashes if it has permissions to access a Logitech device with no per-key lighting.

I re-open your issue as this is severe enough for me to investigate more.
Thanks a lot for reporting.

m1st0, 2017-12-07:
I believe I had the issue due to logitech.rules not being included correctly and restarting udev.

m1st0, 2017-12-11:
This may also help shed more light on hiddraw1 vs hiddraw3 on my system:

    <I>service: device added: /dev/hidraw1
    <E>service: not opening device /dev/hidraw1: synchronization with device failed
    <I>service: device added: /dev/hidraw3
    <I>layout: loaded layout /usr/local/share//keyledsd/layouts/c33500000000_0001.xml
    <I>dev-manager: selected profile <__default__>
    <I>dev-manager: loaded plugin effect wave
    <I>dev-manager: loaded plugin effect fill
    <I>service: opened device /dev/hidraw3 [136036653435], model c33500000000 firmware U1v106.2.7, <G910 Orion Spectrum>

spectras, 2017-12-11:
Yes, most probably without logitech.rules you had to start it as root.
That means it had permission to access `hidraw2`, managed to talk to it somewhat and crashed.
The bug has been hidden because, as you saw, `logitech.rules` is very restrictive on purpose, so in the general case keyledsd never has permissions to open other devices.

Investigating further, it seems whatever device is at `hidraw2` is actually a logitech device too, and must have reported some block of keys with no key in it, a case I did not handle. Fix should be easy.

m1st0, 2017-12-11:
Interesting, and yes as you confirmed I have a Logitech mouse plugged in as well.  I noticed the order changhes of hiddraw1/2/3 switch as I moved the order of connections between the mouse and G910 keyboard.  Good catch.

commit 45a2f319 (2017-12-15): Do not attempt loading a list of keys for empty key blocks. Fixes #15.
closed 2017-12-15 by spectras