keyleds/keyleds #16: Lua Effect Not Found

m1st0, 2017-12-07:
Following instructions to create a test lua effect, I get the following:

    <I>effect-manager: effect hello not loaded, attempting auto-load
    <E>effect-manager: no module <fx_hello.so> in search paths
    <E>dev-manager: plugin for effect hello not found

Since it was a lua effect in $HOME/.local/keyledsd/effects/hello.lua, why did it mention fx_hello.so?  Also, my hello.lua matches that of the documentation.

spectras, 2017-12-07:
Hello @m1st0 and thanks for reporting!

It's normal that it looks for `fx_hello.so`: if it doesn't find the lua plugin, it attempts finding a binary plugin instead.

Did you check the lua plugin is properly loaded? It should show up at the very beginning of the log:

```
<I>main: keyledsd v0.7.0 starting up
<I>effect-manager: loaded plugin <lua>
```

If it doesn't show up, it means the configuration file does not load it (this plugin is not auto-loaded).
If it does show up, then it means keyledsd has trouble locating the file.

Could you find out which it is?

m1st0, 2017-12-07:
<W>main: loading module <lua>: no module <fx_lua.so> in search paths

spectras, 2017-12-07:
Okay, that's the issue.
Actually if you compiled it yourself and didn't have the required development package (`libluajit-5.1-dev`), it may not have been included at all. You should be able to check by locating the other plugins, eg: fx_stars.so

On my system effects get installed into `/usr/lib/x86_64-linux-gnu/keyledsd` but when building manually it may be another place, such as `/usr/local/lib/keyledsd` for instance.

Other option: if you still have the build directory you can check right in it whether you find `fx_lua.so`.

m1st0, 2017-12-07:
After `git checkout v7.1.0`  and `sudo apt install libluajit-5.1-dev liblua5.1-0-dev` it was successful.  Thank you for the following up with me.  Do you want me to add the dependencies on the apt line in documentation?

spectras, 2017-12-07:
I'm glad you could make it work. I hope you like it, and if you have feature suggestions, I'll be glad to read 
them.

I believe the documentation is not editable, I'll add it myself. And I have to install a VM with the latest ubuntu it seems to find what's wrong with the package. Previous version works fine.
Thanks a lot!

closed 2017-12-07 by m1st0