keyleds/keyleds #17: G910: Can't Control Logo, Bottom Light

m1st0, 2017-12-11:
While the logo and bottom g910 light do work in all effects, I cannot isolate them using groups or direct modification as in the fill effect using "logo" or "light" as the intended keys.  Some info on what my G910 Orion Spectrum (not Spark) is detected as:

    <I>service: device added: /dev/hidraw1
    <E>service: not opening device /dev/hidraw1: synchronization with device failed
    <I>service: device added: /dev/hidraw3
    <I>layout: loaded layout /usr/local/share//keyledsd/layouts/c33500000000_0001.xml
    <I>dev-manager: selected profile <__default__>
    <I>dev-manager: loaded plugin effect wave
    <I>dev-manager: loaded plugin effect fill
    <I>service: opened device /dev/hidraw3 [136036653435], model c33500000000 firmware U1v106.2.7, <G910 Orion Spectrum>

m1st0, 2017-12-11:
I also searched [g810-led/Keyboard.h](https://github.com/MatMoul/g810-led)'s project for logo in the logitech_g810_led/src/classes/Keyboard.h file and wondered if it would help with the G910 Spectrum?  Is it possible that logo needs a difference from G810 or fix to its numbering herein?

spectras, 2017-12-11:
Hello again!

Keyleds works in a very different way from g810-led, in that g810-led has raw dumps it injects into the device, while keyleds has a full implementation of the Logitech protocol. This is why logo and light work in effects : they are autodetected by querying the device.

Now, the device does not provide key names and positions, so keyleds ships with layout files providing those. It looks that's the missing piece here: I had a look at your keyboard layout, `c33500000000_0001.xml`, and it indeed lacks some special keys.
I believe multimedia keys should show incoherent colors when running `wave` effect, they are missing too.

Would you mind running those commands and posting the output?
```
keyledsctl -v info
keyledsctl get-leds -b logo
keyledsctl get-leds -b media
```
(and while unrelated to the issue, I'd love the output of this one: `sudo lsusb -vvd 046D:` as part of the effort to make Gkeys work at some point - check out #14 if you want to know more on that)

Thanks for dedicating the time to help me fix those glitches.

m1st0, 2017-12-11:
keyledsctl -v info

    Selecting device /dev/hidraw3                                                                                            
    Name:           G910 Orion Spectrum                                                                                      
    Type:           keyboard                                                                                                 
    Model:          c33500000000                                                                                             
    Serial:         32364713                                                                                                 
    Firmware[c335]: application U1  v106.2.7 [active]                                                                        
    Firmware[aac5]: bootloader BOT v114.0.7                                                                                  
    Features:       [0001, 0003, 4522, 0005, 1e00, 4540, 1eb0, 8010, 8020, 8030, 8060, 00c1, 1801, 1802, 8080, 8070, 1821]   
    Known features: feature version gamemode name reportrate leds led-effects                                                
    Report rates:   [1ms] 2ms 4ms 8ms                                                                                        
    LED block[01]:  104 keys, max_rgb(255, 255, 255)                                                                         
    LED block[02]:    5 keys, max_rgb(1, 0, 0)                                                                               
    LED block[04]:    9 keys, max_rgb(255, 255, 255)                                                                         
    LED block[10]:    2 keys, max_rgb(255, 255, 255)

m1st0, 2017-12-12:
keyledsctl get-leds -b logo

    x01=#4242ff
    x02=#0d0dff

There are other keys reported from get-keys that I wondered about:

    102ND=#000000
    COMPOSE=#3e0000
    RO=#000000
    KATAKANAHIRAGANA=#3e0000
    YEN=#000000
    HENKAN=#000000
    MUHENKAN=#000000

There are also those MACRO (M1, M2, M3, MR) keys which I think only have a on/off orange LED.

spectras, 2017-12-14:
I confirm the M1, M2, M3 and MR keys only have a 1/0 LED, which cannot even be controlled using the LED endpoint. It is an entirely separate on-board feature.
As for the other keys you wonder about, they are used on other keyboard layouts (102ND is present on FR layout for instance), but the firmware still handles them despite them not being actually there.

So that's the issue: G910 does not report some keys. Namely,
  * LOGO key block is missing entirely.
  * Multimedia keys are reported as ON/OFF only.

And though the keyboard reports the following properly, they miss an entry in layout file to give them a name:
  * Mode keys.

Could you try manually editing the layout file, and inserting this into it?
```xml
  <keyboard x="1019" y="20" width="177" height="35" zone="64">
    <row>
      <key glyph="Game" code="0x02" />
      <key style="blank" width="1.54" />
      <key glyph="Light" code="0x01" />
    </row>
  </keyboard>
  <keyboard x="1088" y="90" width="58" height="35" zone="2">
    <row>
      <key glyph="Mute" code="0xe2" />
    </row>
  </keyboard>
```

This should fix the Game, Light and Mute key. If it works fine, I'll add multimedia keys (their position should be off right now) and Logo keys (it needs code update as atm keyleds can fix missing keys but not missing blocks).

m1st0, 2017-12-14:
I added the XML and reloaded keyledsd.  To see if anything changed I used the following effects.  Nothing new appeared.  Keypad and modifiers from a previous effect still operational.

    - effect: fill 
      keypad: ff00ffff # Verified
      modifiers: ff0000ff # Verified
      special: ff0000ff # Verified
      Game: ffffffff # No changes
      Light: ffffffff # No changes
      Mute: ffffffff # No changes
      blank: ffffffff # No changes

Multimedia keys are always lit blue (play/pause, mute, next, previous).  I did notice after adding the XML that the 'previous' multimedia key now flashes off and on.

Also, how does keyledsd report the following info?  How did it connect to the current issue on my command line?  
    
    <I>service: setContext (id=60817424, title=G910: Can't Control Logo, Bottom Light · Issue #17 · spectras/keyleds · GitHub

spectras, 2017-12-14:
The fill effect does not recognize direct key names. They must be in a group. Can you try for instance:
```yaml
groups:
    modekeys: [game, light]
```
And
```yaml
effects:
  - effect: fill
    modekeys: red
```

**Also, how does keyledsd report the following info?**

It connects to X display and watches active window. It watches both when you change active window, and when currently active window changes its title. What you see in the context is the title of a window, probably your browser showing this issue.

If you are interested in specifics, relevant code is here: https://github.com/spectras/keyleds/blob/master/keyledsd/src/tools/XContextWatcher.cxx

m1st0, 2017-12-15:
"It connects to X display and watches active window", should have been something I saw instead of seeing it networking. Silly mistake watching log lines. Thank you.

spectras, 2017-12-16:
I believe I figured most of the issues now.

- media keys actually are just blue on/off on that keyboard, no RGB functionality. Keyleds ignores that, which is why you see one blinking. I don't really know what to do with those, perhaps skip them altogether since they are not RGB-capable?
- logo and badge are missing from layout file. Fixed in latest commit.
- game and light keys are not reported by the keyboard. Actually the whole block is amiss in reported data. That will require code changes, but I'm wondering whether those actually are RGB-capable?

m1st0, 2017-12-20:
Tried the following
 
   groups:
        modekeys: [game, light]

And used those keys in a specific effect that worked for other keys.  I didn't see the change work for now.

spectras, 2017-12-20:
It great to see Logo and Badge now work. So that was the issue, they were lacking a name assignment in the layout file.

I managed to get more info. Actually, it turns out Game and Light keys are not RGB-capable, and are not exposed through the RGB light interface. That's why `keyledsctl show` didn't report them.

It looks like the blob g810-led's dumps into the interface targets another feature than RGB lights, perhaps DFUControl2, which keyleds does not support. That's odd, it's normally used for firmware updates.
I don't believe in dumping chunks of binary data into a device I don't know perfectly, so I'll refrain from touching that interface until I get my hands on a G910 and can make some USB traces.

spectras, 2018-01-31:
As the relevant keys now work and the others have been shown not to be RGB-capable, I am closing this ticket.

closed 2018-01-31 by spectras