keyleds/keyleds #19: keyledsd doesn't find my keyboard

RoxasShadow, 2018-01-24:
Hello,

thanks for this project, it's very nice and useful since it seems that Logitech won't release their software on Linux.

The problem I'm occurring in is that although keyledsctl works perfectly, keyledsd doesn't look like to be able to connect to my keyboard. Sorry for the logs being in Italian, but basically they say that the permission to open `/dev/hidraw1` is denied, which makes sense since my keyboard is `/dev/hidraw3`, but unfortunately I don't know if it's possible to tell keyledsd which device it should connect to.

```sh
└[~/.config]> keyledsctl info
Name:           G910 Orion Spectrum
Type:           keyboard
Model:          c33500000000
Serial:         31374701
Firmware[c335]: application U1  v106.2.7 [active]
Firmware[aac5]: bootloader BOT v114.0.7
Features:       [0001, 0003, 4522, 0005, 1e00, 4540, 1eb0, 8010, 8020, 8030, 8060, 00c1, 1801, 1802, 8080, 8070, 1821]
Known features: feature version gamemode name reportrate leds led-effects
Report rates:   [1ms] 2ms 4ms 8ms
LED block[01]:  105 keys, max_rgb(255, 255, 255)
LED block[02]:    5 keys, max_rgb(1, 0, 0)
LED block[04]:    9 keys, max_rgb(255, 255, 255)
LED block[10]:    2 keys, max_rgb(255, 255, 255)

└[~/.config]> keyledsctl list
/dev/hidraw3 046d:c335 [017437623336]

└[~/.config]> keyledsd -c keyledsd.conf
<E>service: not opening device /dev/hidraw1: Permesso negato
keyledsd: /build/keyleds-fijYzj/keyleds-0.7.1.1/libkeyleds/src/feature_leds.c:100: keyleds_get_leds: asserzione "keys != NULL" non riuscita.
[1]    6889 abort (core dumped)  keyledsd -c keyledsd.conf
```

spectras, 2018-01-24:
Hello!

Thanks for reporting here. Well, it's normal and expected that it tries to open any device that looks like a logitech device, so actually the `/dev/hidraw1: Permesso negato` is not a problem. From the log, the problem arises after, probably while it tries to open /dev/hidraw3. To check this, you can enable verbose mode by adding a few `-v` on the command line (maximum is 3).

It seems related to #15, which occurred on G910 as well and should have been fixed on master, but as I did not publish a new release since, you still have it. I was supposed to get a G910 myself a few days ago, but the would-be seller did not show up so I still don't own one to do more thorough tests.

I guess I could publish those fixes as a v0.7.2.
I need to install a new VM with the latest ubuntu though, it seems the issue only triggered on that one, not on older versions. I'd be interested in knowing which distribution you use, by the way.

RoxasShadow, 2018-01-24:
Oh, I see - it makes sense then. Right now I'm running Ubuntu 17.10.

RoxasShadow, 2018-01-24:
Thank you @spectras and take care!

spectras, 2018-01-26:
So I had a look and it definitely looks like the bug that was fixed by 45a2f319588c58d056b90f931798dad58031b71f, so it should be good as soon as this is released. It will also come with the missing entries in G910 layout files that prevented some keys from lighting up correctly.

However, project architecture was reworked since last release and it broke packaging it seems. I am reinstalling my VMs right now, expect the release tomorrow :)

spectras, 2018-01-27:
I am struggling with SUSE/Fedora packages, but ubuntu one should be ready if you can give it a try.

spectras, 2018-01-28:
Version 0.7.4 should include the fix :)

RoxasShadow, 2018-01-31:
Yup, works nice now!

Only thing to report is that now I get `<E>service: not opening device /dev/hidraw1: feature not found on device` but the message looks quite self-explanatory. Thank you!

closed 2018-01-31 by RoxasShadow

spectras, 2018-01-31:
Great. That would mean you have another Logitech device, which is not a per-key RGB keyboard.
As it's still from Logitech, speaking the same language, it reaches the feature enumeration step and then realizes that device won't do.

That shouldn't have any incidence on your G910. I hope it now works fine, but feel free to open other tickets should you have any issue, feature request or general feedback.