keyleds/keyleds #21: Shared Runtime Libraries Error

m1st0, 2018-06-22:
If I don't have the following workaround as a symbolic link, I get the error

> keyledsd: error while loading shared libraries: keyleds_common.so.0: cannot open shared object file: No such file or directory

While I have repeatedly attempted to load shared libraries by running `sudo ldconfig` .  Within `/etc/ld.so.conf.d` there is a file referencing `/usr/local/lib` to allow for libraries I install.

I repeatedly cannot run keyledsd without making a symbolic link `keyleds_common.so.0 -> /usr/local/lib/keyleds_common.so.0` regardless of trying to fix runtime linked bindings.

m1st0, 2018-06-23:
Issue resolved.  Unsure of ldconfig commands that helped. Adding variable in systemd launch may have helped instead (I statically referenced the _conf_ file if someone uses another location):

> Environment="LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib"
ExecStart=/usr/local/bin/keyledsd -c /etc/keyledsd.conf

spectras, 2018-07-06:
Hello,
Sorry I did not answer earlier. I'm glad you could solve the issue by yourself.

Those lib path settings are tricky to get right, they need to work:
* While running directly from the build directory.
* After a manual build and install
* In packaged builds, including on split arch installs (the standard in x86-64 debian-based distros)

If you have the issue again, please re-open it. Thanks for taking the time to report here :)

closed 2018-07-06 by spectras