keyleds/keyleds #22: GPRO problems

deathtrip, 2018-07-20:
In the AUR package description the GPRO was listed as supported, so i tried it out and here's some problems i encountered:
- when trying the wave plugin i get this : floating point exception (core dumped) 
- the breathe plugin makes the whole keyboard flash different colors rapidly

The nightsky effect example from the documentation works, as does keyledsctl. 
So i think we have only partial GPRO support, would be nice to see it up to par with the other keyboards.

spectras, 2018-07-20:
Hello,

Thanks for reporting! Support for gpro should be complete, it sounds like you found a bug.
The floating point exception in particular is in the effect itself, as the hardware module has no floating point at all. Interesting that it would only show up on gpro.

Would you be able to copy-paste the output of `keyledsctl info`?

deathtrip, 2018-07-24:
Here it is:
> sudo keyledsctl info                                                                                                                                                            
Name:           Pro Gaming Keyboard
Type:           keyboard
Model:          c33900000000
Firmware[c339]: application U1  v113.1.13 [active]
Firmware[aad0]: bootloader BOT v114.0.c4b1
Features:       [0001, 0003, 4522, 0005, 1e00, 4540, 1eb0, 8060, 00c2, 1801, 1802, 8080, 8070, 1821, 1bc0, 1806, 40a2]
Known features: feature version gamemode name layout2 reportrate leds led-effects
Report rates:   [1ms] 2ms 4ms 8ms
LED block[01]:  104 keys, max_rgb(255, 255, 255)
LED block[10]:    1 keys, max_rgb(255, 255, 255)
LED block[40]:    4 keys, max_rgb(255, 255, 255)

This is the log from running keyleds with the default config, it works with KDE's konsole, but when i switch to the chromium window it crashes:
> main: keyledsd v0.7.5 starting up
<I>main: using /etc/keyledsd.conf
<I>effect-manager: loading lua from /usr/lib/keyledsd/fx_lua.so
<I>effect-manager: loaded plugin <lua>
<I>service: setContext ()
<D>filewatcher: subscribed to events on /etc/keyledsd.conf => 1
<D>service: created
<I>service: connected to display :0
<D>xwindow: subscribed to events on :0 => 1
<D>xwindow: subscribed to events on :0 => 2
<I>xinput-watcher: xinput keyboard 6 enabled for device /dev/input/event1
<I>xinput-watcher: xinput keyboard 7 enabled for device /dev/input/event0
<I>xinput-watcher: xinput keyboard 8 enabled for device /dev/input/event4
<I>xinput-watcher: xinput keyboard 9 enabled for device /dev/input/event5
<I>xinput-watcher: xinput keyboard 12 enabled for device /dev/input/event3
<I>service: setContext (id=58720263, title=log : zsh — Konsole, class=konsole, instance=konsole)
<I>service: switching to active mode
<I>service: device added: /dev/hidraw3
<D>filewatcher: subscribed to events on /dev/hidraw3 => 2
<D>dev-manager: profile matches: konsole
<I>dev-manager: selected profile <konsole>
<I>effect-manager: effect fill not loaded, attempting auto-load
<I>effect-manager: loading fill from /usr/lib/keyledsd/fx_fill.so
<I>effect-manager: loaded plugin <fill>
<I>dev-manager: loaded plugin effect fill
<I>effect-manager: effect breathe not loaded, attempting auto-load
<I>effect-manager: loading breathe from /usr/lib/keyledsd/fx_breathe.so
<I>effect-manager: loaded plugin <breathe>
<D>effect-service: created RenderTarget(0x35b3ae92a00)
<I>dev-manager: loaded plugin effect breathe
<I>dev-manager: loaded plugin effect fill
<D>effect-service: created RenderTarget(0x35b3af021d0)
<I>dev-manager: loaded plugin effect idle-dim
<I>dev-manager: loaded plugin effect reactive-hlines
<I>effect-manager: effect feedback not loaded, attempting auto-load
<I>effect-manager: loading feedback from /usr/lib/keyledsd/fx_feedback.so
<I>effect-manager: loaded plugin <feedback>
<D>effect-service: created RenderTarget(0x35b3ae944a0)
<I>dev-manager: loaded plugin effect feedback
<D>dev-manager: enabling 6 effects for loop 0x35b3aec0158
<I>service: opened device /dev/hidraw3 [147036563435], model c33900000000 firmware U1v113.1.13, <Pro Gaming Keyboard>
<I>service: device added: /dev/hidraw1
<D>anim-loop: AnimationLoop(0x35b3aec0158) started
<E>service: not opening device /dev/hidraw1: feature not found on device
<D>xinput-watcher: key 28 released on device 8
<D>dev-manager: key ENTER released on device 147036563435
<I>service: setContext (id=48234497, title=New Tab - Chromium, class=Chromium, instance=chromium)
<I>dev-manager: selected profile <__default__>
<I>effect-manager: effect wave not loaded, attempting auto-load
<I>effect-manager: loading wave from /usr/lib/keyledsd/fx_wave.so
<I>effect-manager: loaded plugin <wave>
<D>effect-service: created RenderTarget(0x35b3af00e50)
zsh: floating point exception  sudo keyledsd -vvv

Here's what happens when i delete other profiles from the config file and leave rainbow as the default effect:

> <I>main: keyledsd v0.7.5 starting up
<I>main: using /etc/keyledsd.conf
<I>effect-manager: loading lua from /usr/lib/keyledsd/fx_lua.so
<I>effect-manager: loaded plugin <lua>
<I>service: setContext ()
<D>filewatcher: subscribed to events on /etc/keyledsd.conf => 1
<D>service: created
<I>service: connected to display :0
<D>xwindow: subscribed to events on :0 => 1
<D>xwindow: subscribed to events on :0 => 2
<I>xinput-watcher: xinput keyboard 6 enabled for device /dev/input/event1
<I>xinput-watcher: xinput keyboard 7 enabled for device /dev/input/event0
<I>xinput-watcher: xinput keyboard 8 enabled for device /dev/input/event4
<I>xinput-watcher: xinput keyboard 9 enabled for device /dev/input/event5
<I>xinput-watcher: xinput keyboard 12 enabled for device /dev/input/event3
<I>service: setContext (id=58720263, title=log : zsh — Konsole, class=konsole, instance=konsole)
<I>service: switching to active mode
<I>service: device added: /dev/hidraw3
<D>filewatcher: subscribed to events on /dev/hidraw3 => 2
<I>dev-manager: selected profile <__default__>
<I>effect-manager: effect wave not loaded, attempting auto-load
<I>effect-manager: loading wave from /usr/lib/keyledsd/fx_wave.so
<I>effect-manager: loaded plugin <wave>
<D>effect-service: created RenderTarget(0x9d4e56a3220)
zsh: floating point exception  sudo keyledsd -vvv

It crashes immediately, and so does setting alert or 'keyleds-default' as the default effect. Basically any profile with wave or breathe effects crash.

spectras, 2018-07-24:
Thanks alot.
So, there are two issues at work here:

- The layout file for your gpro is missing. Not a big issue, that just means keyledsd has no idea how the keys are physically arranged, so effects using that information won't work (engine will report size 0 for all keys to plugins).
- The affected plugins assume key size cannot be 0. That makes sense for an actual key, but is causing the crash here.

I'll see to fix this in the next few days.

spectras, 2018-07-25:
Fix pushed to `master` branch. If you can give it a go, that's great, otherwise I'll package a release with the fix in it within the week.

spectras, 2018-07-28:
Version 0.7.6 should fix this issue. Could you please upgrade and confirm?

deathtrip, 2018-07-30:
With the new version it works when i launch from the terminal, but when i reboot i still get the flashing. Except this time the only way to stop it is to turn off the lighting. Previously i could switch to one of the profiles stored in the keyboard.

spectras, 2018-07-30:
It would seem when you reboot it somehow starts the old version instead of the new one.
To stop the flashing, `killall keyledsd` should work, if it flashes it means the service is still in control.

Would it be possible to post the logs to confirm?
To enable (some) debugging output you can edit `/usr/share/keyledsd/keyledsd.desktop` and add `-vv` to the `Exec` line: `Exec=keyledsd -vv`.

This will cause the service to log stuff, and that logging output should end up in `~/.xsession-errors`.

deathtrip, 2018-08-03:
Did a clrean install and still the same result. I did stop the flashing by terminating the service though.
My keyboard has the us-int layout, but keyleds sees it as the eu layout. As the white wave effect triggers on the enter key when the wave is triggered one layer above the key, beacuse on the eu layout the enter occupies two rows. Meanwhile the " \ " key, which is on the place where on the eu layout the upper row part of the enter key is, isn't being triggered by this effect at all, no matter which row. Best check the keyboard pictures to see exactly what i mean.

spectras, 2018-08-04:
Do you now have the following line appearing when you run `keyledsd -vv`?

```
<I>layout: loaded layout /usr/share/keyledsd/layouts/c33000000000_0005.xml
```
Could you copy it here so I can check the layout it loads?

deathtrip, 2018-08-05:
Here it is:
`<I>layout: loaded layout /usr/share//keyledsd/layouts/c33900000000_0001.xml`

spectras, 2018-08-05:
Okay, that's actually the US layout, so it's detecting it just fine.

* **Horizontal lines**
  The horizontal lines that appear as a reaction to your keypresses are a demo script, which does not use 
  keyboard layout but a hard-coded sequence. You may copy it to `~/.local/share/keyledsd/effects/reactive-hline.lua` and edit the sequence there if you wish (or edit the file in place, but beware that it will be overwritten at next install).

  → On line 10, change `"ENTER"` to `"BACKSLASH"`. That should fix this effect.

* **About the divide by zero crash**
  Can you confirm it no longer happens?

So my understanding now is as follows:
* Keyleds now properly loads the layout for your keyboard.
* When you manually start keyledsd all works correctly, including wave and breathe effect. Is that correct?
* When it starts automatically (from boot that means?), then
    - wave effect goes all flashy. Is that correct?
    - breathe effect… what's its status?

Thanks for your time :)

deathtrip, 2018-08-07:
Yes, when starting manually, all effects configured in the default profiles work. So i think we can say all effects work. 

When starting from boot, it flashes regardless of the active window, which means no effect works, as different profiles would be loaded.

The floating point crash is gone.

spectras, 2018-08-10:
My guess would be for some reason when starting from boot it doesn't find the layout. And it doesn't connect to X session.
* Without connecting to X session, it won't react to active window changes and, in the supplied config file, it defaults to wave effect when active window is unknown.
* The wave effect disables itself if it has no layout information. This result in the rendering buffer keeping whatever content it has. As keyledsd uses double-buffering, what you see is the result of swapping between the two uninitialized buffers.

I wonder how keyleds gets started at boot. The intended setup is to have it be started after login, from within the graphical environment. This is normally accomplished by linking the desktop file into `/etc/xdg/autostart`. Makepkg should do it this way.

You can test whether keyleds is started that way by removing `/etc/xdg/autostart/keyledsd.desktop`. Keyleds should no longer start at boot. If it still does, then something else is starting it probably before the system has required components ready.

deathtrip, 2018-08-14:
When i removed `/etc/xdg/autostart/keyledsd.desktop` keyleds didnt start.

It turns out that it flashes everytime keyleds is run as an unprivileged user, which happens at boot.
And i was running as root manually, that's why it worked there.

However regardless if i run as root or normal user i get this error when starting:
`service: not opening device /dev/hidraw1: feature not found on device`

It doesn't seem to be related to the keyboard model, so i'm wondering why only i encountered this problem.

spectras, 2018-08-14:
This is normal, it just means you have another Logitech device with no RGB leds on it.

> It turns out that it flashes everytime keyleds is run as an unprivileged user, which happens at boot.

This is interesting. So back to the previous question: can you start is manually, as unprivileged user (causing the flashing), with logging, and look for that line again?
```
<I>layout: loaded layout /usr/share//keyledsd/layouts/c33900000000_0001.xml
```
And possible, as that unprivileged user, check you have the permissions to open the file. It could be as simple as the file being unreadable by the user.

deathtrip, 2018-08-16:
I checked the file permissions for the layout and the effect files in /usr/lib/keyledsd, and they are all readable by all users, however `/dev/hidraw3` cannot be accessed by normal users.
I also noticed that when running with sudo it uses `/etc/keyledsd.conf` ,while as a normal user it's `~/.config/keyledsd.conf`. Will check the config files when i get to my other machine.

spectras, 2018-08-17:
Indeed, keyledsd follows the [XDG basedir specification](https://standards.freedesktop.org/basedir-spec/basedir-spec-latest.html). Thus, it will search the following locations:
* Your `$XDG_CONFIG_HOME` directory. If not set, it resolves to `$HOME/.config/keyledsd.conf`.
* Each directory defined in `$XDG_CONFIG_DIRS`, in order.
* `/etc/keyledsd.conf`

It uses the first it finds. This means by default it will used the system-wide one, but users can override it by simply putting theirs in `~/.config` as you noticed.

deathtrip, 2018-08-28:
I just tried to fix the Horizontal lines bug and copied and edited reactive-hline.lua just like you said above. And while now the enter key works as it should, backslash never activates as part of the line. Even If i press it directly i get the horizontal line but the key itself never lights up, and neither did it before editing the file.

Also setting KEYLEDS_DEVICE env variable, shouldn't it stop the `feature not found on device` error?
Because i still get it after setting the variable.

deathtrip, 2018-08-30:
Yes, it works when part of the wave effect, or when i tell keyledsctl to change the color on all keys.
`keyledsctl get-leds` always reports: BACKSLASH=#000000, regardless of the real color.

When i tell keyledsctl to change the color on that particular key nothing happens, and i don't get the `keyledsctl: invalid key in directive` error, which is weird.

Also `keyledsctl set-leds all=somecolor` doesn't set color on the logo and all the modes keys, not sure if it's meant to be that way, or it's a bug.

spectras, 2018-08-31:
> Also keyledsctl set-leds all=somecolor doesn't set color on the logo and all the modes keys, not sure if it's meant to be that way, or it's a bug.

Meant that way. `keyledsctl` is a low-level tool. It manipulates a single key block. Try `keyledsctl set-leds -b modes all=red` ☺

> `keyledsctl get-leds` always reports: BACKSLASH=#000000

Good catch. That means the keyboard doesn't publish that key as BACKSLASH.
This is also why nothing happens when you change the color. The BACKLASH entry in the light table embedded in the keyboard must be a "phantom" key. Indeed, the light table has more entries than actually exist - setting those unused entries is valid but has no effect.

We need to find which key is the right one. This small script should flash all keys in the `keys` block:
```bash
for i in {001..105}; do echo $i; keyledsctl set-leds all=black $i=white; sleep 1; done
```
When the backslash keys turns on, kill the script, so that the key remains lit.

At this point, use `keyledsctl get-leds` and look for the only key that has a non-zero color. That will be the one.

deathtrip, 2018-08-31:
I ran the script and the backslash key didn't light up, on some numbers no key lit up, and on some all the keys.
Running keyleds -vvv reports backslash as the key number 43, trying to change the color of this key using keyledsctl does nothing.

It looks like because there are two layouts of this keyboard, the EU layout backslash key is seen as a seperate key from the INTL layout, because they are placed differently. The international layout key name seems to be "intl_backslash", as reported here:
https://github.com/MatMoul/g810-led/issues/16#issuecomment-263060588

Will check it out once i get back to my other machine.

spectras, 2018-08-31:
It might be in the upper part then…
Could you try the same script, only changing `105` for  `255` to cover the whole range?

deathtrip, 2018-09-02:
Tried it up to 255, and still nothing.
I think the problem could be with the layout file, is the one loaded for the eu or intl version?
Because keyleds can recognise the input of the key, and we know there can be two different names for the backslash key.

spectras, 2018-09-02:
`keyledtsctl` is a low-level tool, it does not use the layout files, nor is it aware of layouts (`set-leds A=blue` will light the Q key on AZERTY keyboards). It sends raw control codes to the LED controller in the keyboard.

But the fact none works, yet keyleds service can use it in wave effect is a very good track, because the way they use the keyboard does not differ much. I should be able to find an explanation by checking that. I'll check it out. Thanks for your patience. I wish I had a gpro myself so I wouldn't have to make you test the things yourself.

spectras, 2018-09-02:
Okay, I think I got something.

In the low-level driver of keyboards on Linux, the key translation is done by this table:
https://github.com/torvalds/linux/blob/master/drivers/hid/usbhid/usbkbd.c#L51

Interestingly, it maps not one but **two** keys to `KEY_BACKSLASH`.
This means there is no way to distinguish between them. I believe that's what is happening there.

→ Since the line shows, it means it receives the `BACKSLASH` event from the input.
→ Except your keyboard doesn't really have the standard BACKLASH key (0x31), but the HASHTILDE key (0x32), that Linux kernel remaps to 0x31.
→ Therefore when keyleds tries to find light number 0x31, it cannot find it.

This remapping is 100% silent. I wonder how to fix that. Probably I'll have to find a way to put it in the layout file somehow.

spectras, 2018-09-14:
Time got very scarce lately, but I expect to be able to solve the issue this weekend :)
Sorry for taking so long to reply.

spectras, 2018-09-16:
I think the backslash workaround should work. It is pushed onto master branch, not yet released.

As for keys with wrong positions:
* Can you confirm your keyboard looks like this?<br>https://www.logitechg.com/content/dam/gaming/en/products/pro-kbd/pro-keyboard-hero-feature-1-desktop-1.png.imgw.3600.3600.png
* Scrollock is probably incorrectly tagged in the layout if it's left aligned.<br>It would be amazing if you could run the led-blinking script and see whether it lights the Scrollock key and find out its name.
* Capslock is on the left, isn't it correct to be on the left? Or perhaps I did not get the key you meant right?

Or perhaps you meant the tiny leds, not keys?

deathtrip, 2018-09-17:
Yes, i meant the tiny leds above the F keys, the keys itself work fine.

spectras, 2018-09-18:
I did not even know they were controllable through the RGB interface.
From the picture they dodn't look RGB-capable?

From your logs earlier I see this:
```
LED block[40]: 4 keys, max_rgb(255, 255, 255)
```
As you don't have 4 custom keys but only two (game mode and lights), my best guess is the two others could be the leds above the F keys. Could you try this:

To get the codes used by those keys: `keyledsctl get-leds -b modes`
To change colors manually: `keyledsctl set-leds -b modes x01=white` (using the codes from above command).
*You should disable `keyledsd` first, otherwise the color you set will be overridden by the service immediately.*

If it works, I will add them to the layout file and it should be all that's needed. Thanks for your time :)

deathtrip, 2018-09-20:
Yes, changing colors works. The key codes are:
x01 is the lights key
x02 is game mode
x03 is caps lock
x04 is scroll lock

spectras, 2018-09-25:
Could you try adding this to `/usr/share/keyledsd/layouts/c33900000000_0001.xml`:
```xml
  <keyboard x="648" y="23" width="95" height="28" zone="64">
    <row>
        <key glyph="CapsLockLed" code="0x03" />
        <key glyph="ScrollLockLed" code="0x04" />
    </row>
  </keyboard>
```
Insert this alongside the other `<keyboard>` tags, for instance right after the `<spurious…/>` line.

If I am correct, this should tell keyleds the physical location of the leds, so position-based effects will work. If you use the Lua engine to create custom effects, the leds will be named `CapslockLed` and `ScrollLockLed`.

If it works, I will update the layout files for the GPro and publish a new release. Thanks for taking the time to pioneer GPro support :)

deathtrip, 2018-09-29:
I added it and everything works as expected. It's time for a new release i guess :)

spectras, 2018-11-17:
V 0.7.9 includes everything from that issue. Thanks again for your time deathtrip.

closed 2018-11-17 by spectras