keyleds/keyleds #23: Permissions hidraw and udev

m1st0, 2018-08-06:
Using Ubuntu 18.04.1, I cannot provide access using the logitech.rules file to allow keyledsd to work with the keyboard without additional access such as running with `sudo`.  However, I have found a workaround that helps with other USB device access on my system.  To begin with an example:
    
    SUBSYSTEMS=="usb", ATTRS{manufacturer}=="Logitech", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c335", MODE="0660", GROUP="plugdev"

Allows me to access devices on my system as a member of the group. In addition, this will set the correct permissions in `/dev/bus/usb/001/` with `003` referencing the G910 attached to my system:

    total 0
    crw-rw-r-- 1 root root    189, 0 Aug  6 09:17 001
    crw-rw-r-- 1 root root    189, 1 Aug  6 09:17 002
    crw-rw---- 1 root plugdev 189, 2 Aug  6 09:46 003
    crw-rw-r-- 1 root root    189, 3 Aug  6 09:17 004
    crw-rw-r-- 1 root root    189, 4 Aug  6 09:17 005
    crw-rw-r-- 1 root root    189, 5 Aug  6 09:17 006
    crw-rw-r-- 1 root root    189, 6 Aug  6 09:17 007
    crw-rw-r-- 1 root root    189, 7 Aug  6 09:17 008
    crw-rw-r-- 1 root root    189, 8 Aug  6 09:18 009

My quest is to see whether we can update code to go over `/dev/bus/usb/001/` instead of parsing permissions on `/dev/hidraw*` to access the keyboard?  Updating logitech.rules, line 9, from 1 to 2 did not work on hidraw:
1. ATTRS{bInterfaceProtocol}=="00", TAG+="uaccess"
2. ATTRS{bInterfaceProtocol}=="00", MODE="0660", GROUP="plugdev", TAG+="uaccess"
The hidraw device nodes are yet in either case:

    crw------- 1 root root 243, 0 Aug  6 09:17 /dev/hidraw0
    crw------- 1 root root 243, 1 Aug  6 09:17 /dev/hidraw1
    crw------- 1 root root 243, 2 Aug  6 09:18 /dev/hidraw2
    crw------- 1 root root 243, 3 Aug  6 09:18 /dev/hidraw3

spectras, 2018-08-07:
Hello,

I am running Ubuntu 18.04 as my main development system, and have no issue.

Keyledsd explicitly uses `uaccess` directive, which causes systemd to update ACLs (access control list) on devices, so multiple users might use the device concurrently when switching sessions. Keyledsd supports that protocol, and will disable itself an re-enable itself whenever systemd detects a user switch.

For your suggestion, I see two issues:
* `keyledsd` needs access to the actual hid interface. The usb device you showed cannot be used to talk to the required endpoint. You can see from the device majors that they are not the same (243 for hid devices vs 189 for usb devices on your system).
* The `plugdev` group is the old way to handle local devices. It is now obsolete and is being slowly removed from most distribs. In fact, Debian now considers it an [error](https://lintian.debian.org/tags/udev-rule-missing-uaccess.html) not to use uaccess for device rules.

It would probably be better to understand why `uaccess` fails to grant correct permissions on your system. Do you know:
* whether ACL support is enabled? It is by default on most distribs, but better check.
* whether systemd is in control of devices? Same thing, that's the default setting on most distribs but better check.

When it works properly, it should appear like this (notice the small `+`):
```
crw-------  1 root root 247, 0 Jul 29 12:30 /dev/hidraw0
crw-rw----+ 1 root root 247, 1 Jul 29 12:30 /dev/hidraw1
crw-------  1 root root 247, 2 Jul 29 12:30 /dev/hidraw2
```

m1st0, 2018-08-08:
Thanks for the knowledge.  There was an article on mode and group permissions which now is out of date but seemed at the time to say that it was the way to go with udev.

After confirming my filesystem has ACL support, I was able to obtain the result you noted with the following modification that someone else also had found with their Udev [issue](https://github.com/Yubico/libu2f-host/issues/28):

    ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c335", TAG+="uaccess"
    
The following also didn't independently work for interest due to specificity of hidraw items or even usb subsystems respectively.  I also tried of course the current _logitech.rules_ in the repo.  Only the above line worked for me.  I get that any of the lines below are not correct in terms of obtaining the device's correct tree from `sudo udevadm info -a /dev/hidraw0` or `sudo udevadm info -a /dev/hidraw1`, but I had to respectively see why logitech.rules was not working specifically enough.

- `KERNEL=="hidraw*", SUBSYSTEMS=="hidraw",  ATTRS{manufacturer}=="Logitech", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c335", TAG+="uaccess"'`
- `SUBSYSTEMS=="usb", ATTRS{manufacturer}=="Logitech", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c335", TAG+="uaccess"`

Do you want me to update logitech.rules for Ubuntu 18.04.1 in a branch?

spectras, 2018-08-10:
Thanks a lot for sharing the result of your investigation.
I wonder what would be best here. Your approach will definitely work for your device, but has two major caveats:
* Hardcoding the product id might be workable, but then it will be a pain to maintain, as Logitech likes to change the product ids when one upgrades the firmware.
* I would like to keep the restriction that access is only granted to the device with `bInterfaceProtocol==0`. The point is the keyboard exposes two interfaces. One allows bypassing X security and reading keypresses directly from the device, I'd rather not be the one unlocking that box. The one we use is somewhat safe, it allows changing some config and colors, but that's it (though I guess someone crafty enough could use it to flash an infected firmware, but that significantly raises the challenge).

In your examples at the end of your post, I see the following things that could make them fail:
* For first example, I believe it should be `SUBSYSTEM` (singular) for this level. Yeah that's tricky, udev uses sometimes singular and sometimes plural, for some reason.
* For second example I don't really know. It looks like it should work, but perhaps it only grants access to the root of the device and not to the hidraw subsystem itself?

spectras, 2019-03-03:
Hello,
I am closing stall issues. If this is still causing trouble in the latest releases, please feel free to re-open this or open a new one referencing it.

closed 2019-03-03 by spectras