keyleds/keyleds #24: G810 on Ubuntu 18.04 - std::logic_error with no parent for device type

foehnph, 2018-08-28:
Hi Devs

Your library looks pretty cool, I'd like to use it on my Ubuntu 18 system (Dell XPS13).
I installed it via apt repo and upon launching I get the following error:
(tested USB 3.0, USB 2.0 and USB ports via dock, with added udev rules and both as normal user and root/sudo)

```
terminate called after throwing an instance of 'std::logic_error'
  what():  No parent with specified type for device /sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DLL075B:01/0018:06CB:76AF.0007/hidraw/hidraw3
Aborted
```

The `keyledsctl info` outputs the following infomration summary:
```
Name:           G810 Orion Spectrum
Type:           keyboard
Model:          c33100000000
Serial:         31354711
Firmware[c331]: application U1  v102.0.14 [active]
Firmware[aabd]: bootloader BOT v114.0.8
Features:       [0001, 0003, 4522, 0005, 1e00, 4540, 1eb0, 8060, 00c1, 1801, 1802, 8080, 8070, 1821]
Known features: feature version gamemode name layout2 reportrate dfu-control leds led-effects
Report rates:   [1ms] 2ms 4ms 8ms
LED block[01]:  105 keys, max_rgb(255, 255, 255)
LED block[02]:    5 keys, max_rgb(255, 255, 255)
LED block[10]:    1 keys, max_rgb(255, 255, 255)
LED block[40]:    5 keys, max_rgb(255, 255, 255)
```

and `keyledsd -vv` results in
```
<I>service: setContext ()
<I>service: connected to display :0
<I>xinput-watcher: xinput keyboard 6 enabled for device /dev/input/event3
<I>xinput-watcher: xinput keyboard 7 enabled for device /dev/input/event5
<I>xinput-watcher: xinput keyboard 8 enabled for device /dev/input/event1
<I>xinput-watcher: xinput keyboard 9 enabled for device /dev/input/event2
<I>xinput-watcher: xinput keyboard 11 enabled for device /dev/input/event14
<I>xinput-watcher: xinput keyboard 16 enabled for device /dev/input/event6
<I>xinput-watcher: xinput keyboard 17 enabled for device /dev/input/event8
<I>xinput-watcher: xinput keyboard 18 enabled for device /dev/input/event10
<I>xinput-watcher: xinput keyboard 19 enabled for device /dev/input/event9
<I>xinput-watcher: xinput keyboard 20 enabled for device /dev/input/event12
<I>xinput-watcher: xinput keyboard 21 enabled for device /dev/input/event4
<I>xinput-watcher: xinput keyboard 23 enabled for device /dev/input/event16
<I>xinput-watcher: xinput keyboard 24 enabled for device /dev/input/event15
<I>service: setContext (id=67108867, title=Drop Down Terminal, class=DropDownTerminalWindow, instance=Drop Down Terminal)
<I>service: switching to active mode
terminate called after throwing an instance of 'std::logic_error'
  what():  No parent with specified type for device /sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DLL075B:01/0018:06CB:76AF.0007/hidraw/hidraw3
Aborted
```

Do you know what could cause this error or how I can hunt it down?

foehnph, 2018-08-29:
thanks for the reply.
Well, I'm not very good with low-level-debugging but from what I can get out of my notebook, it has 2 additional `hidraw` devices, connected by i2c. could be my touchpad and touchscreen.
When I use `lspci` and check id `0000:00:15.1` (and `0000:00:15.0`) I see that it is
```
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
	Subsystem: Dell Sunrise Point-LP Serial IO I2C Controller
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at dc433000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel driver in use: intel-lpss
	Kernel modules: intel_lpss_pci

00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
	Subsystem: Dell Sunrise Point-LP Serial IO I2C Controller
	Flags: bus master, fast devsel, latency 0, IRQ 17
	Memory at dc432000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel driver in use: intel-lpss
	Kernel modules: intel_lpss_pci
```
No clue if this helps you in any way... seems to be a general way that touchpads and touchscreens are interfaced.
Googling arround I could find other people having problems with 3rd-party HID drivers and those devices (many reporting logitech joysticks or general gaming devices causing problems/interference).

I would be very grateful if you could add a check for that and skip the devices not connected by USB...

Thanks for your time! Cheerio

spectras, 2018-08-29:
Thanks for taking the time to confirm my initial analysis.
Yes indeed, as almost all HID devices are USB, it is easy to forget there also are non-usb HID devices.

I pushed a commit that should fix the issue. Would you feel comfortable building from source to test the fix, or would you rather me to package it?

foehnph, 2018-08-29:
Thanks a lot!!
I already built it and it seems to fix the problem. 
On my system I had to add your install path `/usr/local/lib/` for the shared libraries to my library search path and rerun `ldconfig` (don't know if my config is weird or if you assume a non-generic configuration).

Also I just added the example `keyleds.conf` file you provide and with this the whole keyboard is flashing in colors... but I guess that's because many effects are loaded/applied.... I will look into configuring it appropriately.

Thanks for the fast and awesome support!

spectras, 2018-08-29:
I'm glad this fixes the issue. I will publish a bugfix release with it in it.

And this will put plugins and data files in the locations ubuntu expects. When building from source, the default is to put things in a distribution-independent place, `/usr/local`. But indeed it's distribution-independent because it's non-standard in all of them. This is a common practice, it ensures manual build won't touch distribution files.

foehnph, 2018-08-29:
ok, my bad with the directory.
thanks again! I really appreciate devs that provide such good support to the community!
Cheers

closed 2018-08-29 by foehnph

spectras, 2018-08-29:
Thanks for the feedback, it is much appreciated.

I published the bugfix for Ubuntu.
I recommend you remove the manually-installed version. That should be:
* rm `/usr/local/bin/keyleds*`
* rm -r `/usr/local/lib/*keyleds*`
* rm -r `/usr/local/share/keyledsd`
* rm `/usr/local/share/man/man1/keyleds*`

Then the usual `apt-get update && apt-get install keyleds` should get you release 0.7.7 with the fix.

foehnph, 2018-08-29:
:+1: will do