keyleds/keyleds #26: how to set key blocks in config file?

V3XATI0N, 2018-11-15:
How do I set the color for key blocks in the config file? I can't find anywhere in the documentation that even mentions them except for `keyledsctl`. Also the `-b` switch only kind of works on by G810. I can do `set-leds -b logo 1=red`, and that works, but `set-leds -b media [1-5]=red` does nothing (for any value 1 through 5).

spectras, 2018-11-16:
Hello,

`keyledsd` has not notion of key blocks (well it works with them internally but it is not exposed).
Media keys are exposed as any other key.

The most practical way to know which key is which is to start keyledsd in very-verbose mode:
```bash
killall keyledsd
keyledsd -vvv
```
It will then show detected keys in this way:
```
<D>dev-manager: key LCTRL pressed on device XXXXXXXXXXXX
```
The key name here is `LCTRL`. This should work with all keys. If it does not, then you have found a bug.

If you want to make a group of such keys, simply use the [usual syntax](https://github.com/spectras/keyleds/wiki/Key-Group), eg:
```yaml
groups:
    special: [game, light]
    functions: [F1, F2, F3, F4, F5, F6, F7, F8, F9, F10, F11, F12]
```

spectras, 2018-11-16:
Alright. Well, for indicators they should be named NumLock CapsLock and ScrollLock.

For the logo, looked it up straight from the layout file, only to realize I forgot to give it a name on G810. It is there, but has no name you can use in the description file of your device.

I'll add it and publish an update in the weekend.
You can fix this manually in the meantime if you want:

- open the layout file of your device (`keyledsd -v` will tell you its name, like `<I>layout: loaded layout /usr/share//keyledsd/layouts/c33000000000_0005.xml`).
- search for a line that says `<key code="0x01" style="logo" />`. Should be close to the top, line 6 or 7.
- add the name: `<key glyph="Logo" code="0x01" style="logo" />`.

This should make the logo available as `logo` in the config file.

V3XATI0N, 2018-11-16:
NumLock, CapsLock, and ScrollLock control those respective keys, not the mode indicators. Thanks for the logo fix, tho!

V3XATI0N, 2018-11-16:
nevermind, I fixed it by renaming those in the layout file to NumLockMode/ScrollLockMode/CapsLockMode so they don't conflict with the key names defined later in the layout. Thanks for your help.

spectras, 2018-11-16:
I didn't realize there are double names. I'll fix that too.
Thanks for taking the time to report here :)

spectras, 2018-11-16:
Fixed layouts have been pushed onto master. I'll release a bit later, probably tomorrow.
Just so you know, names will be NumLockLed ScrollLockLed and CapsLockLed to be consistent with GPro layouts, which has them already.

spectras, 2018-11-17:
Published in v0.7.9. Thanks for your time V3XATI0N.

closed 2018-11-17 by spectras