keyleds/keyleds #27: color cycle effect

deathtrip, 2019-02-01:
The only common effect missing from keyleds atm, is color cycling.
I read the documentation, trying to figure out if it would be possible to achieve such effect using existing plugins, but i don't think it's doable with more than two colors. Would be cool if you could specify more than that, or the entire RGB spectrum.

deathtrip, 2019-02-02:
Here's a video:
https://www.youtube.com/watch?v=bhRJDm_hjto
and at the beginning of the following ones:
https://www.youtube.com/watch?v=S45bZ50R2Es
https://www.youtube.com/watch?v=g_PdFBMQ4rw
Hope this helps :)

spectras, 2019-02-02:
So it slowly changes the whole keyboard basically?
It can be built with a script easily if you feel so inclined, otherwise I'll be adding it in next release myself.

spectras, 2019-02-10:
It's now in `master` and will be in next release.
Instead of creating a whole new effect, it will be a modification of `wave`:

 * `length` parameter normally gives size of the wave. For instance, 500 is half-keyboard, which means keyboard will have two narrow waves at any time, while 10000 makes a very long wave, of which the keyboard displays only 10% at any time.
 * it makes no sense to use `0` for `length` (in fact in current version, effect crashes trying to divide by 0).

 * in `master` and upcoming release, `length: 0` will become a valid special value that makes the effect static (it just cycles color in place)

Other parameters will still apply (eg: `color0`, `color1`, `color2`, ... to specify colors, and `period` to tell how fast to cycle the colors). `direction` becomes moot since there is no movement.

closed 2019-03-03 by spectras