keyleds/keyleds #29: Layout issue with GPRO

jtyr, 2019-02-14:
I have bought [this](https://www.amazon.co.uk/gp/product/B06VVQKF7S) keyboard:

```
$ keyledsctl info                                
Name:           Pro Gaming Keyboard
Type:           keyboard
Model:          c33900000000
Serial:         3138470e
Firmware[c339]: application U1  v113.1.13 [active]
Firmware[aad0]: bootloader BOT v114.0.c4b1
Features:       [0001, 0003, 0022, 0005, 0000, 0040, 00b0, 0060, 00c2, 0001, 0002, 0080, 0070, 0021, 00c0, 0006, 00a2]
Known features: feature version name root feature
Report rates:   [1ms] 2ms 4ms 8ms
LED block[01]:  105 keys, max_rgb(255, 255, 255)
LED block[10]:    1 keys, max_rgb(255, 255, 255)
LED block[40]:    4 keys, max_rgb(255, 255, 255)
```

Unfortunately I cannot set color of the follwoing keys via config file:

- `backslash`
- `light`
- `game`
- `CapsLockLed`
- `ScrollLockLed`
- `logo`

I can change it by using the following `keyledsctl` commands:

```
keyledsctl set-leds -b keys x2b=blue
keyledsctl set-leds -b modes x01=blue
keyledsctl set-leds -b modes x02=blue
keyledsctl set-leds -b modes x03=blue
keyledsctl set-leds -b modes x04=blue
keyledsctl set-leds -b logo x01=red
```

I guess it's a problem with the layout loading as I cannot see any information about the loaded profile in the log output:

```
$ keyledsd -vvv
<I>main: keyledsd v0.7.9 starting up
<I>main: using /etc/keyledsd.conf
<I>service: setContext ()
<I>service: connected to display :0
<I>xinput-watcher: xinput keyboard 6 enabled for device /dev/input/event1
<I>xinput-watcher: xinput keyboard 7 enabled for device /dev/input/event0
<I>xinput-watcher: xinput keyboard 9 enabled for device /dev/input/event15
<I>xinput-watcher: xinput keyboard 11 enabled for device /dev/input/event2
<I>xinput-watcher: xinput keyboard 12 enabled for device /dev/input/event14
<I>xinput-watcher: xinput keyboard 13 enabled for device /dev/input/event18
<I>xinput-watcher: xinput keyboard 14 enabled for device /dev/input/event17
<I>xinput-watcher: xinput keyboard 15 enabled for device /dev/input/event16
<I>service: setContext (id=52428807, title=/bin/zsh, class=Terminator, instance=terminator)
<I>service: switching to active mode
<I>service: device added: /dev/hidraw2
<I>dev-manager: selected profile <__default__>
<I>effect-manager: effect fill not loaded, attempting auto-load
<I>effect-manager: loading fill from /usr/lib/keyledsd/fx_fill.so
<I>effect-manager: loaded plugin <fill>
<I>dev-manager: loaded plugin effect fill
<I>service: opened device /dev/hidraw2 [0E7D386B3532], model c33900000000 firmware U1v113.1.13, <Pro Gaming Keyboard>
```

spectras, 2019-02-14:
Hello,

Thanks for reporting here. You are right, it looks like a layout issue.
And looking at the log output, it makes me realize I forgot to add a logging message should the layout file not be found, so I don't know what it is looking for.

We should be able to see it looking for the layout file by using this command:
```bash
strace -efile -o'|grep layouts' keyledsd -v
```
It should add the relevant calls in the log output, eg:
```
<I>main: keyledsd v0.8.0 starting up
<I>effect-manager: loaded plugin <lua>
<I>service: connected to display :0
lstat("/home/spectras/.local/share/keyledsd/layouts", {st_mode=S_IFDIR|0775, st_size=4096, ...}) = 0
lstat("/home/spectras/.local/share/keyledsd/layouts/c33000000000_0005.yaml", 
```
However it might write nothing. In that case, it means keyleds could not determine the layout (so it won't try to locate the matching layout file). Please tell me so.

In particular, I notice the output of `keyledsctl info` is wrong, as it says your keyboard lacks RGB leds. That means Logitech changed something in the way it's reported. This might be why keyleds fails to read the layout.

By the way, is your keyboard new? The change might have been caused by some firmware update, if so I'll have to upgrade my keyboard's firmware to inspect this more in depth.

jtyr, 2019-02-14:
Thanks for showing me how to find out which layout file it's trying to load. I have found out that it's looking for `c33900000000_0037.xml`. After I copied `c33900000000_0002.xml`to `c33900000000_0037.xml`, almost all started to work. The only thing I had to add was the `Logo` glyph from `c33500000000_0002.xml`.

I have also noticed that the `duration` value in the `feedback` plugin is not respected regardless what value I set (tried `10000`, `1000`).

Also the version the `keyledsd` shows in the logs is `v0.7.9` even if the version I have installed is `v0.8.0.1`. Is that intentional or an error?

spectras, 2019-02-14:
You are observant, indeed it's a glitch in my packaging process, I usually test the packaging before validating the release. Thus that testing package still has previous release number. I normally rebuild it after validating so it gets the correct version, but I forgot to do it on v0.8.0.

So your keyboard reports layout 0x0037. Interesting. Might be that logitech decided to rework layout numbering. Is it a US layout?
Anyway, in addition to adding that layout, I'm identifying two improvements here:
 * Logging the layout so using debugging tools for that is not required.
 * Perhaps an automatic fallback to the INTL layout would be a good idea.

As for the `feedback` plugin, it's a documentation bug. I just edited it, but here it is:
 * `duration` no longer exists, it was split into:
 * `sustain` for how long the key stays at maximum feedback level
 * `decay` for how long it takes for it to turn back to normal

That's shamelessly stolen from computer music domain, sound envelope:
[image]
(`attack` cannot be configured and is always 0 - it's under the user's fingers so it should not matter)

jtyr, 2019-02-14:
Yes, my keyboard has US layout.

Thanks for updating the documentation of the `feedback` plugin. Using the `sustain` and `decay` options works as expected. Only if I set `decay: 0` the `keyledsd` throws `floating point exception`.

spectras, 2019-02-14:
Noted, I have to admit, even though the engine itself is built very solidly, I did not put the same bullet-proofing effort in plugins, so some values might crash them. I have to do an input-checking pass on them, it's in my todo list :)

closed 2019-03-03 by spectras