keyleds/keyleds #30: Groups of groups

jtyr, 2019-02-14:
It would be nice if we could define groups which contain other groups:

```yaml

groups:
  chars:
    first_row: [q, w, e, r, t, y, u, i, o, p]
    second_row: [a, s, d, f, g, h, j, k, l]
    third_row: [z, x, c, v, b, n, m]

effects:
  keyleds-default:
    plugins:
      - effect: fill
        color: black
        chars: cyan
      - plugin: breathe
        group: first_row
        period: 5000
        color: blue
```

spectras, 2019-02-14:
You mean, as way to make the file look better, or carrying some feature?
I mean, from your example, how is it functionally different from this?
```yaml
groups:
  # some char groups
  first_row: [q, w, e, r, t, y, u, i, o, p]
  second_row: [a, s, d, f, g, h, j, k, l]
  third_row: [z, x, c, v, b, n, m]
```

jtyr, 2019-02-14:
No. The goal here is to be able to address individual rows as well as all rows together via group called `chars`.

spectras, 2019-02-14:
I see. Let's add it to todo list as well.

spectras, 2019-02-21:
Completed. It will be part of next release.

Any keygroup definition will now support nested definitions. There is no nesting limit, should you want to go all Inception with your keygroups.

closed 2019-03-03 by spectras