keyleds/keyleds #31: Custom color names

jtyr, 2019-02-14:
It woudl be nice if we could define custom colors:

```yaml

colors:
  my_cyan: 008374

effects:
  keyleds-default:
    plugins:
      - effect: fill
        color: my_cyan
```

spectras, 2019-02-14:
That's a good idea.
By the way, though not  technically the same thing, you can already use YAML anchors/references to achieve something similar:

```yaml
effects:
  some-effect:
    plugins:
      - effect: wave
        color0: &my_cyan 008374
        color1: &my_red f02010
  some-other-place:
    plugins:
      - effect: breathe
        color: *my_cyan
```
Such anchors work for any simple field, and anchor must appear above reference.

Alas, the parser explicitly disallow sections it does not know, so you cannot make a dummy section at top to hold those. You *could* hijack the `devices` section to put color definitions, like
```yaml
devices:
    unused1: &my_cyan 008374
```
but that's ugly.
Adding a proper section for that would be great indeed.

jtyr, 2019-02-14:
I thought about the anchors as well but it's too hacky ;o)

spectras, 2019-02-21:
Completed. Will be part of next release.

Configuration file will accept a `colors` section to define custom colors:
```yaml
colors:
    mycolor: red
    myothercolor: ff00ff80
````
Custom colors will override predefined ones, so this works as expected:
```yaml
colors:
    red: green     # redefine "red" to mean green
    mycolor: red   # mycolor is red (section is read at once, then overrides are applied)
    ff0000: 00ff00 # it works with color values too
```

This will also works for custom effects. For instance if some custom effect does `tocolor("red")` with above section, it will actually get a green.

closed 2019-03-03 by spectras