keyleds/keyleds #32: Multiple groups for one plugin definition

jtyr, 2019-02-14:
It would be nice if we could specify multiple groups for which a plugin applies:

```yaml
- plugin: breathe
  # Either string with single color or list with multiple colors could be defined
  group: [group1, group2]
  period: 5000
  color: 009090
```

spectras, 2019-02-14:
Indeed. Adding to my todo list.

spectras, 2019-03-03:
I have been assessing the complexity of that feature, and it turns out providing this transparently would have, with the current design, a performance cost that outweighs the benefit of making configuration a bit easier.

May I suggest using nested groups and yaml anchors to define such groups? This kind of things should work:
```yaml
groups:
    function:
        F1toF4: &F1F4 [F1, F2, F3, F4]
        F5toF8: [F5, F6, F7, F8]
        F9toF12: &F9F12 [F9, F10, F11, F12]
    sides:
        left: *F1F4
        right: *F9F12
```
It is also possible to limit such definitions to a single effect stack:
```yaml
groups:
    F1toF4: &F1F4 [F1, F2, F3, F4]
    F5toF8: [F5, F6, F7, F8]
    F9toF12: &F9F12 [F9, F10, F11, F12]
effects:
    my-effect-stack:
        groups:
            sides: [left: *F1F4, right: *F9F12]
        plugins:
          - plugin: fill
            color: darkblue
          - plugin: breathe
            group: sides
            period: 5000
            color: 009090
```
Untested, but I have similar things in my own config. If it gives you trouble, that's a bug.