keyleds/keyleds #33: Alpha has no effect on the final color

jtyr, 2019-02-15:
I'm trying to use color in the format `RRGGBBAA` but the Alpha part has no effect on the final color. Is that something what has to be explicitly supported by the keyboard?

spectras, 2019-02-15:
Hello,

It's handled by the engine, and should work fine regardless of keyboard model.
The engine makes "virtual" keyboards available to plugins, then uses alpha to blend them together.

There is one exception to that model: the `fill` plugin does not obey alpha.
The intent is for this plugin to be used at top of effect list to define the base which downstream effects will draw on.

If you want to make some kind of "coloring filter", this must be done through a custom effect.

Something like this should work:
```lua
-- Filter effect, in my-filter-effect.lua

config = {
    color = tocolor(keyleds.config.color) or tocolor('white')
    keys = keyleds.groups[keyleds.config.group] or keyleds.db
}

function init()
    buffer = RenderTarget:new()

    for i, key in ipairs(config.keys) do
        buffer[key] = config.color
    for 
end

function render(ms, target) target:blend(buffer) end
```
It would be used like this:
```yaml
    plugins:
      - effect: my-filter-effect
        color: ff000080
        group: somegroup   # or omit for full keyboard
```

spectras, 2019-02-22:
After thinking about this for a while, I realized `alpha` has no use at all in current state for fill plugin.

Giving it a use by allowing blending like other plugins do would actually be a sensible approach.
I rewrote it so it will use alpha in next version, if some is provided to it.

closed 2019-03-03 by spectras