keyleds/keyleds #34: Add support for G513

jtyr, 2019-02-23:
I love `keyleds` with my GPro so I have decided to buy G513:

```
$ keyledsctl list
/dev/hidraw1 046d:c33c [1660375C3135]
$ keyledsctl info                   
Name:           G513 Carbon Tactile
Type:           keyboard
Model:          c33c00000000
Serial:         35374716
Firmware[c33c]: application U1  v119.0.12 [active]
Firmware[aadc]: bootloader BOT v165.0.60c0
Features:       [0001, 0003, 0022, 0005, 0000, 0040, 00b0, 0060, 00c2, 0001, 0002, 0080, 0070, 0021, 00c0, 0006, 00a2, 0004, 0020]
Known features: feature version name root feature
Report rates:   [1ms] 2ms 4ms 8ms
LED block[01]:  104 keys, max_rgb(255, 255, 255)
LED block[40]:    3 keys, max_rgb(255, 255, 255)
```

Unfortunately there is no support for G513 yet so I at least tried to use existing layouts (`c33900000000_0001.yaml`, `c33300000000_0001.yaml`, `c32b00000000_0001.yaml`, `c33100000000_0001.yaml` and `c33000000000_0001.yaml`) but with no success. The keyboard only starts to do the blue wave effect and keep re-syncing forever:

```
$ keyledsd -v -c /etc/keyledsd.conf
<N>service: connected to display :0
<N>service: switching to active mode
<N>service: opened device /dev/hidraw1 [1660375C3135], model c33c00000000 firmware U1v119.0.12, <G513 Carbon Tactile>
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
^C<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
^C<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
```

At this stage I'm not even able to quit the process with CTRL+C nor kill it (`SIGTERM`).

If I remove the `c33c00000000_0001.yaml` completely and try to run `keyledsctl set-leds all=red` some of the keys turn to red but as soon as I touch a key in the row, the whole row changes color to white. Only some keys (`lshift`, `lmeta`, `lalt`, `ralt`, `fn`, `rctrl` and `rshift`) stay red for ever and I can change their color as I wish.

Are there some instructions how to add support for a new keyboard? Or would you be able to guide me?

jtyr, 2019-02-23:
Just realized that if I stop `keyledsd` and then run `keyledsctl set-leds all=red` then all keys turn to red.

spectras, 2019-02-23:
Hello,

Thanks for taking the time to post this  :)
I guess you are using  the latest `master` version since you have yaml layouts?

 *  Can you confirm that `keyledsctl` can correctly update key colors? `All` seems to work, but confirmation for individual keys would be nice (it's a different low-level command).
 * Do you also see `<W>render-loop: error on device: device busy re-syncing device` messages with no layout file?  With the service refusing to shutdown?

Could you possibly post:
 * `keyledsd` log output with `-vvv`
 * `keyledsctl -dd get-leds`

Also, after running `keyledsctl set-leds -b modes 1=red 2=green 3=blue`, if you could tell me which  keys turn red green and blue that would be great :)

spectras, 2019-02-24:
(by the way, feel free to send SIGKILL to keyledsd if needed, it's perfectly safe)

jtyr, 2019-02-24:
I have actually made it working. I just didn't notice that the `/etc/keyleds.conf` got rewritten by the default config and this is why I saw the blue wave effect instead of my configuration. After I put back my config and used one of the layouts, it started to work just fine.

But the `device busy re-syncing device` issue is still there:

```
$ keyledsd -vvv
<I>main: keyledsd v0.8.0 starting up
<I>main: using /etc/keyledsd.conf
<I>service: setContext ()
<N>service: connected to display :0
<I>xinput-watcher: xinput keyboard 6 enabled for device /dev/input/event1
<I>xinput-watcher: xinput keyboard 7 enabled for device /dev/input/event0
<I>xinput-watcher: xinput keyboard 8 enabled for device /dev/input/event2
<I>xinput-watcher: xinput keyboard 9 enabled for device /dev/input/event3
<I>xinput-watcher: xinput keyboard 10 enabled for device /dev/input/event4
<I>xinput-watcher: xinput keyboard 12 enabled for device /dev/input/event7
<I>xinput-watcher: xinput keyboard 14 enabled for device /dev/input/event8
<I>service: setContext (id=44040199, title=/bin/zsh, class=Terminator, instance=terminator)
<N>service: switching to active mode
<I>service: device added: /dev/hidraw1
<I>layout: loading layout /usr/share/keyledsd/layouts/c33c00000000_0001.yaml
<I>dev-manager: selected profile <__default__>
<I>effect-manager: loading fill from /usr/lib/keyledsd/fx_fill.so
<I>effect-manager: loaded plugin <fill>
<I>dev-manager: loaded plugin effect fill
<N>service: opened device /dev/hidraw1 [1660375C3135], model c33c00000000 firmware U1v119.0.12, <G513 Carbon Tactile>
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
<W>render-loop: error on device: device busy re-syncing device
....
```

Additional information:

```
$ keyledsctl -dd get-leds
libkeyleds: Opening device /dev/hidraw0
libkeyleds: Parsing report descriptor (65 bytes)
libkeyleds: invalid device (no hid++ support)
libkeyleds: Opening device /dev/hidraw1
libkeyleds: Parsing report descriptor (119 bytes)
libkeyleds: Found report ID 0x11 (19 bytes)
libkeyleds: Found report ID 0x12 (63 bytes)
libkeyleds: Opened device /dev/hidraw1 protocol version 4
libkeyleds: Opening device /dev/hidraw2
libkeyleds: Permission denied
libkeyleds: Opening device /dev/hidraw1
libkeyleds: Parsing report descriptor (119 bytes)
libkeyleds: Found report ID 0x11 (19 bytes)
libkeyleds: Found report ID 0x12 (63 bytes)
libkeyleds: Opened device /dev/hidraw1 protocol version 4
libkeyleds: feature 8080 is at 12 [00]
A=#00ffff
B=#00ffff
C=#00ffff
D=#00ffff
E=#00ffff
F=#00ffff
G=#00ffff
H=#00ffff
I=#00ffff
J=#00ffff
K=#00ffff
L=#00ffff
M=#00ffff
N=#00ffff
O=#00ffff
P=#00ffff
Q=#00ffff
R=#00ffff
S=#00ffff
T=#00ffff
U=#00ffff
V=#00ffff
W=#00ffff
X=#00ffff
Y=#00ffff
Z=#00ffff
1=#00ffff
2=#00ffff
3=#00ffff
4=#00ffff
5=#00ffff
6=#00ffff
7=#00ffff
8=#00ffff
9=#00ffff
0=#00ffff
ENTER=#00ffff
ESC=#ff0000
BACKSPACE=#ff0000
TAB=#00ffff
SPACE=#00ffff
MINUS=#00ffff
EQUAL=#00ffff
LBRACE=#00ffff
RBRACE=#00ffff
BACKSLASH=#00ffff
BACKSLASH=#000000
SEMICOLON=#00ffff
APOSTROPHE=#00ffff
GRAVE=#00ffff
COMMA=#00ffff
DOT=#00ffff
SLASH=#00ffff
CAPSLOCK=#ff0000
F1=#8a2be2
F2=#8a2be2
F3=#8a2be2
F4=#8a2be2
F5=#8a2be2
F6=#8a2be2
F7=#8a2be2
F8=#8a2be2
F9=#8a2be2
F10=#8a2be2
F11=#8a2be2
F12=#8a2be2
SYSRQ=#8a2be2
SCROLLLOCK=#8a2be2
PAUSE=#8a2be2
INSERT=#ff0000
HOME=#d2691e
PAGEUP=#d2691e
DELETE=#ff0000
END=#d2691e
PAGEDOWN=#d2691e
RIGHT=#d2691e
LEFT=#d2691e
DOWN=#d2691e
UP=#d2691e
NUMLOCK=#8a2be2
KPSLASH=#00ffff
KPASTERISK=#00ffff
KPMINUS=#00ffff
KPPLUS=#00ffff
KPENTER=#00ffff
KP1=#00ffff
KP2=#00ffff
KP3=#00ffff
KP4=#00ffff
KP5=#00ffff
KP6=#00ffff
KP7=#00ffff
KP8=#00ffff
KP9=#00ffff
KP0=#00ffff
KPDOT=#00ffff
102ND=#000000
COMPOSE=#ff0000
RO=#000000
KATAKANAHIRAGANA=#ff0000
YEN=#000000
HENKAN=#000000
MUHENKAN=#000000
LCTRL=#ff0000
```

If I run:

```
$ keyledsctl set-leds all=black
$ keyledsctl set-leds -b modes 1=red 2=green 3=blue
```

no key lids up.

jtyr, 2019-02-24:
Btw. it would be nice if `keyledsd` would reset all key colors to what's configured in the config file if I send `SIGHUP` to the process. That would be really handy when I start `keyledsd` on the background at the start of my session and then play with colors manually with `keyledsctl`.

spectras, 2019-02-24:
Great that you made it work. I'll add layout files in the next release, I got a picture from google to figure out where keys are exactly.

Hmmm, I believe the syncing issue to be due to updating the state too fast, perhaps I should slow down from 16fps to 12 fps. It's barely noticeable and would probably avoid saturating the G513.

----
I'm not sure I understand what you mean about sighup?
If it's working fine, it should react like this:
 * when the config file it uses is modified, it should reload the configuration automatically.
 * on SIGHUP, it should re-do the search for the config file, then reload it (it's useful if you moved it / deleted it / added one in `$HOME/.config/keyledsd.conf` so it switches to the new file).

In both cases, if it's valid, the new configuration is applied immediately. Do you experience a different behavior?

`keyledsctl` is mostly a debugging tool. It does not communicate with `keyledsd` at all. Although it can be used while `keyledsd` runs, setting leds won't work because whatever you change just gets overwritten immediately if any effect is animating it.

I'd be curious as to what you normally do with `keyledsctl` so I can perhaps provide support for a less hacking alternative :)

jtyr, 2019-02-24:
Thanks for adding the layout. Please just make sure you also add `Game` and `CapsLockLed` keys there. This keyboard doesn't have `Game` logo which you can set with a color. It only has a `Game` indicator right next to the Caps Lock. Perhaps adding it as `GameLed` would be better?

I'm using only `fill` plugin in my configuration. Therefore if I set some key to a different color via `keyledsctl`, the `keyledsd` doesn't reset it back even if I change the config file. I was hoping to be able to reset the colors with `SIGHUP` like this:

```
# Assuming the keyledsd is running
# Set color of one key to a different color
keyledsctl set-leds enter=red
# This should reset all keys to what's defined in the config
pkill -HUP keyledsd
```

spectras, 2019-02-24:
Thanks for the info, indeed I have still pictures so I could not know whether logo just happened to be off or it didn't have light at all, that cleared it up :)

Okay, I understand what you meant about the reset. Indeed, to save on USB bandwidth, only keys that actually changed colors are sent to the keyboard, so if it's still no update will happen at all. Making it force-refresh everything on `SIGHUP` sounds doable.

spectras, 2019-03-02:
Update: `master` version will respond to `SIGUSR1` by refreshing all keys on all devices.
It's starting to be quite a few tweaks and features, I should make an official release soon.

spectras, 2019-03-02:
And `master` should have a layout file. I cannot test it though. I think I'll publish anyway and when you test it open an issue if keys are amiss. (especially the game/capslockled).

jtyr, 2019-03-02:
I have just teste the build from master and these are my findings:

1) The layout file is found but the FN key (code `126`) is not addressable via `fn` but it responds to `rmeta`.
2) The `<W>render-loop: error on device: device busy re-syncing device` is still present. The `leyledsd` process do not react on `SIGTERM` while the loop problem occurs.
3) `SIGUSR1` works just fine.
4) The alpha doesn't seem to work. I tried the following two variants and coun't see any difference:

```yaml
effects: 
  myeffect: 
    plugins: 
      - effect: fill
        modifiers: ff000000
```

```yaml
effects: 
  myeffect: 
    plugins: 
      - effect: fill
        modifiers: ff0000ff
```
I also tried to set alpha with `keyledsctl` but also without the desired effect:

```shel
# Fully bright red
keyledsctl set-leds all=black enter=ff0000
# Black
keyledsctl set-leds all=black enter=ff000000
# Dimmed blue
keyledsctl set-leds all=black enter=ff000050
# Fully bright blue
keyledsctl set-leds all=black enter=ff0000ff
```

spectras, 2019-03-02:
Hello again, thanks for taking the time to test!

  1. Looking into it!
  2. Yes, I will reduce FPS in release but I did not do it yet in master. Sorry for the inconvenience.
  3. Great!

**For item4:**
 * The keyboard itself has no concept of alpha, it just accepts `(red, green, blue)` values.
 * `keyledsctl` is a low-level tool, it talks to the keyboard directly and therefore has no concept of alpha at all. That's why you see no difference. In fact it just cuts after 6 digits.
 * Now, to allow you to mix several effects, `keyledsd` implements a synthetic alpha channel. It goes like this:
    - Effects each own a virtual copy of your keyboard. They draw whatever they want on it. It has an alpha channel so they can set transparent keys.
    - Then `keyledsd` blends them together using the alpha channel.
    - The alpha channel is then removed, and the resulting colors is sent to the keyboard.

Thus, alpha channel is only relevant to let an effect appear over or under another effect.
Most notably, the very first effect in the stack should be completely opaque. If you want it to have dimmer color, simply use a darker color (thus, replace `FF000050` with `500000FF`).

spectras, 2019-03-03:
I published version 1.0.0. It includes the fix for fn key on your keyboard, and refresh frequency is now 12 instead of 16. Hopefully that should be enough for the G513.
I'm having an issue with Fedora packages, but others are uploaded already :)

jtyr, 2019-03-03:
I just tested the version from `master` and the re-syncing issue still persists. It keeps re-syncing for few minutes and then it stops printing the message. But if I try `SIGUSR1` it prints `.<W>render-loop: error on device: device busy re-syncing device`. Please let me know if you want me to run some more tests.

The layout for G513 seems to work but I'm still not possible to change the `FN` led via `keyledsctl` (`126`, `x7e` and `rmeta` works just fine via `kleyledsctl` as well as `fn` via `keyledsd` config). I guess the `keyledsctl` doesn't read the layout file?

spectras, 2019-03-05:
You are correct, `keyledsctl` does not know of layout, it speaks directly to the keyboard using the raw underlying codes. The layout file causes `keyledsd` to translate fn to rmeta, but `keyledsctl` is a much simpler tool. It is intended for debugging and poking leds around to debug `keyledsd` layouts.

For the resyncing issue, I guess that's still too fast for the keyboard? That seems odd.
If that's not too much to ask, can you try to change the update speed in your version?
 * open keyledsd/include/config.h.in, on line ~36
 * change KEYLEDSD_RENDER_FPS to 2
 * rebuild

If it makes the syncing error disappear, try increasing the number to see when it comes back.

It would be great!

jtyr, 2019-03-06:
I tried to apply this patch:

```diff
--- ./keyleds.orig/keyledsd/include/config.h.in	2019-02-23 12:33:37.904558634 +0000
+++ ./keyleds/keyledsd/include/config.h.in	2019-03-05 23:48:23.056692831 +0000
@@ -33,7 +33,7 @@
 #cmakedefine KEYLEDSD_USE_SSE2
 #cmakedefine KEYLEDSD_USE_AVX2
 #define KEYLEDSD_APP_ID         (0x4)
-#define KEYLEDSD_RENDER_FPS     (12)
+#define KEYLEDSD_RENDER_FPS     (1)
 
 // Feature detection results
 #cmakedefine HAVE_BUILTIN_CPU_SUPPORTS
```

And I'm still getting the re-syncing message for a random period of time. Please let me know if you want me to try something else.

spectras, 2019-03-06:
Thanks for taking the time to try.
Well, it seems there might be a change in the underlying protocol, or perhaps this device is reacting more slowly. I'll try to find a way to get more information and tell you soon. Perhaps enabling low-level usb debugging or something like this.

If I am lucky I might find a 513 close to me that I can use to test myself.

spectras, 2019-03-08:
Hello again,

When you have the time, I would like to know if you could try this:

 1. Open `libkeyleds/src/logging.c`.
 2. Edit line 24, to set debug level to `KEYLEDS_LOG_DEBUG`.
 3. Rebuild.

Then run `keyledsd` with `-vvv`.
This should generate *a lot* of debugging output, including a dump of all the transfers to/from the device. If you could record those and either post them here, or upload them and post a link, it would be great.

I am trying to understand what step exactly makes the device choke.
Sorry for the inconvenience, I really hope this will have enough information for me to fix it for good.

jtyr, 2019-03-08:
I have applied this patch:

```diff
--- keyleds.orig/libkeyleds/src/logging.c	2019-03-05 23:57:58.643923163 +0000
+++ keyleds/libkeyleds/src/logging.c	2019-03-08 00:11:16.534091062 +0000
@@ -21,5 +21,5 @@
 #include "keyleds/logging.h"
 
 KEYLEDS_EXPORT FILE * g_keyleds_debug_stream = NULL;
-KEYLEDS_EXPORT int g_keyleds_debug_level = KEYLEDS_LOG_WARNING;
+KEYLEDS_EXPORT int g_keyleds_debug_level = KEYLEDS_LOG_DEBUG;
 KEYLEDS_EXPORT int g_keyleds_debug_hid = KEYLEDS_LOG_ERROR;
```

and this is what `keyledsd` logged:

```
<6>main: keyledsd v1.0.0 starting up
<6>main: using /etc/keyledsd.conf
<6>service: setContext ()
<5>service: connected to display :0
<6>xinput-watcher: xinput keyboard 6 enabled for device /dev/input/event1
<6>xinput-watcher: xinput keyboard 7 enabled for device /dev/input/event0
<6>xinput-watcher: xinput keyboard 8 enabled for device /dev/input/event2
<6>xinput-watcher: xinput keyboard 9 enabled for device /dev/input/event3
<6>xinput-watcher: xinput keyboard 10 enabled for device /dev/input/event4
<6>xinput-watcher: xinput keyboard 12 enabled for device /dev/input/event16
<6>xinput-watcher: xinput keyboard 14 enabled for device /dev/input/event17
<6>service: setContext (id=56623111, title=/bin/zsh, class=Terminator, instance=terminator)
<5>service: switching to active mode
<6>service: device added: /dev/hidraw1
<6>layout: loading layout /usr/share/keyledsd/layouts/c33c00000000_0001.yaml
<6>dev-manager: selected profile <__default__>
<6>effect-manager: loading fill from /usr/lib/keyledsd/fx_fill.so
<6>effect-manager: loaded plugin <fill>
<6>dev-manager: loaded plugin effect fill
<5>service: opened device /dev/hidraw1 [1660375C3135], model c33c00000000 firmware U1v119.0.12, <G513 Carbon Tactile>
<4>render-loop: error on device: device busy re-syncing device
<4>render-loop: error on device: device busy re-syncing device
...
```

Then I tried to apply also this patch:

```diff
--- keyleds.orig/libkeyleds/src/logging.c	2019-03-08 00:19:21.086259513 +0000
+++ keyleds/libkeyleds/src/logging.c	2019-03-08 00:18:22.602935414 +0000
@@ -21,5 +21,5 @@
 #include "keyleds/logging.h"
 
 KEYLEDS_EXPORT FILE * g_keyleds_debug_stream = NULL;
-KEYLEDS_EXPORT int g_keyleds_debug_level = KEYLEDS_LOG_WARNING;
-KEYLEDS_EXPORT int g_keyleds_debug_hid = KEYLEDS_LOG_ERROR;
+KEYLEDS_EXPORT int g_keyleds_debug_level = KEYLEDS_LOG_DEBUG;
+KEYLEDS_EXPORT int g_keyleds_debug_hid = KEYLEDS_LOG_DEBUG;
```

and this is what it logged:

```
<6>main: keyledsd v1.0.0 starting up
<6>main: using /etc/keyledsd.conf
<6>service: setContext ()
<5>service: connected to display :0
<6>xinput-watcher: xinput keyboard 6 enabled for device /dev/input/event1
<6>xinput-watcher: xinput keyboard 7 enabled for device /dev/input/event0
<6>xinput-watcher: xinput keyboard 8 enabled for device /dev/input/event2
<6>xinput-watcher: xinput keyboard 9 enabled for device /dev/input/event3
<6>xinput-watcher: xinput keyboard 10 enabled for device /dev/input/event4
<6>xinput-watcher: xinput keyboard 12 enabled for device /dev/input/event16
<6>xinput-watcher: xinput keyboard 14 enabled for device /dev/input/event17
<6>service: setContext (id=56623111, title=/bin/zsh, class=Terminator, instance=terminator)
<5>service: switching to active mode
<6>service: device added: /dev/hidraw1
libkeyleds: Opening device /dev/hidraw1
libkeyleds: Parsing report descriptor (119 bytes)
libkeyleds: Found report ID 0x11 (19 bytes)
libkeyleds: Found report ID 0x12 (63 bytes)
libkeyleds: Opened device /dev/hidraw1 protocol version 4
libkeyleds: feature 0005 is at 4 [00]
libkeyleds: feature 0003 is at 2 [00]
libkeyleds: feature 4540 is at 6 [00]
libkeyleds: feature 8080 is at 12 [00]
<6>layout: loading layout /usr/share/keyledsd/layouts/c33c00000000_0001.yaml
<6>dev-manager: selected profile <__default__>
<6>effect-manager: loading fill from /usr/lib/keyledsd/fx_fill.so
<6>effect-manager: loaded plugin <fill>
<6>dev-manager: loaded plugin effect fill
<5>service: opened device /dev/hidraw1 [1660375C3135], model c33c00000000 firmware U1v119.0.12, <G513 Carbon Tactile>
libkeyleds: device busy
<4>render-loop: error on device: device busy re-syncing device
libkeyleds: device busy
<4>render-loop: error on device: device busy re-syncing device
libkeyleds: device busy
...
```

spectras, 2019-03-08:
So it did not log usb traces.
Hmmm. Are you compiling it in release mode maybe? I believe this removes some of the debugging output.

Yeah, here it is: release mode excludes usb traces.

Could you please either build in debug mode (if using cmake, `cmake -DCMAKE_BUILD_TYPE ..`), or apply this patch:
```patch
diff --git a/libkeyleds/src/device.c b/libkeyleds/src/device.c
index ef3e25f..5318ce4 100644
--- a/libkeyleds/src/device.c
+++ b/libkeyleds/src/device.c
@@ -176,7 +176,6 @@ KEYLEDS_EXPORT bool keyleds_flush_fd(Keyleds * device)

 /****************************************************************************/

-#ifndef NDEBUG
 static void format_buffer(const uint8_t * data, size_t size, char * buffer)
 {
     for (size_t idx = 0; idx < size; idx += 1) {
@@ -184,7 +183,6 @@ static void format_buffer(const uint8_t * data, size_t size, char * buffer)
     }
     buffer[3 * size - 1] = '\0';
 }
-#endif

 /** Send a report to the device, running an on-device function.
@@ -222,13 +220,11 @@ bool keyleds_send(Keyleds * device, uint8_t target_id, uint8_t feature_idx,
     if (length > 0) { memcpy(&buffer[4], data, length); }
     memset(&buffer[4 + length], 0, report_size - 3 - length);

-#ifndef NDEBUG
     if (g_keyleds_debug_level >= KEYLEDS_LOG_DEBUG) {
         char debug_buffer[3 * (1 + report_size) + 1];
         format_buffer(buffer, (1 + report_size), debug_buffer);
         KEYLEDS_LOG(DEBUG, "Send [%s]", debug_buffer);
     }
-#endif

     /* Send the report to the device */
     ssize_t nwritten = write(device->fd, buffer, 1 + report_size);
@@ -293,13 +289,12 @@ bool keyleds_receive(Keyleds * device, uint8_t target_id, uint8_t feature_idx,
             keyleds_set_error_errno();
             return false;
         }
-#ifndef NDEBUG
         if (g_keyleds_debug_level >= KEYLEDS_LOG_DEBUG) {
             char debug_buffer[3 * nread + 1];
             format_buffer(message, (size_t)nread, debug_buffer);
             KEYLEDS_LOG(DEBUG, "Recv [%s]", debug_buffer);
         }
-#endif
+
         /* Check the received report type against our known report types */
         for (idx = 0; device->reports[idx].id != DEVICE_REPORT_INVALID; idx += 1)
         {
```

Thanks for all the time you're spending on this, it's really appreciated :)

jtyr, 2019-03-08:
Here is the output of the debug build:

```
<6>main: keyledsd v1.0.0 starting up
<6>main: using /etc/keyledsd.conf
<6>service: setContext ()
<7>filewatcher: subscribed to events on /etc/keyledsd.conf => 1
<7>service: created
<5>service: connected to display :0
<7>xwindow: subscribed to events on :0 => 1
<7>xwindow: subscribed to events on :0 => 2
<6>xinput-watcher: xinput keyboard 6 enabled for device /dev/input/event1
<6>xinput-watcher: xinput keyboard 7 enabled for device /dev/input/event0
<6>xinput-watcher: xinput keyboard 8 enabled for device /dev/input/event2
<6>xinput-watcher: xinput keyboard 9 enabled for device /dev/input/event3
<6>xinput-watcher: xinput keyboard 10 enabled for device /dev/input/event4
<6>xinput-watcher: xinput keyboard 12 enabled for device /dev/input/event16
<6>xinput-watcher: xinput keyboard 14 enabled for device /dev/input/event17
<6>service: setContext (id=56623111, title=/bin/zsh, class=Terminator, instance=terminator)
<5>service: switching to active mode
<6>service: device added: /dev/hidraw1
libkeyleds:keyleds/libkeyleds/src/device.c:61: Opening device /dev/hidraw1
libkeyleds:keyleds/libkeyleds/src/device.c:77: Parsing report descriptor (119 bytes)
REPORT item type a [0001]
    Usage: 00010006
REPORT item type 8 [0000]
    Report ID:          01
    Report size:         8
    Report count:       20
    Logical minimum:     0
    Logical maximum:   231
REPORT item type c [0000]
REPORT item type a [0001]
    Usage: 000c0001
REPORT item type 8 [0002]
    Usage: 000c00e2
    Report ID:          02
    Report size:         1
    Report count:        7
    Logical minimum:     0
    Logical maximum:     1
REPORT item type 8 [0001]
    Report ID:          02
    Report size:         1
    Report count:        1
    Logical minimum:     0
    Logical maximum:     1
REPORT item type c [0000]
REPORT item type a [0001]
    Usage: ff430602
REPORT item type 8 [0000]
    Usage: ff430002
    Report ID:          11
    Report size:         8
    Report count:       19
    Logical minimum:     0
    Logical maximum:   255
REPORT item type 9 [0000]
    Usage: ff430002
    Report ID:          11
    Report size:         8
    Report count:       19
    Logical minimum:     0
    Logical maximum:   255
REPORT item type c [0000]
REPORT item type a [0001]
    Usage: ff430604
REPORT item type 8 [0000]
    Usage: ff430004
    Report ID:          12
[... 101 lines cut ...]
libkeyleds:keyleds/libkeyleds/src/device.c:300: Recv [12 ff 0c 24 00 01 00 62 87 00 00 00 88 ff 00 00 89 00 00 00 8a 00 00 00 8b 00 00 00 e0 ff 00 00 e1 ff 00 00 e2 ff 00 00 e3 ff 00 00 e4 ff 00 00 e5 ff 00 00 e6 ff 00 00 e7 ff 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:229: Send [11 ff 0c 24 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:300: Recv [12 ff 0c 24 00 40 00 00 01 00 dc ff 02 00 00 00 03 00 00 00 04 00 dc ff 05 00 dc ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
<7>anim-loop: AnimationLoop(0x562ca4f87c78) started
libkeyleds:keyleds/libkeyleds/src/device.c:229: Send [11 ff 0c 34 00 01 00 01 88 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:300: Recv [11 ff 0c 34 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:229: Send [11 ff 0c 34 00 40 00 03 01 00 00 00 02 00 00 80 03 00 80 00]
libkeyleds:keyleds/libkeyleds/src/device.c:300: Recv [11 ff 0c 34 00 40 00 03 00 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:229: Send [11 ff 0c 54 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:300: Recv [11 ff ff 0c 54 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/error.c:116: device busy
<4>render-loop: error on device: device busy re-syncing device
libkeyleds:keyleds/libkeyleds/src/device.c:229: Send [11 ff 00 14 00 00 68 00 00 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:300: Recv [11 ff 00 14 04 02 68 00 00 00 00 00 00 00 00 00 00 00 00 00]
<7>anim-loop: AnimationLoop(0x562ca4f87c78) started
libkeyleds:keyleds/libkeyleds/src/device.c:229: Send [11 ff 0c 34 00 01 00 01 88 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:300: Recv [11 ff 0c 34 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:229: Send [11 ff 0c 34 00 40 00 03 01 00 00 00 02 00 00 80 03 00 80 00]
libkeyleds:keyleds/libkeyleds/src/device.c:300: Recv [11 ff 0c 34 00 40 00 03 00 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:229: Send [11 ff 0c 54 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/device.c:300: Recv [11 ff ff 0c 54 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
libkeyleds:keyleds/libkeyleds/src/error.c:116: device busy
<4>render-loop: error on device: device busy re-syncing device
...
```

spectras, 2019-03-09:
That's great! Now we know it is the committing that chokes the keyboard.
Perhaps the keyboard needs some delay to apply sent data before it can actually commit.

I created a branch: https://github.com/keyleds/keyleds/tree/issues/34 

I modified the render loop so every time the sending fails, it will insert more delay.
The initial attempt should still show the error but if you let it run, hopefully it will reach a delay amount where the error disappears.
Please let me know if it gets rid of the error, and if so which value for the `increased commit delay to XXXus` is last seen.

jtyr, 2019-03-09:
The issue stopped when the delay reached `3560us`:
```
...
<4>render-loop: increased commit delay to 3540us
<7>anim-loop: AnimationLoop(0x5642defc9c78) started
<3>render-loop: error on device: device busy re-syncing device
<4>render-loop: increased commit delay to 3550us
<7>anim-loop: AnimationLoop(0x5642defc9c78) started
<3>render-loop: error on device: device busy re-syncing device
<4>render-loop: increased commit delay to 3560us
<7>anim-loop: AnimationLoop(0x5642defc9c78) started
```

spectras, 2019-03-10:
Thanks!

Wow, that's awfully slow. Normally, keyleds does not wait at all.
I wonder whether it's normal for G513 or if your specific device is faulty.
Either way, I guess I'll keep the automatic delay. It needs to grow much faster though, 356 steps is a bit long. Or perhaps save it for subsequent times?

I'm glad a working solution finally is at hand.

ritze, 2019-03-10:
Hey,

I have also a Logitech G513.

```
<E>render-loop: error on device: device busy re-syncing device
<W>render-loop: increased commit delay to 3890us
<E>render-loop: error on device: device busy re-syncing device
<W>render-loop: increased commit delay to 3910us
<E>render-loop: error on device: device busy re-syncing device
<W>render-loop: increased commit delay to 3920us
```

After 3890µs the delay value stabilizes. I think all further device busy errors are more random and the delay value has not to be increased.

spectras, 2019-03-11:
@ritze thanks a lot for stepping in with a comparison point.

So it looks like the G513 itself is really slow at applying color. I think I'll try to get in touch with Logitech to signal the issue. They've not been very supportive so far but you never know.

In the meantime I have merged the fix, along with a guard: delay value will only increase if errors happen in rapid succession, and will never exceed 8ms no matter what. This is on `master` now and will be in next release.

Are you encountering many further device errors? On other keyboard models I have yet to see any at all, with no delay. Well, yes, when computer comes back from sleep I sometimes witnessed a single device error.

spectras, 2019-03-11:
I sent an email to logitech support. It would be amazing if they actually answered in a useful way.
In the meantime I also read the G513 specifications.

I noticed something interesting: G513 saves custom lights so they can be called through `FN-F6`.
This might explain why it is significantly slower than other models. It might be busy saving the color data.

Which leads me to think, either:
1) it is hardcoded, and there is not much to be done.
2) the saving can be disabled. In this case, Logitech's own software probably does it. The only way to know would be to capture USB frames of the latest LGS talking to a G513 device.

If someone is open to doing this and lives in a juridiction that does not make it illegal, please let me know, I'll help you set up the capture.

msagerup, 2019-04-14:
Hi. I am having issues also with this
<W>render-loop: error on device: device busy re-syncing device

Since I am rather new I am not sure how I could create a new build and apply these builds that you are talking about. Can you provide me with steps on what do do when i Install from the official PPA ?

msagerup, 2019-04-14:
Got it working. 

This is my output: 
<E>render-loop: error on device: device busy, re-syncing device
<E>render-loop: error on device: device busy, re-syncing device
<W>render-loop: increased commit delay to 1000us
<E>render-loop: error on device: device busy, re-syncing device
<W>render-loop: increased commit delay to 2000us
<E>render-loop: error on device: device busy, re-syncing device
<W>render-loop: increased commit delay to 3000us
<E>render-loop: error on device: device busy, re-syncing device
<W>render-loop: increased commit delay to 4000us

spectras, 2019-04-14:
@msagerup Yeah, the issue is fixed on master branch as you saw, but there was no release since.

spectras, 2019-04-21:
The fix is in version 1.1.0 that was just released. Closing the issue :)

closed 2019-04-21 by spectras