keyleds/keyleds #35: Failing to build from the AUR repository - Manjaro

tigerstrikemedia, 2019-03-18:
I'm getting t he following error building this from the AUR repository for Manjaro. Is there another dependency I'm missing?

```
error: missing initializer for member ‘sd_bus_vtable::<unnamed union>::<unnamed struct>::features’ [-Werror=missing-field-initializers]
```

This is during the following step:

```
[ 83%] Building CXX object keyledsd/CMakeFiles/keyledsd.dir/src/service/dbus/DeviceManager.cxx.o
```

spectras, 2019-03-18:
Hello,
Thanks for reporting here. You just  found a bug it seems,  I'm looking into ti.

spectras, 2019-03-19:
Okay, apparently it's a bug in either systemd or gcc on Manjaro.
I was able to work around it like this:

1) Edit `/etc/makepkg.conf`
→ Look for the line that starts with `CXXFLAGS`
→ At the end, before the final quotes, add `-Wno-missing-field-initializers`.
2) Ensure you don't have `gtest` package installed _(it is outdated and I do not check the version so it will break the build, I'll fix this in next release — it's only needed for running the tests anyway)_.
→ `sudo pacman -Rs gtest appimagelauncher`
3) Re-rerun `makepkg`. You might need to clear the `src` directory for it to account for changed settings.

I should perhaps reconsider the policy of using `-Werror` when packaging. It's a good thing when working on the project to ensure code quality, but it's not the first time it breaks the build on some distribution because of some third-party package.

I hope this workaround works for you, please let me know. If it does, I will include it in next release too.

tigerstrikemedia, 2019-03-19:
@spectras 

This has worked for me just had to trash the cache so it all pulled fresh `rm -rf ~/.cache/pacaur/keyleds`

Thanks for the help!

spectras, 2019-03-19:
I'm glad it worked. Thanks for  taking the time to report here and try the workaround.

I'm modifying the build system so next release should be more resistant to inconsistencies in the dependencies I use. That way the workaround will not be needed anymore.

spectras, 2019-04-21:
Fix is now included in v1.1.0, closing the issue.
Thanks again for reporting here!

closed 2019-04-21 by spectras