keyleds/keyleds #36: G910 <C>main: Could not connect to session bus: No such file or directory

zirisdarkskullz, 2019-04-24:
G910 Orion / Manjaro / Kernel 5.0.7-1-MANJARO / XFCE4 / AMD Ryzen Threadripper 2950X @ 3.8 Ghz @ 35 C / GPU: AMD ATI Radeon RX 580 / Mem: 2660MiB / 64363MiB 

I installed Manjaro yesterday. 

┌─[odin@Asgard]─[~]
└──╼ sudo keyledsctl list
┌─[odin@Asgard]─[~]
└──╼ sudo keyledsctl info
More than one device found, use -d device or set KEYLEDS_DEVICE environment variable.
┌─[✗]─[odin@Asgard]─[~]
└──╼ sudo keyledsd -vv
<I>main: keyledsd v1.1.0 starting up
<I>main: using /etc/keyledsd.conf
<I>effect-manager: loading lua from /usr/lib/keyledsd/fx_lua.so
<I>effect-manager: loaded plugin <lua>
<C>main: Could not connect to session bus: No such file or directory
<I>effect-manager: unloaded plugin <lua>
┌─[✗]─[odin@Asgard]─[~]
└──╼ ls /dev/hidraw*
crw-------  1 root root 242, 0 Apr 24  2019 /dev/hidraw0
crw-rw----+ 1 root root 242, 1 Apr 24  2019 /dev/hidraw1
crw-rw-rw-  1 root root 242, 2 Apr 24  2019 /dev/hidraw2
┌─[odin@Asgard]─[~]
└──╼ 

Please help/advise, let me know if you need further information.

spectras, 2019-04-25:
Hello,

Thanks for reporting here. It seems to be a problem with the way systemd is setup.
I see at least to issues here:

1. udev is not granting you access to the keyboard device.
2. dbus service is not available. Could you try those commands?

    ```bash
    echo $DBUS_SESSION_BUS_ADDRESS
    systemctl --user daemon-reload
    systemctl --user status dbus.socket
    systemd-run --user /bin/sleep 1
    systemctl status dbus.socket
    ```
    They will not change anything but their output might help pinpoint the problem.

If you can tell me which Manjaro release you are using, I'll setup a VM to try it myself too.

zirisdarkskullz, 2019-04-25:
This is Manjaro 18.0
Here is the info you requested, with a little bit about my system included:

odin@Asgard 
OS: Manjaro Linux x86_64 
Host: X399 DESIGNARE EX 
Kernel: 5.0.7-1-MANJARO 
Shell: bash 5.0.2 
DE: Xfce 
WM: Xfwm4 
Terminal: xfce4-terminal 
Terminal Font: Monospace 12 
CPU: AMD Ryzen Threadripper 2950X 16- (32) @ 4.1 
GPU: AMD ATI Radeon RX 470/480/570/570X/580/580X 
Memory: 1870MiB / 64363MiB 

odin@Asgard ~ # sudo echo $DBUS_SESSION_BUS_ADDRESS
[sudo] password for odin: 
unix:path=/run/user/1000/bus
odin@Asgard ~ # sudo systemctl --user daemon-reload
Failed to connect to bus: No such file or directory
odin@Asgard ~ # sudo systemctl --user status dbus.socket
Failed to connect to bus: No such file or directory
odin@Asgard ~ # sudo systemd-run --user /bin/sleep 1
Failed to create bus connection: No such file or directory
odin@Asgard ~ # sudo systemctl status dbus.socket
● dbus.socket - D-Bus System Message Bus Socket
   Loaded: loaded (/usr/lib/systemd/system/dbus.socket; static; vendor preset: >
   Active: active (running) since Thu 2019-04-25 15:04:09 CDT; 4h 57min left
   Listen: /run/dbus/system_bus_socket (Stream)
   CGroup: /system.slice/dbus.socket

Apr 25 15:04:09 Asgard systemd[1]: Listening on D-Bus System Message Bus Socket.
odin@Asgard ~ # sudo su
[Asgard odin]# echo $DBUS_SESSION_BUS_ADDRESS

[Asgard odin]# systemctl --user daemon-reload
Failed to connect to bus: No such file or directory
[Asgard odin]# systemctl --user status dbus.socket
Failed to connect to bus: No such file or directory
[Asgard odin]# systemd-run --user /bin/sleep 1
Failed to create bus connection: No such file or directory
[Asgard odin]# systemctl status dbus.socket
● dbus.socket - D-Bus System Message Bus Socket
   Loaded: loaded (/usr/lib/systemd/system/dbus.socket; static; vendor preset: >
   Active: active (running) since Thu 2019-04-25 15:04:09 CDT; 4h 57min left
   Listen: /run/dbus/system_bus_socket (Stream)
   CGroup: /system.slice/dbus.socket

Apr 25 15:04:09 Asgard systemd[1]: Listening on D-Bus System Message Bus Socket.
[Asgard odin]# exit
exit
odin@Asgard ~ #

spectras, 2019-04-25:
Thanks!

I am sorry I did not make myself explicit: those should be run as a regular user, not as root.
Same thing with `keyledsd`, it is intended to be run by the user from his graphical session.
In fact, starting `keyledsd` as root won't work, as (oddly) root will be denied access to the desktop.

Would it please be possible to:

1. Re-run the same commands without `sudo`.
2. If they work, try to start `keyledsd -vv` (without `sudo`).
3. If keyledsd starts but does nothing, try to run
    * `keyledsctl -dv info -d /dev/hidraw0`
    * `keyledsctl -dv info -d /dev/hidraw1`
    * `keyledsctl -dv info -d /dev/hidraw2`
    As regular user first, then if none worked, with `sudo`.

Thanks. I am installing myself a Manjaro 18.0 vm so I can test too.

zirisdarkskullz, 2019-04-25:
No worries. Thank YOU for all the effort! I really appreciate it!

odin@Asgard ~ # echo $DBUS_SESSION_BUS_ADDRESS
unix:path=/run/user/1000/bus
odin@Asgard ~ # systemctl --user daemon-reload
odin@Asgard ~ # systemctl --user status dbus.socket
● dbus.socket - D-Bus User Message Bus Socket
   Loaded: loaded (/usr/lib/systemd/user/dbus.socket; static; vendor preset: enabled)
   Active: active (running) since Thu 2019-04-25 15:04:22 CDT; 4h 14min left
   Listen: /run/user/1000/bus (Stream)
   CGroup: /user.slice/user-1000.slice/user@1000.service/dbus.socket

Apr 25 15:04:22 Asgard systemd[2702]: Starting D-Bus User Message Bus Socket.
Apr 25 15:04:22 Asgard systemd[2702]: Listening on D-Bus User Message Bus Socket.
odin@Asgard ~ # systemd-run --user /bin/sleep 1
Running as unit: run-rce98b1169d5f4a2e8af06b353054ff77.service
odin@Asgard ~ # systemctl status dbus.socket
● dbus.socket - D-Bus System Message Bus Socket
   Loaded: loaded (/usr/lib/systemd/system/dbus.socket; static; vendor preset: disabled)
   Active: active (running) since Thu 2019-04-25 15:04:09 CDT; 4h 14min left
   Listen: /run/dbus/system_bus_socket (Stream)
   CGroup: /system.slice/dbus.socket

Apr 25 15:04:09 Asgard systemd[1]: Listening on D-Bus System Message Bus Socket.
odin@Asgard ~ # keyledsd -vv
\<I>main: keyledsd v1.1.0 starting up
\<I>main: using /etc/keyledsd.conf
\<I>effect-manager: loading lua from /usr/lib/keyledsd/fx_lua.so
\<I>effect-manager: loaded plugin <lua>
\<C>main: Could not reserve name on session bus: File exists
\<I>effect-manager: unloaded plugin <lua>
odin@Asgard ~ # keyledsctl -dv info -d /dev/hidraw0
Cannot open /dev/hidraw0: Permission denied
odin@Asgard ~ # keyledsctl -dv info -d /dev/hidraw1
Cannot open /dev/hidraw1: invalid device (hid++ v1)
odin@Asgard ~ # keyledsctl -dv info -d /dev/hidraw2
Cannot open /dev/hidraw2: invalid device (no hid++ support)
odin@Asgard ~ # keyledsctl -dv info -d /dev/hidraw0
Cannot open /dev/hidraw0: Permission denied
odin@Asgard ~ # sudo keyledsctl -dv info -d /dev/hidraw0
[sudo] password for odin: 
Cannot open /dev/hidraw0: invalid device (no hid++ support)
odin@Asgard ~ # sudo keyledsctl -dv info -d /dev/hidraw1
Cannot open /dev/hidraw1: invalid device (hid++ v1)
odin@Asgard ~ # sudo keyledsctl -dv info -d /dev/hidraw2
Cannot open /dev/hidraw2: invalid device (no hid++ support)
odin@Asgard ~ #

spectras, 2019-04-25:
Okay, let's see…

Well, first some good news:

* your system is not broken :+1: , it was just `sudo` not working with `systemctl --user`.
* `keyledsd` is in fact already loaded, even though is does nothing:
    ```
    <C>main: Could not reserve name on session bus: File exists
    ```
    → This happens because `keyledsd` is already running.
* `keyledsctl` has correct permissions, since it can communicate with the devices.

So everything is working fine, except the device is reporting Logitech protocol v1 only (v2 is required for animation support). That's odd.

You say it's a G910. Perhaps it is old and it needs a firmware upgrade using Logitech Gaming Software?

zirisdarkskullz, 2019-04-25:
Keyboard was purchased from Amazon on February 28, 2019.  It's still pretty new BUT it has never been connected with any Logitech Gaming Software as I don't run any version of Windows on this machine. I do have a very, very old laptop with Win10 on it I could possibly try to hook it up to and download the software to see if it needs updating/upgrading. 

That being said, I'm not really looking for animation, per se, I just want my lights to work. It appears that brightness is currently set to zero. I have no way of turning up the brightness unless we can get this going. Is there a command I'm missing somewhere that needs to be run to turn up the brightness? I didn't see such a thing in the ReadMe or Documentation.  

I know everything works properly because it works fine in LinuxMint, just not in Manjaro. I have been using a different git with it in Mint, but only because I had not yet found your git, which I like better. I plan to switch to this one on the Mint install at some point.  (dual boot Mint & Manjaro) The other git doesn't work *at all* in Manjaro. With yours, I have some hope. Again, yours is lots better.

zirisdarkskullz, 2019-04-25:
Additional info:  

I unplugged the keyboard, getting ready to plug it into my old laptop to see about updates, and just for giggles plugged it back in. To my surprise, all the keys lit up, but they're all blue, I'd prefer them red. At this point, I'm taking what I can get, I can live with blue for a bit, but I'd really prefer red. So I set to changing the colors via keyledsctl and it told me no. See below:

```
odin@Asgard ~ # keyledsctl set-leds all=red
More than one device found, use -d device or set KEYLEDS_DEVICE environment variable.
odin@Asgard ~ # keyledsctl info
More than one device found, use -d device or set KEYLEDS_DEVICE environment variable.
odin@Asgard ~ # keyledsctl list
odin@Asgard ~ # keyledsctl set-leds all=red
More than one device found, use -d device or set KEYLEDS_DEVICE environment variable.
odin@Asgard ~ # keyledsctl info
More than one device found, use -d device or set KEYLEDS_DEVICE environment variable.
odin@Asgard ~ # keyledsctl list
odin@Asgard ~ # keyledsctl get-leds
More than one device found, use -d device or set KEYLEDS_DEVICE environment variable.
odin@Asgard ~ #
```
I'll go ahead and see about the update(s) and let you know how it turns out.

zirisdarkskullz, 2019-04-25:
Logitech has a file specifically for updating the keyboard's firmware. I downloaded it and it told me the firmware is completely up-to-date, no update needed. 
So I also downloaded Logitech Hub, which allows for key customizations. I ran it and was able to set the keys to all red. 
I unplugged from the laptop and back into my desktop. I now have the flowing rainbow effect. My keys are at least lit now, but I still can't set them to all red from within Manjaro; I can't do anything with them from within Manjaro.

I tried the same commands as above and got the exact same results.

zirisdarkskullz, 2019-04-25:
I plugged back into the laptop, set the startup effect to "disabled" closed the software and plugged back into the Desktop. I now have all blue lights again, so at least that annoying, flowing rainbow effect is gone. 
Just for giggles, I tried the commands again, but still got the exact same output. No real changes.

spectras, 2019-04-26:
I'm sorry, life called me ;). I'll read what you wrote tomorrow and hopefully figure out what's going on :)

spectras, 2019-04-26:
Okay, so here is what I understand so far:

* Your keyboard is new and up to date :+1:
* `keyledsctl` can access the /dev/hidraw devices, but believes none is a valid keyboard.

Additional info:

* Keyboard turning all blue is just the default color when no software talks to it and no starting effect is defined.

I would like to try to find what exactly is happening when you plug your keyboard.
Could you do this?

1. Unplug your keyboard
2. Plug it again
3. Run `dmesg`. You should see something like at the bottom of the output:
    ```
    [263759.806035] usb 1-8: Product: Gaming Keyboard G410
    [263759.806037] usb 1-8: Manufacturer: Logitech
    [263759.806040] usb 1-8: SerialNumber: 0123456789AB
    [263759.808441] input: Logitech Gaming Keyboard G410 as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:046D:C330.000E/input/input47
    [263759.864819] hid-generic 0003:046D:C330.000E: input,hidraw3: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G410] on usb-0000:00:14.0-8/input0
    [263759.866303] input: Logitech Gaming Keyboard G410 Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:046D:C330.000F/input/input48
    [263759.924878] input: Logitech Gaming Keyboard G410 Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:046D:C330.000F/input/input49
    [263759.925134] hid-generic 0003:046D:C330.000F: input,hiddev2,hidraw4: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G410] on usb-0000:00:14.0-8/input1
    ```
    The key information here is `hidraw3`, `hidraw4`.
4. Then run<br>
    `sudo keyledsctl -dddv info -d /dev/hidrawX`<br>
    `sudo keyledsctl -dddv info -d /dev/hidrawY`<br>
    …replacing **hidrawX** and **hidrawY** with the numbers you noted in step 3.

    Hopefully, this should generate quite a lot of output, and the key to understanding what doesn't work should be in there.

zirisdarkskullz, 2019-04-26:
Here is my output after unplugging/replugging in my keyboard:
```

██████████████████  ████████   odin@Asgard 
██████████████████  ████████   ----------- 
██████████████████  ████████   OS: Manjaro Linux x86_64 
██████████████████  ████████   Host: X399 DESIGNARE EX 
████████            ████████   Kernel: 5.0.7-1-MANJARO 
████████  ████████  ████████   Uptime: 1 min 
████████  ████████  ████████   Packages: 1200 (pacman) 
████████  ████████  ████████   Shell: bash 5.0.2 
████████  ████████  ████████   Resolution: 2560x1440, 1920x1080 
████████  ████████  ████████   DE: Xfce 
████████  ████████  ████████   WM: Xfwm4 
████████  ████████  ████████   WM Theme: Plasma Fire 
████████  ████████  ████████   Theme: Plasma Fire [GTK2], Adwaita [GTK3] 
████████  ████████  ████████   Icons: OSDark [GTK2], Adwaita [GTK3] 
                               Terminal: xfce4-terminal 
                               Terminal Font: Monospace 12 
                               CPU: AMD Ryzen Threadripper 2950X 16- (32) @ 3.3 
                               GPU: AMD ATI Radeon RX 470/480/570/570X/580/580X 
                               Memory: 2492MiB / 64363MiB 

                                                       

 08:08:36 up 1 min,  1 user,  load average: 1.79, 0.61, 0.22
┌─[odin@Asgard]─[~]
└──╼ dmesg
[    0.000000] Linux version 5.0.7-1-MANJARO (builduser@development) (gcc version 8.2.1 20181127 (GCC)) #1 SMP PREEMPT Sat Apr 6 08:50:27 UTC 2019
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.0-x86_64 root=UUID=e3d824bf-8bc9-4619-b07a-d6b1bcd97be6 rw quiet
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009deffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009df0000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x000000000affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000b000000-0x000000000b01ffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000b020000-0x00000000782fffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000078300000-0x000000007850cfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007850d000-0x00000000785f5fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000785f6000-0x0000000078a0ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000078a10000-0x00000000798e5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000798e6000-0x0000000079979fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007997a000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007c000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b7a00000-0x00000000b7a00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b7b00000-0x00000000b7b7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ef800000-0x00000000ef8fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000efc00000-0x00000000efc00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000efd00000-0x00000000efd7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000feafffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec30000-0x00000000fec30fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc0000-0x00000000fedc0fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc2000-0x00000000fedcffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedd4000-0x00000000fedd5fff] reserved
[... 1455 lines cut ...]
Cannot open /dev/hidraw1: invalid device (hid++ v1)
┌─[✗]─[odin@Asgard]─[~]
└──╼ sudo keyledsctl -dddv info -d /dev/hidraw2
libkeyleds: Opening device /dev/hidraw2
libkeyleds: Parsing report descriptor (65 bytes)
libkeyleds: invalid device (no hid++ support)
Cannot open /dev/hidraw2: invalid device (no hid++ support)
┌─[✗]─[odin@Asgard]─[~]
└──╼ sudo keyledsctl -dddv info -d /dev/hidraw3
libkeyleds: Opening device /dev/hidraw3
libkeyleds: No such file or directory
Cannot open /dev/hidraw3: No such file or directory
┌─[✗]─[odin@Asgard]─[~]
└──╼ sudo keyledsctl -dddv info -d /dev/hidraw1
libkeyleds: Opening device /dev/hidraw1
libkeyleds: Parsing report descriptor (122 bytes)
libkeyleds: Found report ID 0x10 (6 bytes)
libkeyleds: Found report ID 0x11 (19 bytes)
libkeyleds: invalid device (hid++ v1)
Cannot open /dev/hidraw1: invalid device (hid++ v1)
┌─[✗]─[odin@Asgard]─[~]
└──╼ 

```
I hope this helps! When I got the error, I just went ahead and tried to run the same thing with the other hidraws, just to see if it would make any difference but it didn't seem to. I've also included the full outputs just to be thorough. And, as you can see in the uptime at the top, this was directly after booting up into Manjaro; I was in LinuxMint over night.

spectras, 2019-04-26:
Thanks!

Okay, something odd is definitely going on.
```
[89.424036] hid-generic: input,hiddev2,hidraw3: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910]
```
Yet immediately after:

```
libkeyleds: Opening device /dev/hidraw3
libkeyleds: No such file or directory
```
Something unrelated to keyleds is deleting the device between the time your system detects it and you enter the `keyledsctl` command. This is very strange.

Do you have some other software that might try to grab the device for some reason?

spectras, 2019-04-26:
I am running out of ideas, especially as this happens *before keyleds is even started*.

Perhaps `sudo cat /sys/kernel/debug/usb/devices` would help, it sometimes has useful info.

zirisdarkskullz, 2019-04-26:
Ok, I updated my BIOS just now. After reading through my initial dmesg output, I saw my current BIOS version and wondered if there's an update for it. (I agree, this is very odd.) I checked and there *is* an update available. I downloaded it, rebooted and installed it. I then had to reconfigure most of my BIOS options. After doing that, I booted back into Manjaro and noted that my lights work without me doing anything with them now, although they're blue. (which is fine for now, at least I get *some* light over none at all. I ran dmesg and your following commands above and got basically the same output. So I unplugged/replugged again, then ran dmesg again. I got some expected output from dmesg, but unfortunately the same output from your following commands. So I ran the sudo cat /sys/kernel....... command and noted my keyboard there. I'll paste the relevant info, but not the whole dmesg again, a lot of that seem irrelevant now. 

```
 09:35:10 up 0 min,  1 user,  load average: 2.18, 0.59, 0.20
┌─[odin@Asgard]─[~]
└──╼ dmesg
...   
[    6.790739] iwlwifi 0000:05:00.0: Detected Intel(R) Dual Band Wireless AC 8265, REV=0x230
[    6.814589] kvm: disabled by bios
[    6.817408] MCE: In-kernel MCE decoding enabled.
...   
[    2.099996] ata8: SATA max UDMA/133 abar m131072@0xba380000 port 0xba380480 irq 121
[    2.100066] xhci_hcd 0000:09:00.3: xHCI Host Controller
[    2.100070] xhci_hcd 0000:09:00.3: new USB bus registered, assigned bus number 3
[    2.100164] xhci_hcd 0000:09:00.3: hcc params 0x0270f665 hci version 0x100 quirks 0x0000000000000410
[    2.100475] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.00
[    2.100476] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.100477] usb usb3: Product: xHCI Host Controller
[    2.100478] usb usb3: Manufacturer: Linux 5.0.7-1-MANJARO xhci-hcd
[    2.100478] usb usb3: SerialNumber: 0000:09:00.3
[    2.100551] hub 3-0:1.0: USB hub found
[    2.100556] hub 3-0:1.0: 4 ports detected
[    2.100651] xhci_hcd 0000:09:00.3: xHCI Host Controller
[    2.100655] xhci_hcd 0000:09:00.3: new USB bus registered, assigned bus number 4
[    2.100657] xhci_hcd 0000:09:00.3: Host supports USB 3.0  SuperSpeed
[    2.100667] usb usb4: We don't know the algorithms for LPM for this host, disabling LPM.
[    2.100682] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.00
[    2.100683] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.100684] usb usb4: Product: xHCI Host Controller
[    2.100685] usb usb4: Manufacturer: Linux 5.0.7-1-MANJARO xhci-hcd
[    2.100685] usb usb4: SerialNumber: ████████
[    2.100739] hub 4-0:1.0: USB hub found
[    2.100744] hub 4-0:1.0: 4 ports detected
[    2.100881] ahci 0000:0a:00.2: AHCI 0001.0301 32 slots 1 ports 6 Gbps 0x1 impl SATA mode
[    2.100882] ahci 0000:0a:00.2: flags: 64bit ncq sntf ilck pm led clo only pmp fbs pio slum part 
[    2.100991] xhci_hcd 0000:42:00.3: xHCI Host Controller
[    2.100997] xhci_hcd 0000:42:00.3: new USB bus registered, assigned bus number 5
[    2.101031] scsi host8: ahci
[    2.101086] ata9: SATA max UDMA/133 abar m4096@0xba708000 port 0xba708100 irq 132
[    2.101092] xhci_hcd 0000:42:00.3: hcc params 0x0270f665 hci version 0x100 quirks 0x0000000000000410
[    2.101414] usb usb5: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.00
[    2.101416] usb usb5: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.101417] usb usb5: Product: xHCI Host Controller
[    2.101419] usb usb5: Manufacturer: Linux 5.0.7-1-MANJARO xhci-hcd
[    2.101420] usb usb5: SerialNumber: ████████
[    2.101500] hub 5-0:1.0: USB hub found
[    2.101507] hub 5-0:1.0: 4 ports detected
[    2.101976] xhci_hcd 0000:42:00.3: xHCI Host Controller
[    2.101979] xhci_hcd 0000:42:00.3: new USB bus registered, assigned bus number 6
[    2.101981] xhci_hcd 0000:42:00.3: Host supports USB 3.0  SuperSpeed
[    2.101993] usb usb6: We don't know the algorithms for LPM for this host, disabling LPM.
[    2.102010] usb usb6: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.00
[    2.102012] usb usb6: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.102013] usb usb6: Product: xHCI Host Controller
[    2.102014] usb usb6: Manufacturer: Linux 5.0.7-1-MANJARO xhci-hcd
[    2.102015] usb usb6: SerialNumber: ████████
[    2.102080] hub 6-0:1.0: USB hub found
[    2.102086] hub 6-0:1.0: 4 ports detected
[    2.413564] ata9: SATA link down (SStatus 0 SControl 300)
[    2.426729] usb 1-7: new full-speed USB device number 2 using xhci_hcd
[    2.430050] usb 3-1: new high-speed USB device number 2 using xhci_hcd
[    2.430058] usb 5-3: new full-speed USB device number 2 using xhci_hcd
[    2.570065] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    2.584373] ata1.00: ATA-10: WDC WD6001FZWX-00A2VA0, 01.01A01, max UDMA/133
[    2.584375] ata1.00: 11721045168 sectors, multi 16: LBA48 NCQ (depth 32), AA
[    2.585137] usb 3-1: New USB device found, idVendor=04f9, idProduct=0328, bcdDevice= 1.00
[    2.585140] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.585141] usb 3-1: Product: DCP-L2540DW
[    2.585143] usb 3-1: Manufacturer: Brother
[    2.585144] usb 3-1: SerialNumber: U63885F6N571358
[... 187 lines cut ...]
S:  SerialNumber=████████
C:* #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr= 98mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
[... cut ...]
I:* If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs
E:  Ad=82(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
┌─[odin@Asgard]─[~]
└──╼ 
```

Ok, I ran a few other commands as well, just to give a full story here, post BIOS update. 

I do have several softwares installed relative to the keyboard right now, I'll go see about removing them, and maybe try again. I installed so many because I was trying them one-by-one to see which one works and/or will do something with my keyboard, but none have, yet.

zirisdarkskullz, 2019-04-26:
Alright! Some new output after uninstalling the other softwares. 
```
┌─[odin@Asgard]─[~/Downloads]
└──╼ sudo keyledsctl -dddv info -d /dev/hidraw2
libkeyleds: Opening device /dev/hidraw2
libkeyleds: Parsing report descriptor (65 bytes)
libkeyleds: invalid device (no hid++ support)
Cannot open /dev/hidraw2: invalid device (no hid++ support)
┌─[✗]─[odin@Asgard]─[~/Downloads]
└──╼ ls /dev/hidraw*
crw-------  1 root root 242, 0 Apr 26 09:34 /dev/hidraw0
crw-rw----+ 1 root root 242, 1 Apr 26 09:34 /dev/hidraw1
crw-rw-rw-  1 root root 242, 2 Apr 26 09:41 /dev/hidraw2
┌─[odin@Asgard]─[~/Downloads]
└──╼ keyledsd list
<C>main: Could not reserve name on session bus: File exists
┌─[✗]─[odin@Asgard]─[~/Downloads]
└──╼ keyledsd info
<C>main: Could not reserve name on session bus: File exists
┌─[✗]─[odin@Asgard]─[~/Downloads]
└──╼ 
```

I'm still not able to really do anything with it, yet, but we have made progress. 

Also, I want to take a moment to say THANK YOU VERY MUCH for all of your assistance! You really are an awesome Developer and the community as a whole appreciates you and everything you've done. Keep up the great work, Sir.

spectras, 2019-04-26:
No worries, I would really like to find out what is wrong :) And I think I start to understand what is going on.

This line here is interesting:
```
I:* If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs
                                                              ^^^^^
```
The "usbfs" part at the end means that some software running on your system claimed the keyboard control interface for its exclusive use. This is why /dev/hidraw3 disappears: that software is removing it.

1. Could you try to find what it is and kill it?
2. And then we will try to rebind the correct driver. We run this as root (use `sudo bash` to get a root prompt). If you did not reboot / unplug your keyboard, it should be:
    ```
    echo -n "5-4:1.1" > /sys/bus/usb/drivers/usbfs/unbind
    echo -n "5-4:1.1" > /sys/bus/usb/drivers/usbhid/bind
    ```
    If you did reboot or unplug your keyboard we need to find the correct address. See at bottom of this message[*].
3. Redo `cat /sys/kernel/debug/usb/devices`. Now the line that had `usbfs` should have `usbhid` instead. If not, then the software that grabs the device is still active.<br>If you do see `usbhid`, then proceed to 4.
4. `ls /dev/hidraw*` should show a new hidraw device (should be hidraw3).
5. Try `sudo keyledsctl -dv info -d /dev/hidrawX` (replacing X with the new hidraw device).

-----
[*] So how to find out address if it changed:
 * Run `dmesg |grep 046D:C335`
 * Look for line that reads `[1...] input: Logitech Gaming Keyboard G910 Consumer Control as /devices/pci0000:40/0000:40:07.1/0000:42:00.3/usb5/5-4/5-4:1.1/0003:046D:C335.0006/input/input30`
 * The address is the `x-x:x.x` part in the middle of the long path.

zirisdarkskullz, 2019-04-26:
Just for giggles, after removing the software, I rebooted. After I got back up, I noted that I have an "effect" happening on my keyboard. (More Progress!)
I don't know what the effect is called, so I'll just describe it as best I can.   
The keys are blue, but wave across from 10-key pad to caps lock with darkness, while the keys themselves turn white when I press them. 
I tried the same few commands as my previous post and got the same outputs, until I tried something a little different on a whim:

```
 10:26:02 up 3 min,  1 user,  load average: 1.10, 0.88, 0.38
┌─[odin@Asgard]─[~]
└──╼ keyledsd list
<C>main: Could not reserve name on session bus: File exists
┌─[✗]─[odin@Asgard]─[~]
└──╼ keyledsd info
<C>main: Could not reserve name on session bus: File exists
┌─[✗]─[odin@Asgard]─[~]
└──╼ sudo keyledsctl -dddv info -d /dev/hidraw2
[sudo] password for odin: 
libkeyleds: Opening device /dev/hidraw2
libkeyleds: Parsing report descriptor (65 bytes)
libkeyleds: invalid device (no hid++ support)
Cannot open /dev/hidraw2: invalid device (no hid++ support)
┌─[✗]─[odin@Asgard]─[~]
└──╼ sudo keyledsctl -dddv info -d /dev/hidraw1
libkeyleds: Opening device /dev/hidraw1
libkeyleds: Parsing report descriptor (122 bytes)
libkeyleds: Found report ID 0x10 (6 bytes)
libkeyleds: Found report ID 0x11 (19 bytes)
libkeyleds: invalid device (hid++ v1)
Cannot open /dev/hidraw1: invalid device (hid++ v1)
┌─[✗]─[odin@Asgard]─[~]
└──╼ sudo keyledsctl -dddv info -d /dev/hidraw0
libkeyleds: Opening device /dev/hidraw0
libkeyleds: Parsing report descriptor (67 bytes)
libkeyleds: invalid device (no hid++ support)
Cannot open /dev/hidraw0: invalid device (no hid++ support)
┌─[✗]─[odin@Asgard]─[~]
└──╼ sudo keyledsctl -dddv info -d /dev/hidraw3
libkeyleds: Opening device /dev/hidraw3
libkeyleds: Parsing report descriptor (119 bytes)
libkeyleds: Found report ID 0x11 (19 bytes)
libkeyleds: Found report ID 0x12 (63 bytes)
libkeyleds: Opened device /dev/hidraw3 protocol version 4
libkeyleds: feature 0005 is at 4 [00]
Name:           G910 Orion Spectrum
Type:           keyboard
libkeyleds: feature 0003 is at 2 [00]
Model:          c33500000000
Serial:         31384714
Firmware[c335]: application U1  v106.2.7 [active]
Firmware[aac5]: bootloader BOT v114.0.7
libkeyleds: feature 4522 is at 3 [00]
libkeyleds: feature 1e00 is at 5 [40]
libkeyleds: feature 4540 is at 6 [00]
libkeyleds: feature 1eb0 is at 7 [60]
libkeyleds: feature 8010 is at 8 [00]
libkeyleds: feature 8020 is at 9 [00]
libkeyleds: feature 8030 is at 10 [00]
libkeyleds: feature 8060 is at 11 [00]
libkeyleds: feature 00c1 is at 12 [00]
libkeyleds: feature 1801 is at 13 [60]
libkeyleds: feature 1802 is at 14 [60]
libkeyleds: feature 8080 is at 15 [00]
libkeyleds: feature 8070 is at 16 [00]
libkeyleds: feature 1821 is at 17 [00]
Features:       [0001, 0003, 4522, 0005, 1e00, 4540, 1eb0, 8010, 8020, 8030, 8060, 00c1, 1801, 1802, 8080, 8070, 1821]
Known features: feature version gamemode name layout2 gkeys mkeys mrkeys reportrate dfu-control leds led-effects
G-keys: 9
Report rates:   [1ms] 2ms 4ms 8ms
LED block[01]:  104 keys, max_rgb(255, 255, 255)
LED block[02]:    5 keys, max_rgb(1, 0, 0)
LED block[04]:    9 keys, max_rgb(255, 255, 255)
[... 121 lines cut ...]
└──╼ keyledsctl info
Name:           G910 Orion Spectrum
Type:           keyboard
Model:          c33500000000
Serial:         31384714
Firmware[c335]: application U1  v106.2.7 [active]
Firmware[aac5]: bootloader BOT v114.0.7
Features:       [0001, 0003, 4522, 0005, 1e00, 4540, 1eb0, 8010, 8020, 8030, 8060, 00c1, 1801, 1802, 8080, 8070, 1821]
Known features: feature version gamemode name layout2 gkeys mkeys mrkeys reportrate dfu-control leds led-effects
G-keys: 9
Report rates:   [1ms] 2ms 4ms 8ms
LED block[01]:  104 keys, max_rgb(255, 255, 255)
LED block[02]:    5 keys, max_rgb(1, 0, 0)
LED block[04]:    9 keys, max_rgb(255, 255, 255)
LED block[10]:    2 keys, max_rgb(255, 255, 255)
┌─[odin@Asgard]─[~]
└──╼ keyledsctl list
/dev/hidraw3 046d:c335 [148738733532]
┌─[odin@Asgard]─[~]
└──╼ keyledsctl set-leds -d 148738733532 all=red
┌─[odin@Asgard]─[~]
└──╼ 
```
It appears to be working now. I guess the other software was holding the keyboard. I apologize for this inconvenience. I had no idea they would conflict like that. 
So how do I disable the wave effect and make it just a solid color all the time?

spectras, 2019-04-26:
> The keys are blue, but wave across from 10-key pad to caps lock with darkness, while the keys themselves turn white when I press them.

That's keyleds' default effect!

spectras, 2019-04-26:
So there it is, it was the other software grabbing the keyboard and not letting go!
For the record, what's that other software, so I can go straight for it should someone else have the same issue?

So, yes, you can follow the [guide on the wiki](https://github.com/keyleds/keyleds/wiki/Configuration-File) to configure keyleds.

You can skip the part "finding the config file". Just do:
```
cp /etc/keyledsd.conf $HOME/.config/
killall keyledsd
keyledsd -vv&
```
Then you can edit `$HOME/.config/keyledsd.conf` following the guide. You can do many things such as having different colors depending on application. If you have any question or need help, feel free to ask them.

zirisdarkskullz, 2019-04-26:
Ok, we're SO close. I ran the following commands: 

```
┌─[odin@Asgard]─[~]
└──╼ killall keyledsd
┌─[odin@Asgard]─[~]
└──╼ keyledsctl
Usage: keyledsctl [-dqv] help|list|info|gkeys|get-leds|set-leds|gamemode
┌─[✗]─[odin@Asgard]─[~]
└──╼ keyledsctl set-leds -d 148738733532 all=red
┌─[odin@Asgard]─[~]
└──╼ 

```

All of the major keys turned red, but the G-Keys, the big G at the top left and the "G910" led at the wrist are all still blue. Is there any way to change those as well? 

The other softwares were [this one](https://github.com/JSubelj/g910-gkey-macro-support) and [this one](https://github.com/MohamadSaada/LogiGSK). I had them both installed, but the latter one's service was not running and I had to start it to get it uninstalled. Moreover, the author misspelled the command to uninstall his software in his documentation, so before I noticed the typo, it just opened the software. After I noticed it, it looked like he did it on purpose to prevent you from uninstalling his product, even though that's probably not the case, it just *looked* that way. 

One other question, if I may: is there some way within your software to assign anything to the G-Keys, like make G1 also press "a" for example. If so, is there also a way to make a G-Key do something like ctrl-v?

spectras, 2019-04-26:
**Is there any way to change those as well?**

Yes. `keyledsctl` is intended as a low-level tool, so it exposes what Logitech calls key blocks.
By default `all=red` applies to normal key block. You can change switch block using `-b`. For instance:
```
keyledsctl set-leds -b media all=red
```
Possible blocks are `keys` (normal keys), `media`, `gkeys`, `logo` and `modes`. Available blocks depend on specific models.

**Is there some way within your software to assign anything to the G-Keys?**
Alas no, not yet. It's part of the planned features, but I have no ETA yet. The way I see it, I want to bring proper support even if it means taking more time.

If you need it asap, I would suggest using https://github.com/JSubelj/g910-gkey-macro-support like this:
1) make sure neither `g910-gkey-macro-support` nor `keyledsd` starts automatically.
2) run your `keyledsctl` commands to change the colors to your liking.
3) run  `g910-gkey-macro-support` after.

And stay tuned. When GKeys support eventually comes to keyleds, it should bring the whole functionality set, like switching macros with Mkeys and the like. This does come with a few challenges though, especially making it work together with animations, that's what it's taking a bit of time (also I don't have a g910 myself to test, which slows me down a bit - that'll come eventually).

zirisdarkskullz, 2019-04-26:
Cool! 
I found the whack-a-mole and set it to red. It's a pretty neat effect. I might just stick with it. Idk, yet, I might play around until I find something else I like. For now, though, you rock. Thanks a TON. This is awesome. 
It was the G910-GKey-Macro-Support, I think, that was conflicting with your software, since the service for the other one wasn't running at all. 
I'm very glad you're working on this. I look forward to updates in the future, but until then, keep being a Rock Star!

closed 2020-06-11 by spectras