keyleds/keyleds #37: Leds feature not found on G213

nrittsti, 2019-05-02:
Hello spectras,

first of all, thank you so much for supporting this awesome project.
I bought this week a new G213 and run into this issue.
To be on the safe side, i checked the device with Logitech Gamecenter, it works so far.

```
keyledsctl set-leds all=yellow
keyledsctl: set led block 01 -- feature not found on device
```

` keyledsctl info` does not show the **leds** feature

```
$ keyledsctl info
Name:           Gaming Keyboard G213
Type:           keyboard
Model:          c33600000000
Serial:         32384705
Firmware[c336]: application U1  v109.0.6 [active]
Firmware[aac8]: bootloader BOT v131.0.2
Features:       [0001, 0003, 4522, 0005, 1e00, 4540, 1eb0, 8060, 00c2, 1801, 1802, 8070, 1821, 1bc0, 1806]
Known features: feature version gamemode name layout2 reportrate led-effects
Report rates:   [1ms] 2ms 4ms 8ms
```
Maybe i missed something, any ideas?

Distro: Manjaro Linux
Kernel: 5.0.9-2-MANJARO x86_64 bits: 64 
Compiler: gcc v: 8.3.0 
Keyleds: 1.1.0 (https://aur.archlinux.org/packages/keyleds/)
Systemd: 242.16

```
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 05e3:0612 Genesys Logic, Inc. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 046d:c336 Logitech, Inc. 
Bus 001 Device 006: ID 147e:2016 Upek Biometric Touchchip/Touchstrip Fingerprint Sensor
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 1532:005c Razer USA, Ltd 
Bus 001 Device 002: ID 045e:0719 Microsoft Corp. Xbox 360 Wireless Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ sudo lsusb -v -s 005
Bus 001 Device 005: ID 046d:c336 Logitech, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc336 
  bcdDevice            9.00
  iManufacturer           1 Logitech
  iProduct                2 Gaming Keyboard G213
  iSerial                 3 056F38743330
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 U109.00_B0006
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      65
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     119
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

spectras, 2019-05-03:
Hello,

Thanks for reporting here. Well, that looks odd.
That feature list here:
```
[0001, 0003, 4522, 0005, 1e00, 4540, 1eb0, 8060, 00c2, 1801, 1802, 8070, 1821, 1bc0, 1806]
````
Well, as per-key RGB is 8080, indeed it's missing.
That's really odd.

And the firmware on the keyboard looks recent enough. In fact it's even newer than mine.
With Logitech game software, you are positive that you can manually change specific keys (not just keyboard-wide effect)?

nrittsti, 2019-05-04:
> With Logitech game software, you are positive that you can manually change specific keys (not just keyboard-wide effect)?

I couldn't find a configuration option for setting keygroups colors.
I can only set a keyboard-wide colors or activate some effects like breathing, waving and color cycling.

I also tried another project called G213Colors. (https://github.com/SebiTimeWaster/G213Colors)
G213Colors lets me change the color for 5 keygroups.

`sudo python G213Colors.py -c FFFFFF FF0000 00FF00 FFFFFF FFFFFF`

Well, for whatever reason, it works.

spectras, 2019-05-04:
Hmmm if Logitech gaming software does not offer you to change keys one by one, then it does not detect the feature either. That's odd, I though it could do that. I guess you can try to check for a firmware update through Logitech Gaming Software.

If it is not per-key RGB capable, then keyledsd cannot support it in its current state. This could definitely be added in a future release, but it requires a rework of some key parts if the main on-device feature used by keyleds is missing.

nrittsti, 2019-05-05:
Okay, thanks for your effort.
Unfortunately my firmware is up to date.
I'll let you know if the function is reactivated in the future.

pearsonk, 2019-06-06:
The G213 isn't per-key-RGB. It is simply 5 "region" LEDs. I implemented using a simple replay from a usbcap, but a more deterministic method would be much-appreciated, since replay is very model-specific.

spectras, 2019-06-07:
@pearsonk That's nice to know.
Well, if you have those replays, I'm starting to know enough about Logitech's proprietary protocol to make sense of it :)

txtsd, 2019-10-31:
I'm looking to buy this keyboard. What kind of additional information do you need to be able to support [Lightsync](https://www.logitechg.com/en-us/innovation/lightsync-rgb.html)?

There are 2 other keyboards that have this system: G910 and G513.