keyleds/keyleds #38: "Wait" variable causing crash

AllOgreTheWorld, 2019-05-08:
Attempting to create a flashing effect for a key when it is pressed. I have pasted the code for hello.lua to test this issue. When the "wait" variable is used in an onKeyEvent, I recieve an error stating: 

attempt to yield across C-call boundary
stack traceback:
	[C]: in function 'wait'
        [string "KF2Breathe"]:19: in function <[string "KF2Breathe"]:16>

When "wait" is called from the 'hello' function, no error occurs. When called form the onKeyEvent function, it crashes. I will post my code below. (changed to TXT file to allow for attachment)
[KF2Breathe.txt](https://github.com/keyleds/keyleds/files/3155440/KF2Breathe.txt))

spectras, 2019-05-08:
Hello,
Thanks for taking the time to report here.

Yes, two things combine here:

* `wait` may only be called from a function started with `thread`.
* `onKeyEvent` is called whenever a key is pressed (so, not with `thread`).

The typical approach would be to put you animation in a separate function, and have `onKeyEvent` start that function with `thread`. Something like this should do:

```lua
transparent = tocolor(0, 0, 0, 0)
red = tocolor('red')
white = tocolor('white')

function keypressAnimation(keyboard, key)
    keyboard[key] = red
    wait(1)
    keyboard[key] = white
    wait(1)
    keyboard[key] = transparent
end

function onKeyEvent(key, isPress)
    if not isPress then return end
    thread(keypressAnimation, keyboard, key)
end

keyboard = RenderTarget:new()
function render(ms, target) target:blend(keyboard) end
```— I did not test it, please tell me how it goes. :)

AllOgreTheWorld, 2019-05-09:
That worked perfectly, thank you so much!

Now a question, if I wanted to set this to work ONLY with a specific key (I'm trying this on 'Q'), how would I go about creating an if statement that would do that? I have previously attempted this:
`function onKeyEvent(key, isPress)
    if not isPress then return end
    if keyboard.Q == isPress then 
        thread(keypressAnimation, keyboard)
    end
end`
However this produces no effect. I know isPress should take a boolean value, and that is likely what is causing the problem. But how would I be able to check for just one specific key?

spectras, 2019-05-10:
The idea is good, but you want to test whether `key` is `keyboard.Q`.
`isPress` is just a boolean, it's `true` if the event is the key being pressed down, `false` if the event is the key being released.

AllOgreTheWorld, 2019-05-11:
Ok so I currently have it set like this: 
```
function onKeyEvent(key, isPress)
    if not isPress then return end
    if keyboard.key == Q then
        thread(keypressAnimation, keyboard)
    end
end
```
However it still registers any key to trigger the animation, not just Q. I know this is probably something simple but its just sailing right past my head!  =P

spectras, 2019-05-13:
Hello again!
Sorry for the delay, I was on weekend.

You are making it more complicated than it is ;-)
This should work:
```lua
function onKeyEvent(key, isPress)
    if not isPress then return end
    if key == keyboard.Q then thread(keypressAnimation, keyboard) end
end
```
or
```lua
function onKeyEvent(key, isPress)
    if not isPress then return end
    if key.name == "Q" then thread(keypressAnimation, keyboard) end
end
```
It's my fault, I should have provided the full example. I hope this helps :)

AllOgreTheWorld, 2019-05-14:
My dude, you are the absolute best. This is working exactly as I hoped it would.

Thank you for making such an incredible program and providing such wonderful support for it.

closed 2019-05-14 by AllOgreTheWorld