keyleds/keyleds #40: keyledsd crashing unknowningly

VirtualTorus, 2019-08-31:
Hi, 

I'm using the Logitech G810 Orion keyboard and running keyledsd with `keyledsd -v --no-dbus &`. Long after (or sometimes shortly after) starting the process, keyledsd crashes and I'm not sure what's causing it to crash. Here is a stack trace:
```
Aug 30 14:59:30 virtualtorus systemd-coredump[1912]: Process 777 (keyledsd) of user 1000 dumped core.
                                                
                                                Stack trace of thread 777:
                                                #0  0x00007fef7c40d755 raise (libc.so.6)
                                                #1  0x00007fef7c3f8851 abort (libc.so.6)
                                                #2  0x00007fef7c64781f _ZN9__gnu_cxx27__verbose_terminate_handlerEv (libstdc++.so.6)
                                                #3  0x00007fef7c65430a _ZN10__cxxabiv111__terminateEPFvvE (libstdc++.so.6)
                                                #4  0x00007fef7c654367 _ZSt9terminatev (libstdc++.so.6)
                                                #5  0x00007fef7c6545bd __cxa_throw (libstdc++.so.6)
                                                #6  0x000056476bde06e1 n/a (keyledsd)
                                                #7  0x000056476bde0c5d n/a (keyledsd)
                                                #8  0x000056476bdc6075 n/a (keyledsd)
                                                #9  0x000056476bdc63dd n/a (keyledsd)
                                                #10 0x000056476bdc66bc n/a (keyledsd)
                                                #11 0x000056476bddf8ed n/a (keyledsd)
                                                #12 0x000056476bdc51f4 n/a (keyledsd)
                                                #13 0x00007fef7ca046a8 uv__io_poll (libuv.so.1)
                                                #14 0x00007fef7c9f4d47 uv_run (libuv.so.1)
                                                #15 0x000056476bdb585a n/a (keyledsd)
                                                #16 0x00007fef7c3f9ee3 __libc_start_main (libc.so.6)
                                                #17 0x000056476bdb5c0e _start (keyledsd)
                                                
                                                Stack trace of thread 780:
                                                #0  0x00007fef7c9d7145 __nanosleep (libpthread.so.0)
                                                #1  0x000056476bddddb4 n/a (keyledsd)
                                                #2  0x000056476bddd147 n/a (keyledsd)
                                                #3  0x00007fef7c680cd4 execute_native_thread_routine (libstdc++.so.6)
                                                #4  0x00007fef7c9cd57f start_thread (libpthread.so.0)
                                                #5  0x00007fef7c4cf0e3 __clone (libc.so.6)
-- Subject: Process 777 (keyledsd) dumped core
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- Documentation: man:core(5)
-- 
-- Process 777 (keyledsd) crashed and dumped core.
```
I'm not sure how to go about debugging this or what's going on, any ideas? 

I've been manually running the command `keyledsd -D -vv &> keyledsd.log &` to try to capture anything before a crash. Oddly enough, no crashes occurred (yet?) when I run keyledsd in this way.

spectras, 2019-08-31:
Hello,

Thanks for reporting here! Could you please tell me:
- what version you are running?
- what distribution you are running it on?
- whether it is a packaged version or a manually-compiler one?

As it is compiled without debugging information, I need to get the exact same binary to make sense of the trace.

If you are curious, here is what I understand as of now:
- Thread 777 is the controlling thread. It is crashing because of some unhandled exception.
- Thread 780 is the rendering thread for your keyboard. It is idle at the time of the crash.

VirtualTorus, 2019-08-31:
I'm running version 1.1.0 of keyleds on Arch Linux. I'm using a packaged version, as provided from the AUR [here](https://aur.archlinux.org/packages/keyleds/).

spectras, 2019-08-31:
Aw, arch linux. I has supporting this one because everyone has a different binary, that makes reproducing issues difficult.

Do you think you could upload the binary, `/usr/bin/keyledsd` somewhere? Or the build package, I'll get the binary from it.

I hope I can match it with the stack trace and figure out where it failed. If that does not work, we will need to enable debugging on your package, that's more annoying on your side, so let's keep that if my first idea fails :)

VirtualTorus, 2019-09-01:
Sure thing, [here's the binary](https://send.firefox.com/download/869fdbd18dd6391a/#U_PKLzBN9rAY4OAcAjLyoQ).

I'm sure it's not a big deal for me to enable debugging mode, but let me know how the first part goes. If it helps, you can take a look at the Arch [build script](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=keyleds) that was used in making the keyleds package.

spectras, 2019-09-01:
Thanks!
It should be enough, I could match the stack trace to actual code using the binary. So I know where the issue arises. The X server is returning some unexpected value to keyleds, and it aborts by throwing an exception.

If you are curious, here is how the stack trace maps:<details><summary>Click to unfold</summary>
```
#6  0x000056476bde06e1 n/a (keyledsd)   456e1: xlib::Window::getProperty, line 208
#7  0x000056476bde0c5d n/a (keyledsd)   45c5d: xlib::Window::name, line 172
#8  0x000056476bdc6075 n/a (keyledsd)   2b075: XContextWatcher::setContext, line 125
#9  0x000056476bdc63dd n/a (keyledsd)   2b3dd: XContextWatcher::onActiveWindowChanged, line 104
#10 0x000056476bdc66bc n/a (keyledsd)   2b6bc: XContextWatcher::handleEvent, line 60
#11 0x000056476bddf8ed n/a (keyledsd)   448ed: std::function::operator()
#12 0x000056476bdc51f4 n/a (keyledsd)   2a1f4: xlib::Display::processEvents(), line 113
#13 0x00007fef7ca046a8 uv__io_poll (libuv.so.1)
#14 0x00007fef7c9f4d47 uv_run (libuv.so.1)
#15 0x000056476bdb585a n/a (keyledsd)   main()
```
The #6 throws the exception, items #1 to #5 are just how it unfolds under the hood.</details>

I have to look into what could cause an invalid value to be received at that point. Do you happen to know what versions of Xlib and X server (Xorg / wayland / ...) you are running?

spectras, 2019-09-01:
I might have a fix. Are you comfortable with the idea of recompiling after applying a 1-line change?

If so, could you go to `keyledsd/src/tools/XWindow.cxx`, line 195 and change it so it looks like this?
```cpp
        actualType != type ||           // modify this one
        actualFormat == 0) {            // insert this one
        return std::string();           // unchanged
```

I believe this should be enough to fix the issue. Please let me know whether it works if you try it, or if you don't feel like patching manually — in that case I will push the patch and repackage. It might take a few days though, as I am starting a new job tomorrow so things are likely to get busy.

VirtualTorus, 2019-09-01:
> I have to look into what could cause an invalid value to be received at that point. Do you happen to know what versions of Xlib and X server (Xorg / wayland / ...) you are running?

I'm running Xlib v1.6.8 and Xorg server v1.20.5. 

> I might have a fix. Are you comfortable with the idea of recompiling after applying a 1-line change?
> 
> If so, could you go to `keyledsd/src/tools/XWindow.cxx`, line 195 and change it so it looks like this?
> 
> ```c++
>         actualType != type ||           // modify this one
>         actualFormat == 0) {            // insert this one
>         return std::string();           // unchanged
> ```
> 
> I believe this should be enough to fix the issue. Please let me know whether it works if you try it, or if you don't feel like patching manually — in that case I will push the patch and repackage. It might take a few days though, as I am starting a new job tomorrow so things are likely to get busy.

Yeah I'll patch it later today, rebuild, and give it a spin. Take your time.

VirtualTorus, 2019-09-06:
I believe the patch fixed it :-) haven't had a crash since I last commented here.

spectras, 2019-09-15:
Thanks a lot for the feedback. I will make it into a proper patch and include it in next release then.

spectras, 2020-06-11:
Sorry for the long delay. Since I started my new position, I have very little time for side projects.
(hey it was on September 2nd, that was right in those days :))

I am planning to work on the backlog of fixes and publish a bugfix release this weekend.

commit 5ebb7f29 (2020-06-11): Check for invalid actualFormat returned by XGetWindowProperty. Fixes #40.
closed 2020-06-11 by spectras