keyleds/keyleds #42: keyledsd does not start in wayland (Fedora 30)

domrim, 2019-10-14:
Hey,

I installed keyleds via the repository running fedora 30.
In fedora 30 wayland is default and i want to stick with it (multiple reasons).

Under wayland the keyleds deamon seems to fail to start. I've debugged it with my small knowledge about x and wayland but have no clue what exactly is the problem.
Output of `keyledsd -vvv`:
```
<I>main: keyledsd v1.1.0 starting up
<I>main: using /home/dominik/.config/keyledsd.conf
<I>effect-manager: loading lua from /usr/lib64/keyledsd/fx_lua.so
<I>effect-manager: loaded plugin <lua>
<N>main: connected to display :0
X Error of failed request:  BadAtom (invalid Atom parameter)
  Major opcode of failed request:  131 (XInputExtension)
  Minor opcode of failed request:  59 ()
  Atom id in failed request:  0x0
  Serial number of failed request:  21
  Current serial number in output stream:  21
```
Output of `sudo keyledsd -vvv`
```
<I>main: keyledsd v1.1.0 starting up
<I>main: using /etc/keyledsd.conf
<I>effect-manager: loading lua from /usr/lib64/keyledsd/fx_lua.so
<I>effect-manager: loaded plugin <lua>
<C>main: Could not connect to session bus: No such file or directory
<I>effect-manager: unloaded plugin <lua>
```

Is the problematic known? Is wayland yet not supported?
If you need more data to debug/analyse the problems just tell me what data you need ;)

spectras, 2019-10-14:
Hello!
Thanks for reporting here. Well, I think Wayland is not very widespread those days, which explains why that was not discovered sooner.
I will investigate the issue this week, though perhaps only on next weekend as my new job keeps me pretty busy :)

Also, please note keykedsd is intended to run within user session, that's why starting it with sudo doesn't work out of the box, in that second capture :)

Thanks again for taking the time to report the issue!

domrim, 2019-10-14:
Thanks for the fast reply.

> Also, please note keykedsd is intended to run within user session, that's why starting it with sudo doesn't work out of the box, in that second capture :)

Oh, i did miss this hint in the doc. Now i read it ;)

> Well, I think Wayland is not very widespread those days, which explains why that was not discovered sooner.

Now that Debian 10 also defaults to wayland, i think it will spread more and more with ubuntu and so on.

spectras, 2019-10-14:
I had a quick look just so I can think about it during the week.

It seems it's much deeper than I thought. The very feature keyledsd uses, listening to key events to enable dynamic feedback, is not supported by Wayland.

Apparently, it's on purpose, to remove the ability to write keyloggers. Unfortunately, this also prevents software such as keyleds that need key events for legitimate purposes.

It's all news to me, there might be things I missed, but it does not look good.

domrim, 2019-10-15:
AFAIK wayland has still support for libinput and xinput.
The more problematic factor is that we can't get the context-changes as before.
So we don't know which window is active.

spectras, 2019-10-15:
I just had a look, wayland explicitly does not support the features of Xinput that keyleds needs.
In fact, this is what we see in the error message you posted, the Xinput part of Xwayland tells keyleds that it does not support the requested property.

I have to get in touch with wayland developers about keyleds use case, but I have been looking at the mailing list archives, and every other project getting in touch on this got rejected without much consideration (tools to setup global hotkeys mostly). They were told global hotkeys should be handled by the compositor, and their project should be discontinued.

I happen to know some people from KDE whom I talked to shortly, and they said it is unlikely I would see support from KWin/plasma or other compositors for that matter, because animating the keyboard is too far from the role of a compositor.

So the current situation is not only keyleds cannot run under wayland, but **no other program** can implement such a feature under wayland. Except the compositor itself.
Unless something is changed, gaming keyboards will remain Xorg-only (and possibly MSWindows-only if development of Xorg stops).

It's bad timing, I have a lot of work atm, I am not sure I have time to spend on getting the case of keyleds heard by wayland developers.

domrim, 2019-10-17:
okay. so i would not close this issue, but mark it as future project to enhance keyleds.

I think i will be in February at FOSDEM in Brussels and try to talk as well as with the wayland people, also with the gnome people (and other devs of compositors) if they have ideas / plans to help the matter.

in my opinion the lack of support of some widely spread keyboard-manufacturers is still an issue for some people who are thinking about changing to gnu/linux. So hopefully there will be an solution one day.

domrim, 2019-10-18:
addition:

key-event based effects still don't work as far as i can see it. (this won't change soon under wayland i believe).
But lighting-effects and so on are working.

spectras, 2019-11-05:
Hello again!
Sorry for the long delay, as I feared work is ramping up and I have less time to spend on most personal stuff, including keyleds.

I have been watching the wayland mailing list though and I saw that wayland still seems very young, when it comes to governance. They are working on formalizing a way to submit protocol extension proposals, and this might be the opening that's needed for keyleds and similar tools: submitting and championing a protocol extension that makes it possible to request keyboard events (that way the compositor could ask the user and accept or deny the request safely).

It is a nice workaround you found to at least let it start in the meantime. But I believe switching profiles automatically based on current window shouldn't work either with Wayland. I guess we'll have to wait and see whether and what comes out on this topic, AFAIK at the moment there is nothing for that either. I know that most global keyboard tools had to be discontinued (or merged into their matching compositor for those that were part of a full desktop environment).

The whole situation is a bit Kafkaesque. At this point, there are enough programs this breaks and enough users having their workflow disrupted that I am ready to bet I could build a "wayland-keyserver" project that would run as root and just let any program connect from localhost to access the keyboard, and it would instantly have a sizable community, destroying all the security wayland developers aimed to provide.
Security over functionality has never worked. Security is good, but it has to come on top of functionality, not at its expense.</end-of-rant>