keyleds/keyleds #44: G815 Support

kamushadenes, 2019-12-24:
Hello,

I just purchased a G815 keyboard and I was wondering what would be necessary for this project to support it. It detects the keyboard but any attempt to get or set led state fails.

What info could be useful?

### keyledsctl info

`Name:           G815 RGB MECHANICAL GAMING KEYBOARD
Type:           keyboard
Model:          c33f00000000
Serial:         31384702
Firmware[aaea]: bootloader BOT v184.0.3
Firmware[c33f]: application U1  v131.0.16 [active]
Firmware[0000]: application  v100.0.0
Firmware[c33f]: (null) EXT v100.0.0
Firmware[c33f]: (null) EXT v100.0.0 [active]
Features:       [0001, 0003, 0005, 0020, 00c2, 00d0, 1bc0, 4522, 4540, 8010, 8020, 8030, 8040, 8060, 8071, 8081, 8100, 1801, 1802, 1806, 18b0, 1e00, 1eb0]
Known features: feature version name gamemode layout2 gkeys mkeys mrkeys reportrate
G-keys: 5
Report rates:   [1ms] 2ms 4ms 8ms`

### lspci

`Bus 001 Device 010: ID 046d:c33f Logitech, Inc.`

 

The errors I'm receiving are:

### keyledsctl get-leds

`Fetching led info failed: feature not found on device`

### keyledsctl set-leds all=yellow enter=green f1=ff00cd f2=ff00cd

`keyledsctl: set led block 01 -- feature not found on device
keyledsctl: set leds -- feature not found on device`

Thanks in advance.

spectras, 2019-12-24:
Hello and thanks for reporting here.
Well, from the output you pasted it seems the per-key RGB feature is missing from this model.

If Logitech software for Windows can actually perform per-key RGB on this device, then it means they created a newer protocol to that effect. If it is the case, it needs to be reverse-engineered and re-implemented for keyleds.

The requirements are basically having in the same place a G815, a Windows box (possibly a VM) and some time. I will check on the web whether people have been posting captures already, otherwise I will either need someone with a G815 to do some, or I will have to get a G815 here to do it myself.

Those days are pretty busy for me, don't expect it to come tomorrow, but I'll definitely try to get this working.

yawor, 2020-03-07:
Hi,
I'm buying G915 (I'm waiting for confirmation from three shops regarding keys layout as I'm looking for a proper US international layout with a wide, single row enter key version), which is basically wireless version of G815, so probably the protocol is going to be the same.
Both G815 and G915 have per-key RGB control with an ability to save three presets in the keyboard itself.
I should be able to capture the USB traffic from/to the keyboard. If I remember correctly, I can run a Windows guest in VirtualBox and capture USB traffic on Linux host using Wireshark.

yawor, 2020-03-13:
I'm playing around with G915, doing some captures and comparing my results with work being done on supporting G815 by g810-leds project. They've got G815 somehow working, but G915 not.
What I found out is that the G915 has three interfaces in USB descriptor, each containing one HID endpoint:
interface 0 - endpoint 0x81 with keyboard usage
interface 1 - endpoint 0x82 with mouse usage (used for multimedia keys)
interface 2 - endpoint 0x83 with vendor usage

Output from dmesg:
```
[210243.692543] input: Logitech USB Receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:046D:C541.00E5/input/input270
[210243.749010] hid-generic 0003:046D:C541.00E5: input,hidraw7: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-0000:00:14.0-1/input0
[210243.749763] input: Logitech USB Receiver Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:046D:C541.00E6/input/input271
[210243.749926] input: Logitech USB Receiver Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:046D:C541.00E6/input/input272
[210243.805589] input: Logitech USB Receiver System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:046D:C541.00E6/input/input273
[210243.805721] hid-generic 0003:046D:C541.00E6: input,hidraw8: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-1/input1
[210243.806348] hid-generic 0003:046D:C541.00E7: hiddev3,hidraw9: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-1/input2
```

To correctly open and send commands to the keyboard, the last interface must be used, so in this case it's /dev/hidraw9 device.

There are at least 2 different commands to set LED colours
1. Set up to 4 LEDs, each with its own color:
```
11 01 0b 1d [AA] [RR] [GG] [BB] [AA] [RR] [GG] [BB] [AA] [RR] [GG] [BB] [AA] [RR] [GG] [BB]
```
where AA is a led address and following it is the RGB value for that LED.
If less than 4 LEDs are being set, then first non-used AA value is set to 0xFF and the rest of the data is filled with 0x00. For example:
- set 4 multimedia keys to color `1400ff`:
```11 01 0b 1d 9d 14 00 ff 9e 14 00 ff 9b 14 00 ff 9c 14 00 ff```
- set backlight level button to color `707070`:
```11 01 0b 1d 99 70 70 70 ff 00 00 00 00 00 00 00 00 00 00 00```

Keyboard responds with
```
11 01 0b 1d [AA] [AA] [AA] [AA] 00 00 00 00 00 00 00 00 00 00 00 00
```
where AA are addresses of LEDs being set (positions not used are set to 0xFF).

2. Set up to 13 LEDs to the same color:
```
11 01 0b 6d [RR] [GG] [BB] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA]
```
where first 3 bytes are RGB value, followed by 13 LED addresses (if less, then unused positions are set to 0x00).

Keyboard responds with:
```
11 01 0b 6d [RR] [GG] [BB] [AA] 00 00 00 00 00 00 00 00 00 00 00 00
```
where RGB is the same value as set in the command and AA is an address of the first LED in the command (I wonder if this is some bug - my keyboard is running on newest firmware).

3. Commit color change:
```
11 01 0b 7d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
Keyboard responds with the same data

tJouve, 2021-08-19:
G815 Keyboard is well support by https://github.com/MatMoul/g810-led, maybe source code could be usefull for reverse-engineering

yawor, 2021-08-19:
@tJouve there's no need to copy g810-led solutions. All the bits needed are already described in the #47 issue (for G915 actually, but everything should be true for G815 too, except for the wireless receiver part).
The main issue is that, according to github, @spectras has not been active since October last year. I hope he's well.

mcasaccio, 2022-03-01:
Is there an implementation for g815?

closed 2022-04-02 by kamushadenes