keyleds/keyleds #45: Switch off keys when no used for certain period of time

jtyr, 2020-01-08:
It's known issue that the LEDs on keyboard  (especially blue color LEDs - my own experience) can break over the time. I was wondering if `keyledsd` could switch the keys off (set black color, or dim to minimum brightness) when not used for certain period of time (or when screensaver starts)? If this is not currently possible, adding such feature would definitely be more than appreciated.

spectras, 2020-01-08:
Hello!
Well I am not sure that it really increases the lifetime of the leds, but there is a dim effect shipped with keyleds. The documentation can be found on the wiki, here: https://github.com/keyleds/keyleds/wiki/Idle-dim-plugin

To have it completely turn off the lights, simply use black as the effect color :)

jtyr, 2020-01-09:
Thanks for your prompt response. I have tested it and it works just fine!

closed 2020-01-09 by jtyr