keyleds/keyleds #46: Changes for gcc10 needed

domrim, 2020-03-06:
I tried today to build keyleds for fedora 32 (which will be released in April), and the build failed because of following error: 
```
...
[ 95%] Building CXX object keyledsd/CMakeFiles/keyledsd.dir/src/tools/FileWatcher.cxx.o
[ 96%] Building CXX object keyledsd/CMakeFiles/keyledsd.dir/src/tools/XContextWatcher.cxx.o
In file included from /builddir/build/BUILD/keyleds-1.1.0/keyledsd/src/tools/FileWatcher.cxx:17:
/builddir/build/BUILD/keyleds-1.1.0/keyledsd/include/keyledsd/tools/FileWatcher.h:54:75: error: 'string' is not a member of 'std'
   54 |     using Listener = std::function<void(Event mask, uint32_t cookie, std::string path)>;
      |                                                                           ^~~~~~
/builddir/build/BUILD/keyleds-1.1.0/keyledsd/include/keyledsd/tools/FileWatcher.h:23:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
   22 | #include <sys/inotify.h>
  +++ |+#include <string>
   23 | #include <vector>
```
after some research i saw, that f32 uses gcc10 and on [gcc porting to 10 page](https://gcc.gnu.org/gcc-10/porting_to.html) were some important infos:
> Some C++ Standard Library headers have been changed to no longer include the <stdexcept> header. As such, C++ programs that used components defined in <stdexcept> or <string> without explicitly including the right headers will no longer compile.
> Previously components such as std::runtime_error, std::string and std::allocator were implicitly defined after including unrelated headers such as <array> and <optional>. Correct code should include the appropriate headers for the classes being used.

So it would be nice if the headers get included soon :)

spectras, 2020-03-16:
Hello,
Good spotting. That's indeed something I try to get right, but there was no way to verify up to that point. So it's not a surprise some slipped through.

hwoarang, 2020-04-28:
@spectras some build logs can be found on your Fedora rawhide builds on your OBS repo https://build.opensuse.org/package/live_build_log/home:spectras/keyleds/Fedora_Rawhide/x86_64
It would be nice to have this fixed now that Fedora 32 is also available :)

spectras, 2020-06-14:
Fixed and released :)

closed 2020-06-14 by spectras