keyleds/keyleds #47: G915 support

yawor, 2020-03-13:
I'm playing around with G915, doing some captures and comparing my results with work being done on supporting G815 by g810-leds project. They've got G815 somehow working, but G915 not.
What I found out is that the G915 has three interfaces in USB descriptor, each containing one HID endpoint:
interface 0 - endpoint 0x81 with keyboard usage
interface 1 - endpoint 0x82 with mouse usage (used for multimedia keys)
interface 2 - endpoint 0x83 with vendor usage

Output from dmesg:
```
[210243.692543] input: Logitech USB Receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:046D:C541.00E5/input/input270
[210243.749010] hid-generic 0003:046D:C541.00E5: input,hidraw7: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-0000:00:14.0-1/input0
[210243.749763] input: Logitech USB Receiver Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:046D:C541.00E6/input/input271
[210243.749926] input: Logitech USB Receiver Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:046D:C541.00E6/input/input272
[210243.805589] input: Logitech USB Receiver System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:046D:C541.00E6/input/input273
[210243.805721] hid-generic 0003:046D:C541.00E6: input,hidraw8: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-1/input1
[210243.806348] hid-generic 0003:046D:C541.00E7: hiddev3,hidraw9: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-1/input2
```

To correctly open and send commands to the keyboard, the last interface must be used, so in this case it's /dev/hidraw9 device.

There are at least 2 different commands to set LED colours
1. Set up to 4 LEDs, each with its own color:
```
11 01 0b 1d [AA] [RR] [GG] [BB] [AA] [RR] [GG] [BB] [AA] [RR] [GG] [BB] [AA] [RR] [GG] [BB]
```
where AA is a led address and following it is the RGB value for that LED.
If less than 4 LEDs are being set, then first non-used AA value is set to 0xFF and the rest of the data is filled with 0x00. For example:
- set 4 multimedia keys to color `1400ff`:
```11 01 0b 1d 9d 14 00 ff 9e 14 00 ff 9b 14 00 ff 9c 14 00 ff```
- set backlight level button to color `707070`:
```11 01 0b 1d 99 70 70 70 ff 00 00 00 00 00 00 00 00 00 00 00```

Keyboard responds with
```
11 01 0b 1d [AA] [AA] [AA] [AA] 00 00 00 00 00 00 00 00 00 00 00 00
```
where AA are addresses of LEDs being set (positions not used are set to 0xFF).

2. Set up to 13 LEDs to the same color:
```
11 01 0b 6d [RR] [GG] [BB] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA]
```
where first 3 bytes are RGB value, followed by 13 LED addresses (if less, then unused positions are set to 0x00).

Keyboard responds with:
```
11 01 0b 6d [RR] [GG] [BB] [AA] 00 00 00 00 00 00 00 00 00 00 00 00
```
where RGB is the same value as set in the command and AA is an address of the first LED in the command (I wonder if this is some bug - my keyboard is running on newest firmware).

3. Commit color change:
```
11 01 0b 7d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
Keyboard responds with the same data

_Originally posted by @yawor in https://github.com/keyleds/keyleds/issues/44#issuecomment-598771604_

yawor, 2020-03-13:
I'm going through the keyleds code in debugger while trying to talk to the G915. I get an error that this is an invalid device (hid++ v1). And indeed, in `keyleds_get_protocol` function, I get an error response from the keyboard with value 0x8f, which is interpreted as v1 device. But I think this would be really strange if the newest Logitech keyboard on the market used an old protocol.
I'm going through the capture from the moment of attaching keyboard to Windows virtual machine with G-Hub software and I don't see any requests with the same parameters as `keyleds_get_protocol` does (target = 0xff, feature idx = 0, function = 1).

yawor, 2020-03-13:
If I change `KEYLEDS_TARGET_DEFAULT` to 0x01 (instead original 0xff), I can run `./keyleds info` without any problems. I've tested different values and the keyboard responds only to this value.

Like in #44, keyleds detects all (or most) features except LED support. The feature id for LED in G915 (maybe in G815 too) is not 0x8080, but 0x8081. When keyboard is asked for feature index with 0x8081, it responds with index 0x0b, which is then used when LEDs are being set. By changing feature id the feature is detected, but doesn't work (which was expected, as probably the LED protocol itself is different now).

yawor, 2020-03-14:
The G915 can also work in wired mode when connected using USB cable (the dongle is unplugged).

In wired mode it accepts any target_id value - probably the value is simply ignored as it has no purpose outside of wireless mode, where there can be multiple devices on a single dongle (for example in case of Unifying receiver).

Other differences between wired and wireless modes:
- The USB PID changes: dongle has C541, the keyboard itself has C33E. For comparison, G815 has PID C33F.
- In wired mode the keyboard properly reports its serial number. In wireless mode the dongle doesn't report any serial numbers. This makes it harder for keyleds to select correct keyboard. I also have a Unifying receiver plugged in and it sometimes pops up in the `keyledsctl list` and causes "multiple devices" error. But I can't set the serial number using `-d` option or env variable, because the dongle doesn't have one.

yawor, 2020-03-14:
It seems that LED v2 feature uses different LED addressing. I did few dirty changes to the code which allowed me to use `keyleds set-leds` subcommand with G915, but the mapping is totally off.
I've written a small Python program, which allowed me to quickly map the LED addresses to specific keys. Here's the resulting map:

```
01 - A
02 - B
03 - C
04 - D
05 - E
06 - F
07 - G
08 - H
09 - I
0a - J
0b - K
0c - L
0d - M
0e - N
0f - O
10 - P
11 - Q
12 - R
13 - S
14 - T
15 - U
16 - V
17 - W
18 - X
19 - Y
1a - Z
1b - 1
1c - 2
1d - 3
1e - 4
1f - 5
20 - 6
21 - 7
22 - 8
23 - 9
24 - 0
25 - Enter
26 - Esc
27 - Backspace
28 - Tab
29 - Space
2a - -
2b - =
2c - [
2d - ]
2f - \ (near enter)
30 - ;
31 - '
32 - `
33 - ,
34 - .
35 - /
36 - Caps
37 - F1
38 - F2
39 - F3
3a - F4
3b - F5
3c - F6
3d - F7
3e - F8
3f - F9
40 - F10
41 - F11
42 - F12
43 - PRT SCR
44 - SCROLL LOCK
45 - PAUSE
46 - Insert
47 - Home
48 - Page Up
49 - Delete
4a - End
4b - Page Down
4c - Right Arrow
4d - Left Arrow
4e - Down Arrow
4f - Up Arrow
50 - Num Lock (numpad)
51 - / (numpad)
52 - * (numpad)
53 - - (numpad)
54 - + (numpad)
55 - Enter (numpad)
56 - 1 (numpad)
57 - 2 (numpad)
58 - 3 (numpad)
59 - 4 (numpad)
5a - 5 (numpad)
5b - 6 (numpad)
5c - 7 (numpad)
5d - 8 (numpad)
5e - 9 (numpad)
5f - 0 (numpad)
60 - . (numpad)
62 - Context menu
68 - Left Control
69 - Left Shift
6a - Left Alt
6b - Left Win
6c - Right Control
6d - Right Shift
6e - Right Alt
6f - Right Win
99 - Backlight brightness button
9b - Play/Pause
9c - Mute
9d - Next song
9e - Prev song
b4 - G1
b5 - G2
b6 - G3
b7 - G4
b8 - G5
d2 - G Logo
```

yawor, 2020-03-14:
From what I can see, there's no way to download current LED values from the keyboard. And the LEDs v2 doesn't have a blocks feature - everything is set individually.

I've been playing around in Python, doing some tests and I've found out few ways to set LED values:

**Feature 0x8081 (LED v2)**

Address - 1 byte LED address
RGB - 3 bytes with RGB value

Function 1
Sets up to 4 LEDs to specfied colors for each LED.

Request data:
`[Address] [RGB] [Address] [RGB] [Address] [RGB] [Address] [RGB]`

Function 2
Sets 5 consecutive LEDs to colors specified in a sequence, starting with Start Address.

Request data:
`[Start Address] [RGB] [RGB] [RGB] [RGB] [RGB]`

Function 5
Sets up to 3 ranges of LEDs to specified colors for each range.

Request data:
`[Start Address] [End Address] [RGB] [Start Address] [End Address] [RGB] [Start Address] [End Address] [RGB]`

Function 6
Sets up to 13 individual LEDs to a specified color.

Request Data:
`[RGB] [Addr] [Addr] ... [Addr]`

Function 7
Commit

**Feature 0x8071 (RGB Effects)**

Function 7
Reads or sets power save settings.

Read current settings:
Request data set to all 0x00

Response data:
`00 00 00 00 [Low power mode timeout (2 bytes, big endian)] [Turn of backlight timeout (2 bytes, big endian)]`

Set new settings:
Request data has the same format as response data when reading, but first byte is set to 0x01.

yawor, 2020-03-16:
I think none of the libkeyleds functions should take target_id as an argument. This should be a field in `keyleds_device` struct and it should be set during opening a device.
I could prepare the PR for this, but this is quite big change in the API so I would like to first get some confirmation from @spectras if such change is acceptable.

spectras, 2020-03-16:
Wow this is a lot of work you did over the weekend.

Yes, libkeyleds is still considered internal library, though I try not to break the API without compelling reasons. But I never had a unifying-receiver device to play with, so I did not create a mental model of how the target id should fit in. The additional information you posted there is great and definitely is a good reason to update the API.

This also means that to make it fully usable, there should probably be some way to list devices attached to a unifying receiver?

I had to admit, as much as I tried to give the project a sound architecture, there is one major gap left open: adding an abstraction layer between the main service code and the lower-level implementation, so implementation can be swapped at runtime depending on device capabilities.
With 815/915 having a different protocol, this will have to happen soon.

Thanks a lot for sharing the result of your work, it is very much appreciated :)

yawor, 2020-03-16:
@spectras the protocol itself seems to be the same, as it's still HID++. Only LED control is different as it these two new features (0x8071 and 0x8081) replaced old ones.

Regarding the Unifying receiver, maybe I've mixed things too much. The G915 doesn't use the Unifying receiver (I don't think any Logitech gaming keyboard does). It uses LightSpeed receiver, which is not RF compatible with Unifying. I've mentioned it because the HID++ side seems to be similar in regards to the target_id field.

The LightSpeed right now doesn't have ability to pair more devices to it like Unifying, so at least now there's no need to retrieve a list of devices. It's hard to tell what Logitech will bring in the future though. Maybe they'll release some non-gaming keyboards with LED controls for Unifying for example.

I've checked the Solaar code to check how they deal with Unifying and they basically check all the 6 target ids, as Unifying has ability to pair with up to 6 devices. The ids are always from 1 to 6 for devices and 0xff for the receiver itself and they are constant until some device is paired or unpaired.

Also it seems that both LightSpeed and Unifying receivers themselves use HID++ v1, but the devices can use newer protocols (G915 uses HID++ v4).

Regarding detecting which target_id to use, I wonder if wired keyboards even check this value at all. When I connect G915 by cable instead of LightSpeed dongle, it accepts any target_id value (or rather it ignores it?). I don't have access to any other Logitech gaming keyboard so I can't check if this is true for them too. If they also ignore this value, then maybe changing the default to 1 would be enough for now. If not, then the library could use the fact, that the receiver is HID++ v1 and after trying to get protocol version with target_id set to 0xFF it could try with 0x01. Also maybe some option to manually select target_id would be nice.

yawor, 2020-03-16:
One idea regarding abstraction layer would be to have function pointers in keyleds_device struct for different operations (for example set_leds). The pointers would be NULL by default, which would mean no support.
Upon opening a device, a list of features could be retrieved and if an implementation for specific feature is in the library, it could attach its own functions as pointers to the device struct. Of course the functions for the same function pointer need to have the same specification, so set_leds from old LED feature needs to take the same args as set_leds from the new LED feature.

yawor, 2020-03-16:
Here's an example of what I've meant in previous comment:

```C
struct keyleds_operations {
    // NAME
    bool (*get_device_name)(Keyleds * device, uint8_t target_id,
            /*@out@*/ char ** out);
    void (*free_device_name)(/*@only@*/ /*@out@*/ char *);
    bool (*get_device_type)(Keyleds * device, uint8_t target_id,
            /*@out@*/ keyleds_device_type_t * out);
    // VERSION
    bool (*get_device_version)(Keyleds * device, uint8_t target_id,
            /*@out@*/ struct keyleds_device_version ** out);
    void (*free_device_version)(/*@only@*/ /*@out@*/ struct keyleds_device_version *);
    // GAME MODE
    bool (*gamemode_max)(Keyleds * device, uint8_t target_id, /*@out@*/ unsigned * nb);
    bool (*gamemode_set)(Keyleds * device, uint8_t target_id,
                              const uint8_t * ids, unsigned ids_nb);    /* add some keys */
    bool (*gamemode_clear)(Keyleds * device, uint8_t target_id,
                                const uint8_t * ids, unsigned ids_nb);  /* remove some keys */
    bool (*gamemode_reset)(Keyleds * device, uint8_t target_id);   /* remove all keys */
    // LEDS
    bool (*get_block_info)(Keyleds * device, uint8_t target_id,
            /*@out@*/ struct keyleds_keyblocks_info ** out);
    void (*free_block_info)(/*@only@*/ /*@out@*/ struct keyleds_keyblocks_info * info);
    bool (*get_leds)(Keyleds * device, uint8_t target_id, keyleds_block_id_t block_id,
                          struct keyleds_key_color * keys, uint16_t offset, unsigned keys_nb);
    bool (*set_leds)(Keyleds * device, uint8_t target_id, keyleds_block_id_t block_id,
                          const struct keyleds_key_color * keys, unsigned keys_nb);
    bool (*set_led_block)(Keyleds * device, uint8_t target_id, keyleds_block_id_t block_id,
                               uint8_t red, uint8_t green, uint8_t blue);
    bool (*commit_leds)(Keyleds * device, uint8_t target_id);
};
```

I've created here a new struct instead of keeping it in keyleds_device struct, I don't know which way would be better. I've just copied function headers from current code and changed them to pointers. The library would need to keep some kind of list of all implemented features. Foe example it could extend `keyleds_feature_names` from `strings.c` by adding a function pointer to each item in list: `void (*get_operations)(struct keyleds_operations *operations)` which would set up function pointers for that feature.
I don't know if this is good or bad approach. As mentioned, C is not my main programming language, so maybe there are better ways to do this.

spectras, 2020-03-22:
Hello,
Sorry for not being more present. Lot of work, and current situation does not make things any better.

So, yes, I understood well, I just answered in an ambiguous way. By different protocol, I was thinking about the new 8071/8081 features, not the transmission protocol. So if I read you correctly, it would seem the lightspeed receiver behaves like a unifying receiver, with only one device attached, hardcoded to target id 1.

That also matches the behavior you discovered before, that keyleds detected the g915 as HID++v1: it's actually talking to the receiver. So as a first thought, at this point I believe the options, in order of cleanliness, would be like this:
- Cleanly detect a HID++v1 receiver. Talk to it to enumerate attached devices.
- If that's not possible, cleanly detect a HID++v1 receiver. Then try targets 1-6 in turn as part of device enumeration.
- If that's not possible, hardcode a target to 1. But that's only a last resort as this is the less future-proof approach. The day Logitech enables other devices to talk to lightspeed receiver it no longer works.

Yes, for abstraction this is the kind of thing I would like to see. Except perhaps it could be done at the C++ level rather than inside the lib itself:
- The library would simply expose functions for features 8071 and 8081.
- On the C++ level, there would be multiple adapters for `LogitechDevice`. They would use both use the library, but different functions.

This way we let C++ deal with dynamic dispatching through normal virtual inheritance. That's probably better than reinventing the wheel on C side.

yawor, 2020-03-25:
I only know C++ basics, so I won't be able to help except for testing and providing any needed information regarding the keyboard and data exchanges between it and a PC.

intractabilis, 2020-07-31:
I am proficient in C++. How can I help? Can you give me an initial direction?

yawor, 2020-09-17:
I've played around a little today: https://github.com/yawor/keyleds/tree/g915.
For now I've removed target_id argument from all keyleds_* functions and added target_id member in `keyleds_device` struct. I think it belongs there as definitely a concrete device needs to have a specified target_id.
The target_id needs to be passed to `keyleds_open`.
I've also added 0x8071 and 0x8081 feature definitions, but without implementation for now. I think I should be able to implement them on the libkeyleds side, but I'm not sure about keyledsd as I don't know C++ that good.
Also right now I'm still passing `KEYLEDS_TARGET_DEFAULT` to `keyleds_open` in keyledsctl and keyledsd, so there's no target_id detection.

yawor, 2020-09-18:
I've committed a basic LEDS2 feature implementation in my repo. Right now to test it, it's required to change `KEYLEDS_TARGET_DEFAULT` from 0xff to 0x01 in keyleds.h so the cli tool even detects the keyboard.
Also to actually change some colours, you need to change calls in keyledsctl (keyledsctl_set_leds.c) from `keyleds_set_leds` and `keyleds_commit_leds` to `keyleds_set_leds2` and `keyleds_commit_leds2`. The `keyleds_set_leds2` takes one argument less, as LEDS2 feature doesn't have a notion of blocks (just remove block_id argument).
Setting colours work but the LED addresses have changed in the new API, so for example setting colour for "b" changes "e" key. It's going to need a new translation table. Few posts above there's a list of all LED addresses for G915 US layout.

The old LEDS API uses blocks everywhere, so the keyleds is built around that. This is probably going to be worked around somehow or some layer of emulating key blocks is going to be needed to keep backward compatibility. Emulating blocks actually should be easy to do as there's a function in the new API, that allows setting up to 3 colours for 3 different ranges of keys at once.

There are no functions to get current colours from the keyboard anymore. I can see in the code that there are some usages of `keyleds_get_leds` in both keyledsctl and keyledsd, but I didn't analyse further than that if it's really needed.

@spectras if you could take a look if this is going in the right direction and let me know, I would gladly create a work in progress PR from my repo to work on this further.

Rudgas, 2020-11-19:
Any news on this?