keyleds/keyleds #52: Python not really working

Donald4444, 2020-04-24:
First what I want to do:
I want to be able to set every single key color with python, to be able to make a script instead of using the daemon. (I will autostart the script in background and it will completly control the keybord, for example with time or cpu load.)
The Problem is that I can't get the Keycodes:
I can change the color of every key of a key group but no single.
When I try to do dev.leds['keys'].get_all() I get
`>>> dev.leds['keys'].get_all()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "keyleds.pyx", line 401, in keyleds.KeyBlock.get_all
AttributeError: 'dict' object has no attribute 'keycode'`

And when I try to guess a code it again says there is no attribute keycode:
`>>> dev.leds['keys'].set_keys([pykeyleds.KeyColor(keycode=1, key_id=4, color=pykeyleds.Color(255,0,0))])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "keyleds.pyx", line 421, in keyleds.KeyBlock.set_keys
AttributeError: 'dict' object has no attribute 'keycode'`

I would be really happy, if this could be fixed or somebody could help me, if I did something wrong.

Donald4444, 2020-04-25:
Some output, that could help figuring it out:
`dev.name G910 Orion Spectrum
dev.type keyboard
dev.version DeviceVersion(model=c33500000000, serial=3338470d, transport=8, protocols=(DeviceProtocol(0, product=0xc335, version='U1 v106.2.7', active=True), DeviceProtocol(1, product=0xaac5, version='BOTv114.0.7', active=False)))`

Donald4444, 2020-04-25:
Some output that could help figuring it out:

dev.name G910 Orion Spectrum
dev.type keyboard
dev.version DeviceVersion(model=c33500000000, serial=3338470d, transport=8, protocols=(DeviceProtocol(0, product=0xc335, version='U1 v106.2.7', active=True), DeviceProtocol(1, product=0xaac5, version='BOTv114.0.7', active=False)))

spectras, 2020-06-11:
Hello,
To be honest this part lacks unit tests, so it is possible the problem is actually on keyleds side.
Let's flag this as a bug until I get an opportunity to look into it.

Donald4444, 2020-06-11:
If I can help somehow (e.g. USB-information) just write it here

spectras, 2020-06-11:
Confirming it's a bug. I can reproduce it here.

spectras, 2020-06-11:
Can confirm this was a bug. Fixed on `master`.
Thanks a lot for reporting here, and really sorry for the delay. Since I started my new position in September, I have much, much less time for side projects.

closed 2020-06-11 by spectras

Donald4444, 2020-06-11:
Cool. Can you somehow add to the Python-documentation how you can get all Keycodes and change the RGB effect of a single key? I want to make my own daemon with the pythons-bindings, even though it's a little bit slower because of the interpreter layer

commit f79c452d (2020-06-11): Fix python key binding issues. Closes #52.