keyleds/keyleds #56: G512 (SE) layout.

nodgear, 2020-08-09:
For G512 (SE), the current keys are beeing randomly coloured and with unknow prefix.

Name:           G512 SE
Type:           keyboard
Model:          c34200000000
Serial:         31384707
Firmware[c342]: application U1  v150.0.2 [active]
Firmware[aafa]: bootloader BOT v110.0.dff5
Features:       [0001, 0003, 4522, 0005, 1e00, 4540, 1eb0, 8060, 00c2, 1801, 1802, 8080, 8070, 1821, 1bc0, 1806, 40a2, 1b04, 0020]
Known features: feature version gamemode name layout2 reportrate leds led-effects
Report rates:   [1ms] 2ms 4ms 8ms
LED block[01]:  104 keys, max_rgb(255, 255, 255)

!! These are the key and their numbers when using `keyledsctl -vvv`
left Shit: Key 42
left win: Key 125
left alt: Key 56
right alt: Key 100
fn: can't get
right control: 97
right shift: 54

I don't know if it's also related, but keyleds reports wave effect as not existent (default profile, right after installation)

spectras, 2020-08-09:
Hello, thanks for reporting here.
Well, the message you mention, about wave effect not being found, could be the issue, as the default effect is that one. In the absence of any effect, the led state is undefined (probably some uninitialized memory).

Could you try to edit your profile and add some other effect in the default profile (for instance a simple fill effect)?

Also, if the issue really is about the effect not being found, then it's a problem with the packaging. So knowing what distro you use and how you installed / which package you installed would be helpful :)

nodgear, 2020-08-09:
Made a basic template following the guide/docs about lua scripting.
everything is set to black, except for the unknow keys listed above.

Distro: POP_OS! 20.04
Ram: 16GB
CPU: Intel i3 4170
Conected I/O (USB): Logitech g403, Logitech G512SE, Alctron UM-01 (mic)

nodgear, 2020-08-10:
`set-leds -d 0962375D3734 all=purple`
Works just fine and set EVERY key, including the ones mentioned to purple.

Then i proceeded to run 
`KEYLEDS_DEVICE=0962375D3734`

Now, even if i use -d flag, the response is:
```
➜ keyledsctl set-leds -d 0962375D3734 all=purple
keyledsctl: set led block 01 -- feature not found on device
```

I hope that helps diagnosing my installation/keyboard

closed 2020-09-07 by nodgear