keyleds/keyleds #59: keyleds not working when display manager other than lightdm

stefan-paluch, 2020-11-08:
Hello
I was changed  display manager (`sudo dpkg-reconfigure gdm3`) to other than lightdm and keyleds stopped working - eg. it initialize on boot - set initial colors to key groups - but not execute animations or anything - when switch back to lightdm display manager - everything works as usual

spectras, 2021-04-08:
Hello, sorry for the long delay on answering here. If you are still around and it is still an issue, it would help a lot of you could get the logs from the service. :)

stefan-paluch, 2021-04-09:
> it would help a lot of you could get the logs from the service. :)

what logs will be helpful, and how to obtain them? (change logging verbosity of something?)

also - i downloaded recently "Logitech G HUB" - they upgraded it from last time i see it - now they got something i was thinking about: an audio visualizer (but in their version is horrible - it uses whole keyboard area removing all other effects and animations - i like it better if "visualizer" uses only media buttons row)

also also - they added interesting thing - highlight keyboard shortcuts - eg. in Word press [ctrl] - it light key [p] (ctrl+p = print) and [s] (save) and other keys included in [ctrl]+ shortuts