keyleds/keyleds #72: Compile error

Arian8j2, 2022-05-19:
### Summary
Today i found this repo, and wanted to compile it, But it keep getting errors like:
```
keyledsd/src/service/EffectManager.cxx:39:48: error: variable ‘constexpr const std::array<unsigned char, 16> keyledsdModuleUUID’ has initializer but incomplete type
   39 | static constexpr std::array<unsigned char, 16> keyledsdModuleUUID = {{
```
and some more when i added `#include <array>` to `EffectManager.cxx`.
```
keyledsd/src/tools/Paths.cxx:41:46: error: variable ‘constexpr const std::array<XDGVariables, 4> variables’ has initializ
er but incomplete type
   41 | static constexpr std::array<XDGVariables, 4> variables = {{
      |                                              ^~~~~~~~~
keyledsd/plugins/src/lua/LuaEffect.cxx:34:47: error: variable ‘constexpr const std::array<int (*)(lua_State*), 4> loadMod
ules’ has initializer but incomplete type
   34 | static constexpr std::array<lua_CFunction, 4> loadModules = {{
      |                                               ^~~~~~~~~~~
keyledsd/plugins/src/lua/LuaEffect.cxx:41:47: error: variable ‘constexpr const std::array<const char*, 25> globalWhitelis
t’ has initializer but incomplete type
   41 | static constexpr std::array<const char *, 25> globalWhitelist = {{
```
at the end it compiled without error when i added array includes to these three files:
- keyledsd/plugins/src/lua/LuaEffect.cxx
- keyledsd/src/service/EffectManager.cxx
- keyledsd/src/tools/Paths.cxx

### Versions
I compiled with *gcc* and *clang*, both gave me the same error.

gcc version **12.1.0**
clang version **13.0.1**
keyleds version **1.1.1**

tigaente, 2022-09-07:
I have the same issue. Adding the array-includes fixed it for me as well.

**Note:** If you change the src-files in your src-directory manually, be sure to use `makepkg -sie`. Otherwise, makepkg extracts the tarball again, overwriting your changes.

Urgentemente, 2023-09-18:
Thanks a lot for this fix, I recently did a clean install of CachyOS after running Arco for a couple of years (was moving to a bigger, faster NvME drive...and couldn't get gkeybind to install as keyleds compile was failing...

Just edited the files you mentioned and used makepkg -sie ,  and all looking good now ,saving me a lot of repetitive typing during a working day for starters, and can now how my pretty LED key patterns back :-)

Urgentemente, 2023-09-18:
Thanks a lot for this fix, I recently did a clean install of CachyOS after running Arco for a couple of years (was moving to a bigger, faster NvME drive...and couldn't get gkeybind to install as keyleds compile was failing...

Just edited the files you mentioned and used makepkg -sie ,  and all looking good now ,saving me a lot of repetitive typing during a working day for starters, and can now have my pretty LED key patterns back for nighttime :-)

ticpu, 2026-09-20:
Both this and #73 are fixed in my fork: https://github.com/ticpu/keyleds

The missing `<array>` includes are @NicBOMB's work from #74, carried over as https://github.com/ticpu/keyleds/commit/c99982d6138080af4ee3836b153ac38791b2d33d. The deprecated `std::iterator` warning from #73 is gone too: https://github.com/ticpu/keyleds/commit/eff78da8f0d4b45f0d20321390a816dd6267469c

CI builds and install-tests deb and rpm packages on every push, so this class of breakage gets caught now.